UltimateHackingKeyboard/firmware #401: Feature Request: Windows - Add natural scrolling option

mikhey, 2021-08-30:
This is a feature request for Windows - As a Mac & Windows user, I realize I may be in the minority with this request and it may be something I'd to look into myself - it would be nice to have an option that would toggle the Windows Registry flag `FlipFlopWheel ` that inverts the scrolling and essentially causes it to operate the same as natural scrolling on a Mac.  I've come to actually prefer it, oddly enough, especially since I have a USB switch connecting it to both a Mac & Windows PC.  It would be nice to have this as an option in the UHK Agent.

Here's some links that I've used in the past to do this manually:

https://www.windowscentral.com/how-reverse-scrolling-direction-windows-10

https://superuser.com/questions/310681/inverting-direction-of-mouse-scroll-wheel

---

I was able to accomplish this manually by changing this registry key value to 1, disconnecting and reconnecting the device:

`Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\HID\VID_1D50&PID_6122&MI_04\11&a0151f2&0&0000\Device Parameters\FlipFlopWheel`

mondalaci, 2021-08-30:
We'll implement this via [smart macros](https://github.com/UltimateHackingKeyboard/firmware/issues/351) on a per-module basis.

kareltucek, 2021-08-30:
So... lets sum up for clarity. The the solution to this problem is going to be a macro command to set scroll direction on demand (in the firmware, not in the OS), which is not what the OP asked for, but is the easiest and nearest substitute we are willing to provide (since OP's main problem is that his two OSes interpret scroll direction oppositely, so simply flipping the direction is not enough because it will flip directions on both OSes - this way, explicit action will have to take place on each keyboard startup to choose the direction).

mondalaci, 2021-08-30:
@kareltucek Correct. It'd be possible to have two different keymaps with keymap-specific macro events containing the proposed macro command. Eventually, we will transform Agent into a (non-mandatory) background application, and we could make it possible to specify a default keymap per Agent instance, hence fully solving such issues.

kareltucek, 2021-08-30:
Ok, implemented on top kareltucek/firmware#65

daneel23, 2022-01-19:
@mondalaci When are you expecting to be possible to invert scrolling on modules?

Is there plan for dedicated section for modules at the left sidebar?

mondalaci, 2022-01-24:
@daneel23 It'll be possible in the next release via [smart macros](https://github.com/UltimateHackingKeyboard/firmware/issues/351). Eventually, we'll create a dedicated UI as suggested.

mondalaci, 2022-03-19:
Already done via smart macros. Will be included in the next release.

closed 2022-03-19 by mondalaci