UltimateHackingKeyboard/firmware #402: Hyper modifier

JensAc, 2021-09-06:
Hi, 
I am not really sure, whether this belongs here. However, I am wondering whether it is possible to map the hyper modifier to some key. I know, the hyper modifier is somewhat outdated but it is supported by gnu Emacs. For this reason, it would be great to use it, in order to simplify some of the key chords in Emacs. 

I know, I could remap some key using xkb to hyper. However, this would require some configuration on the host side rather than on the UHK.

kareltucek, 2021-09-06:
Well, short answer - probably not possible.

Attempt on longer and hopefully correct answer would go as follows:

I believe that the ability to produce native hyper is effectively lost somewhere in translation `(scancode x scancodeSet) -> controlSequence`. That is, I believe there is no scancode recognized as hyper in scancode sets that are nowadays in use. This means that supporting native meta would mean having to make UHK use some of those ancient scancode sets.  (Such solution would then miss other nowadays-standard modifiers/keys, and overall would probably bring in its own set of challenges, since the old hardware would have to be simulated including all its quirks). 

If above is not intelligible:
- `scancode` is a one-byte identifier specific to a hardware key, sent by a keyboard when such key is pressed.
- `scancodeSet` is one of couple common translation tables that translate scancodes into some meaning on OS level. More-or-less, each scancode set corresponds to some hardware layout of a keyboard. (Usually less, because manufacturers are not dumb, and will simply make their keyboard work with an existing scancode set if a fitting one already exists.)
- `controlSequence` is some representation of the meaning of the key. I am here assuming terminal control sequences. I.e., printable characters correspond to themselves (usually 1 byte = 1 char), while nonprintable sequences (like `Control + <something>`) will be encoded using escape values - i.e., as a sequence of some special values that modify behaviour of the terminal and possibly interpretation of following bytes. Non-printables out of the first 32 ascii values (that is whites and non-printables) will be used as those escape values.

More reading:
- https://www.win.tue.nl/~aeb/linux/kbd/scancodes.html (especially chapters 1 and 10, on keyboard-level stuff)
- http://catern.com/posts/terminal_quirks.html (on terminal control sequences and their quirks)

luteijn, 2021-09-06:
this relatively old thread at https://bbs.archlinux.org/viewtopic.php?id=77202 may also help understanding some of the issues involved.

mondalaci, 2021-09-06:
USB only supports left and right Shift, Ctrl, Alt, and Super modifiers. Agent exposes all these eight modifiers and [every scancode imaginable](https://github.com/UltimateHackingKeyboard/agent/issues/857).

I'm closing this issue, but if you have further, concrete findings on how to implement Hyper, feel free to follow, and you're welcome to continue this discussion regardless.

closed 2021-09-06 by mondalaci

luteijn, 2021-09-06:
https://www.wisdomandwonder.com/article/10146/every-emacser-can-use-hyper-on-every-usb-hid-keyboard (99% of the time you don't need to differentiate between left and right modifiers, often just done in games to work around key aliasing/limited roll over)

https://irreal.org/blog/?p=5152 also makes some of the things clearer. Of course, these don't need to do anything special on the keyboard, just interpret some of the 8 modifier bits differently in emacs, but what uhk does allow you to do is change the layout so the right control key sets left control bit, and the key you want to be hyper sets the right control bit in the usb reports.

Personally, I still prefer a serial approach to keyboards,  like ps/2 or midi, with the host having to keep track of state, and consider the way usb hid handles things (in the common case) a step back. This is of course something the uhk can't really solve on its own, although I keep being tempted to just convert the uhk to a usb midi keyboard, generate simple note on/offs for the key down/up events and write the rest of the code to map these to input events on the host... and then my limited free time kicks in again.

JensAc, 2021-09-07:
Thank you very much for the prompt responses and the high quality of your replies. I see, there are some difficulties involved, but I can find my way to achieve what I want. If I find anything useful with respect to your comments, I will let you know.