UltimateHackingKeyboard/firmware #403: Mouse lagging when multiple mouse devices are connected at the same time to the same usb controller

ewancoder, 2021-09-07:
I have an issue which is maybe an issue of a motherboard / OS, but maybe it's an issue of the keyboard firmware since I did not encounter this issue previously with any other hardware.

Whenever I have the keyboard connected to some USB controller where I already have connected some other mouse (for example, wireless mouse, but it also reproduces with a wired mouse), **sometimes** (not all the time) mouse starts to really lag. It jitters and jumps from place to place when I move it. It is reproduced both when using trackpoint module and when using mouse controls on the keyboard, and it is also reproduced when moving separate wireless / wired mouse. If I disconnect the wired/wireless mouse - trackpoint/mouse features work perfectly ok on the UHK. If I disconnect UHK - separate wired/wireless mouse works perfectly ok. If I re-connect it again - sometimes it is reproduced as soon as I reconnect it, sometimes it takes some time and I need to wait many hours to catch this issue again.

I've also noticed (but I'm not sure that it's correct) that when I correct both mouse and UHK to the same USB hub, or I connect them both to my monitor which is connected by USB cable to the motherboard, or I connect them both to front USB ports of my PC case, or I connect them both to the top USB 2.0 ports on the back of my motherboard (which are supposed to use the same USB controller) - the issue reproduces, however it looks like it doesn't reproduce if I connect a mouse through usb 2.0 on the back of my motherboard and UHK to USB 3.0 on the front of my PC case (so different USB controllers). But again, this might be because I didn't manage to catch this issue when using devices on different USB controllers.

Also I've noticed that ALL inputs are laggy, not just the mouse. I just had a case when I had a lockscreen on and I needed to input my password. I didn't manage to input password correctly couple times because all the inputs were laggy and when I was pressing keys there were some delay. Until I disconnected my wireless separate mouse from the PC and it started working flawlessly.

I'd really like to solve this problem, be it a UHK issue or my motherboard issue. I have i9 9900k, AORUS Ultra z390 motherboard. Please tell me what I can do / which statistics / debug data I can gather in any way to get closer to the root cause of the problem.

ewancoder, 2021-09-07:
I think this is an OS/mobo issue. I just noticed a VERY strange behaviour. I wanted to record this issue but whenever I open the OBS software everything starts working perfectly. When I close it, mouse starts to jitter again consistently. The same thing repeats with Oculus application. It doesn't repeat with any other applications that I tested so far.

ewancoder, 2021-09-07:
I just tested it with two mouses instead of a mouse and a UHK and it works perfectly, so it looks like the issue reproduces when the UHK is connected to the PC. I'm running the latest official firmware and Keycluster and Trackpoint modules are connected to the keyboard. The issue also reproduces on the @kareltucek firmware.

This is the example of the first sentence of this comment written before I disconnected the mouse (with the issue present):

ijust ested it ih twomoues istead ofa moueand a UH andit worksperetly.

ewancoder, 2021-09-07:
If I disconnect both modules from the keyboard the issue is still present - mouse functionality of the keyboard and wireless separate mouse are still jittering / lagging.

ewancoder, 2021-09-07:
I just noticed that when connecting the wireless mouse to front of the PC case (separate USB controller than the UHK) mouse jitters for about 2 seconds then it starts to work normally. When disconnecting it from the front slot - mouse jitters again for about 2 seconds. Maybe some OS lag when enumerating devices or smth. I don't know, but this is very unnerving because I have some other hardware that requires precise smooth input and I want it to work perfectly every time I use it, but I'm afraid that it can have jitters from time to time when UHK is connected to my PC since it's also connected through USB cable.

mondalaci, 2021-09-07:
Can you reproduce the issue when every one of your devices is connected via USB 3 ports?

ewancoder, 2021-09-07:
Let me try.

ewancoder, 2021-09-07:
Actially it looks like I don't use any USB 2.0 ports right now, but I have my monitor connected through USB 3.0 cable in USB 3.0 port, and monitor has couple USB 3.0 ports and USB 2.0 ports, so I have my mouse connected to USB 3.0 port and keyboard to USB 2.0 port, but they are both effectively going into USB 3.0 port on the PC. Unfortunately I disconnected and reconnected some cables and now the issue doesn't reproduce again. I'll try to reboot the PC and will try to reproduce the issue with all the USB devices disconnected and just mouse and keyboard in two separate USB 3.0 slots.

The issue also reproduced when keyboard and mouse were inserted into USB 3.0 USB hub inserted into USB 3.0 slot on the PC, but I don't remember whether I had any USB 2.0 devices connected at that time.

ewancoder, 2021-09-07:
I can't reproduce it for now even with the same configuration, and even if I connect both UHK and the mouse to adjacent USB 2.0 ports. I'll keep trying and I'll get back to you once it happens again.

mondalaci, 2021-09-07:
I guess that USB 2.0 messes things up, and the UHK should be connected via USB 3.0 all the way, including the host port and USB hub ports. Connecting your other peripherals via 3.0 all the way as well would be a safe bet.

ewancoder, 2021-09-07:
Thanks, I'll try to keep it in 3.0 and will see if it helps. Thought I think it reproduced when I kept it together with a mouse in two USB 3.0 ports on the USB 3.0 HUB connected to a single USB 3.0 port. I can't afford a separate motherboard USB 3.0 port just for the UHK since I have a lot of devices connected that require big bandwidth / stable connection.

mondalaci, 2021-09-07:
Feel free to use a USB hub, but make sure it's purely USB 3.0

ewancoder, 2021-09-07:
Is it ok that the UHK itself has USB 2.0 cable?

mondalaci, 2021-09-07:
No, it shouldn't be a problem. The UHK actually doesn't utilize USB 3.0, but some hosts have problems with USB 2.0 ports nevertheless.

ewancoder, 2021-09-08:
I just had it reproduced through USB 3.0 hub though I was using @kareltucek firmware, it was strange. It was laggy for about 3 seconds then normal again for about 5 seconds, then laggy again for about 3 seconds and like this in a cycle. When I disconnected my wireless mouse from the hub (only UHK remained, USB 3.0 connection) the issue still remained. After disconnecting and reconnecting UHK the issue disappeared and I cannot reproduce it again.

I'll flash the latest official firmware again and will wait for the same result using official firmware just to confirm it's not the fork to blame.

ewancoder, 2021-09-08:
ok I don't think this is the issue in the firmware. I just managed to reproduce it with just a single mouse (without UHK whatsoever, though to reproduce it I was connecting and disconnecting UHK too). I think something is wrong either with my OS or my hardware. sorry for bothering you

ewancoder, 2021-09-08:
I'm not sure that it's the fix but here's what I did and it seemed to fix the issue for now (one day of usage). I noticed that Device Manager constantly updates the list of devices for some reason, as frequent as once per second sometimes. I went into Universal Serial Bus controllers, then selected manually every device found there (including Root Hubs / motherboard USB controllers), went into Power Management tab in properties and unchecked the checkbox "Allow the computer to turn off this device to save power". This fixed the issue with DeviceManager constantly updating the list of devices and it seemingly fixed the issue with the mouse (at least for now). I sincerely hope it was the cause of the issue and I finally fixed it, I'll get back to you in a week or so to confirm that I have never faced it again (if I am right). It looks like my Windows constantly tried to disable and re-enable some devices which might cause mouse lags / jitters, maybe it starts to do this when you have too many USB devices connected (I have a lot).

This is a very specific use case that (as far as I understand) nobody else has faced before, but maybe it makes sense to add this info to some FAQ / troubleshooting page (after I'll be able to confirm that the issue did not reproduce).

mondalaci, 2021-09-08:
Thanks for sharing your findings! I don't include this case in our FAQ because we haven't encountered it so far, so it's not a frequent issue. This thread can be beneficial for some, regardless. I'm closing this issue, but we're interested in your findings, so please keep us updated.

closed 2021-09-08 by mondalaci

pcooke9, 2021-09-09:
I don't know if this could be related or not, but I've had occasional mouse twitches for a few months now. No noticable lag or major jumps, but small twitches of a centimeter or so. I don't have any UHK modules. I first thought it was my new EVGA X17 mouse because it's had several firmware updates recently, but I've only noticed the twitches when I have the UHK connected.

While troubleshooting, I discovered that I get the following message in Windows Event Viewer upon first use of a mouse function after disconnecting/reconnecting the UHK, regardless of any other mice being connected. The message appears with firmware v8.10.10, and doesn't with v8.10.9:

Log Name:      System
Source:        mouhid
Date:          9/9/2021 4:15:53 PM
Event ID:      1
Task Category: None
Level:         Warning
Keywords:      Classic
User:          N/A
Computer:      
Description:
The driver has detected that this HID mouse has bad firmware.  It is reporting a bad physical minimum/maximum for one or more axes.

mondalaci, 2021-09-10:
Can you reproduce the USB error with [this firmware build](https://github.com/UltimateHackingKeyboard/firmware/issues/390#issuecomment-916093400)?

pcooke9, 2021-09-10:
> @pcooke9 Can you reproduce the USB error with [this firmware build](https://github.com/UltimateHackingKeyboard/firmware/issues/390#issuecomment-916093400)?

Unfortunately yes, the warnings continue with that build.

mondalaci, 2021-09-13:
@pcooke9 This is a one-off issue, and it seems to be hardware-specific. Can you reproduce it without any hub and only via the USB 3.0 ports of your computer?

pcooke9, 2021-09-14:
@mondalaci My motherboard (MSI x470 Gaming Plus Max with Ryzen 9 3900X) has USB2, and USB3 Gen1-5Gbps & Gen2-10Gbps. It happens on all of them. I normally only connect the UHK directly to the MB's back panel, but I just tried it through a powered USB3 hub today to see if it could be a power issue. I even tried installing a USB3 PCIe card, but I still get the same results.