UltimateHackingKeyboard/firmware #404: Keycluster mouse keys not working on first plug-in

Mike-Hermans, 2021-09-08:
Tested on the following:
- Ubuntu 20.10 (Dell XPS 9300)
- Windows 10 (HP Omen 15)
- UHK Agent 1.5.15
- Right keyboard half firmware: 8.10.10
- Left keyboard half firmware: 8.10.10
- Key cluster firmware: 8.10.10
- Trackball firmware: 8.10.10 

**First contact with support (from me)**
I've just received my 2 modules, the key cluster and the trackballmodule. The trackpoint works without any issues, but for the keycluster none of the mouse clicks register. The trackball itself can be used to scroll, but when I click it nothing happens. I've 
- Updated the firmware on my UHK
- Reconnected the modules
- Reset the configs
- Reconnected the keyboard itself
but nothing seems to help.

**Solution**
After remapping on of the keys in the firmware, and saving it to the keyboard, the mouse keys worked again. The next day, they didn't work, and after doing the same thing as before, they worked again. They have worked since, without any issues.

I was asked by support to report this as an issue, so I've provided as much information as I could.

ewancoder, 2021-09-08:
This happened a few times for my KeyCluster module as well but it was very rare case for me.

mondalaci, 2021-09-08:
@Mike-Hermans Thank you very much for your report! On second thought, this issue is surely a duplicate of #361, so I'm closing this one.

closed 2021-09-08 by mondalaci

mhantsch, 2021-09-13:
I think this issue is the same as I reported in #389, and therefore would be fixed with #348.