UltimateHackingKeyboard/firmware #405: layer indicator LEDs flashing

adherr, 2021-09-10:
Since adding the key cluster and trackpoint module, and flashing to 8.10.10 firmware (I don't know if this issue was happening before the flash) the `mod`, `fn`, and `mouse` LEDs flash.

- They appear to be on for a few ms, then turn off for several hundred ms. 
- `mod` is the brightest, implying it's on the longest
- flashing is not quite regular. It's mostly around 2 flashes per second, but sometimes 3 and sometimes fewer.
- it is not related to typing, but using the trackpoint makes it slow down somewhat
- rebooting the keyboard makes the flashing go away for a little while, but it always returns

mondalaci, 2021-09-11:
This is likely a power-related issue. Please try points 1-8 of [this guide](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys), and let me know your findings.

adherr, 2021-09-13:
Switching to a powered USB hub seems to have solved the problem. Thanks!

closed 2021-09-13 by adherr