UltimateHackingKeyboard/firmware #408: Customize Colors

Freaksed, 2021-09-28:
Feature Request
It would be nice if you could set the color of keys in different layers/modes.
The use case here is to highlight shortcut keys in different software.
For me it's mostly gaming, but also things like Blender which have basically every key set to do something it would be nice to be able to create a map where you can color the different keys based on what "kind" of shortcut it is.

mondalaci, 2021-09-29:
Thanks for elaborating on your use case! It makes sense, and chances are others will request this feature, too. If someone is reading this and wants to have this feature, please upvote the first comment with a thumbs-up emoji and add your use case.

yungslime, 2021-11-12:
Just received my 60 V2 today, and I agree. customizable colors would be a great addition to an already great keyboard!

mkove, 2021-11-28:
also agree.

djrenren, 2021-11-30:
For myself:

I find that the classes keys for each color can be a little off of what I expect.
For example, I expect non-textual keys such as arrows or escape to be different from the basic letter keys. I understand the classifications now but i have layouts where `~ is escape and ones where it is the character, and it would be nice to have visual distinction. 

(Just got my UHK2 yesterday and it's so great! <3 )

mondalaci, 2021-12-04:
How about making colors customizable on a per-keymap basis, guys? You could enable custom colors per keymap and then customize the color of every key of every layer of the enabled keymaps.

For the keymaps that don't have custom backlighting enabled, <a href="https://ultimatehackingkeyboard.com/blog/2020/11/05/introducing-the-uhk-60-v2#functional-backlighting">functional backlighting</a> would be active. You will be able to configure the 7 colors of functional backlighting.

Alternatively, Agent could enable custom backlighting for all of your keymaps at once. The drawback is that custom backlighting consumes a lot of on-board EEPROM memory. It'd be worth being conservative to have enough EEPROM for future features and only store custom backlight colors for select keymaps.

I guess the question boils down to whether you'd enable custom colors per keymap, or would you only be interested in enabling it globally.

kareltucek, 2021-12-04:
@mondalaci what about main RAM segment? Will we want/need to unpack current led keymap there, or is it feasible to just point into the config segment? (We would have to build module+layer led config-coordinates map during config parse in order to be able to do so fast.)

I was thinking about having a per-keymap 16-colour palette, and then using 4 bits per key to point into the palette. The numbers are:

- full rgb keymap would take 35*4*12*3 = 5040 bytes
- colour palette-based keymap 35*4*12/2 + 3*16 = 888 bytes

Pros, colour palette would allow easier management of the colours - user could specify his segments without having to worry about picking the colour first, and just then could simply adjust colour palette. This might also allow to easily customize colours of the functional highlight.

Cons, it's more complex proposal and you strongly dislike complexity.

mondalaci, 2021-12-04:
@kareltucek I think a fair number of users would be unhappy with a 16-color palette. It's also worth considering that we'll [expose LEDs via USB](https://github.com/UltimateHackingKeyboard/firmware/issues/322), allowing for fancy gradients and animations, so I'd much rather stick with `LedMap` and overwrite it with custom RGB values.

kareltucek, 2021-12-04:
Sure, if you think so.

>  so I'd much rather stick with LedMap and overwrite it with custom RGB values.

You are implying that I suggest removing LedMap. Just to be clear, I am not suggesting that.

scottjl, 2021-12-06:
> I guess the question boils down to whether you'd enable custom colors per keymap, or would you only be interested in enabling it globally.

Well, if I can have everything I want, I'd want to be able to set up custom colors per keymap. But I'm also guessing 90% of the people out there won't go to that degree of definition. I'd be happy enough to be able to define globally with the ability to load a few setups onto the keyboard directly and swap between them with a custom keypress.

nsflamp, 2022-01-31:
Personally I would love to be able to define it per keymap. It would be helpful to have different lighting setups for different applications as often shortcuts are very different across what I use. In terms of onboard memory it could be worth having an easy import export setup perhaps?

Also just wanted to say absolutely loving this keeb.

mondalaci, 2022-01-31:
@nsflamp Thanks for your input and kind words! Per-key colors will be part of the UHK configuration, so they will be imported and exported together with the UHK configuration.

kareltucek, 2022-02-05:
@mondalaci I am still unclear on:

How exactly do you plan to fit it into memory?

  - Do you intend to reparse the config on any layer change in order to set the `LedMap` from there? 

  - Or do you intend to keep a copy of full RGB setup in RAM all the time? (If so, where do you plan to get those 5kb of RAM?)

How exactly do you plan to make editing the maps user friendly? 
  
  - Are you planning to have some concept of a colour palette?

  - If no, then how do you avoid having to manually change colour of every single key on every layer when the user just wants to slightly alter a colour of some subset of keys?

mondalaci, 2022-02-05:
@kareltucek Let's reparse the actual layer colors from the config and set `LedMap` from there upon layer changes.

I'll make an Agent issue regarding color editing, but it won't affect the firmware. I think there are many legit use cases regarding color editing. I don't think that slightly altering the color of some subset of keys will be a frequent use case. In any case, we should keep the UI simple and possibly extend it later according to user feedback.

kareltucek, 2022-02-05:
>  I don't think that slightly altering the color of some subset of keys will be a frequent use case.

I disagree. On the contrary, I think it will be a very frequent usecase, since in order to see how the lighting looks, you need to set it up first. Just when you see the backlight fully set up, you can start figuring out why it looks bad, and start making small incremental changes. 

I think that most usecases will go as follows:

1) First figure out your key "classes". I.e., which types of keys you want to distinguish by colours. In most cases, this will depend on the app in use or some other semantic distinction. (Feedback above supports this hypothesis.) For instance, vim might define classes as "edit commands", "movement commands", "other" and "special".

2) Draft a colour palette for them.

3) Set it up in Agent.

4) Test it on real keyboard.

5) Find out that your RGB codes look very different in real life than what you expected. Maybe some colours blend together, maybe some look outright ugly. Now:
  
    a) Maybe you just need to fine-tune particular colours.

    b) Maybe you just didn't think the aesthetics of your colour palette through, and want to apply a totally different palette, but to the same set of classes.

6) In either of the cases, you end up wanting to change the colours assigned to each class, without actually changing the classes. 

Maybe I am missing something, but expecting people to do these finetunings by assigning new colours one key at a time is madness.

mondalaci, 2022-02-06:
The mentioned use case makes sense. Still, I'm unsure how frequent it will be. First, we'll aim for a simple implementation. We can implement such a feature later in Agent without involving the firmware.

kareltucek, 2022-02-06:
Just some more thoughts about this:

- These maps will cost a lot of memory. 

- I think it might be more efficient to aim for an efficient storage format from the beginning (even if its potential is not fully employed in the Agent).

- Palette-based indirect approach would be 2x - 6x more efficient (w.r.t. config space), as it is form of compression.

- For the same reason, it might make sense to create separate layer colour maps instead, and assign them on per-keymap-per-layer basis. (If these are placed at the end of the config, the indirection can still be easily resolved with a single config-parse pass.)

- As for the memory, I intend to try to free up one of the two 32kb regions eventually anyway. I would prefer not to waste the freed memory inefficiently, but reserving some space for a colour palette of reasonable size (anything between 16 and 256) is definitely possible, and would allow more efficient use of the config space.

- As for the labour needed to implement the firmware side, I don't think the two approaches make a big difference.

- The proposed complex approach might be counterproductive if you already have some other form of compression in mind.

kareltucek, 2022-02-06:
On second thought, the indirect approach might not be very efficient if we assume small number of layers and a big palette.

mondalaci, 2022-02-06:
I'd not rule out the possibility of a big palette, keep things general for the time being, and possibly optimize later. I'm aware of the large storage space of my proposed color storage scheme.

If you plan to free up a 32 kB region, I assume that means that the whole configuration will not be held in the RAM anymore. If so, we must discuss which items you plan not to hold in RAM because such decisions may have far-reaching implications.

kareltucek, 2022-02-06:
I was thinking about the staging buffer.

mondalaci, 2022-02-07:
How do you plan to remove the staging buffer and still provide configuration validation?

Just finished with the [Agent-side specification](https://github.com/UltimateHackingKeyboard/agent/issues/1688) of the per-key colors feature, which contains firmware-relevant information regarding the user configuration format. The next Agent release won't contain this feature yet.

kareltucek, 2022-02-07:
No specific plans yet, but config validation is a single pass procedure, and the space is not used at all during normal operation...

mondalaci, 2022-02-07:
@kareltucek Any specific plans, please let me know.

DarwinAwardWinner, 2022-02-22:
My use case is also to distinguish multiple different "kinds" of shortcuts, where "kind" is defined in some context sensitive way. Just as one example, for the mouse layer, I'd like to make the mouse movement keys a different color from the scrolling keys, since they do semantically different things, and because the "inverted T" shape of the movement keys is very useful as a visual reference point. Another example, I might want to make the tab-manipulating keys on the mod layer a different color from the window/workspace switching keys.

Even simple and subtle things like making the letter keys on the base layer a different color than special keys like numbers, tab, backspace, etc., or making the home row a slightly different shade, again useful as a visual anchor when first placing my hands on the keyboard.

Regarding memory usage, I would generally be leaving most of the keys on the keyboard at their "default" colors and customizing only a few, and I would be using a small number of unique colors. I can't imagine ever wanting more than 16 unique colors on a single layer (or even probably across all layers).

DarwinAwardWinner, 2022-02-22:
Just a thought, but would it make more sense, in terms of memory usage, to associate colors with specific key mappings or actions, instead of physical keys? For example, I could assign a specific color to the arrow keys, and then any key mapped to left/right/up/down arrow would get that color automatically. I guess this would lose some flexibility, but maybe save some memory since the color mapping is no longer specific to a keymap or layer?

mondalaci, 2022-02-22:
The suggested use case would fit some but not others. Let's stick to the original plan of assigning any color to any key for maximum flexibility.

mondalaci, 2023-06-27:
This feature has been implemented and will be included in the next firmware release.

closed 2023-06-27 by mondalaci