UltimateHackingKeyboard/firmware #409: Application specific scroll jumps.

kareltucek, 2021-10-01:
Split from - https://github.com/UltimateHackingKeyboard/firmware/issues/383

ewancoder, 2021-10-01:
Can we make a version of firmware that allows trackpoint to scroll only up and down? (to test one theory) I suspect this might be happening only when there is both vertical and horizontal input. When I'm trying to be very precise I can scroll the page (though jumps still happen). I tried to still scroll very slowly down (page in xodo) but at the same time tilting the trackpoint to the side and it almost didn't let me scrolling the page.

kareltucek, 2021-10-01:
Sure [uhk-firmware-8.10.10-scroll_issue_v1.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7265529/uhk-firmware-8.10.10-scroll_issue_v1.tar.gz) (with x axis removed).

Just for the record - it seems that the browser is separating the axes. I tried the other thing - making firmware always send y coordinate as both x and y, and the log still shows x and y events separately.

ewancoder, 2021-10-01:
Thank you, yes now I can scroll everything with Xodo software without any issues! Windows 10 audio devices list also works flawlessly. Maybe we can add some command to the macro engine to disable horizontal axis / configure which axes are available in trackpoint module? Would be nice to have an option to disable it since I don't really use horizontal scrolling. And even if we find the root cause and fix it in the main firmware, it might still be a good feature to be configurable since in some softwares (like CAD) you might want precise control and to scroll with your mouse only vertically so you could trigger some macros to temporarily disable horizontal axis.

kareltucek, 2021-10-01:
So what exactly is the hypothesis? 

I don't see why should occasional horizontal tick be a problem.

Regarding scrolling, the testing firmware allows to axis-lock scrolling:
```
    COMMAND = set module.MODULEID.scrollAxisLock {0|1}
```

ewancoder, 2021-10-01:
I have no idea why it happens but if both axes are enabled then I'm having a very difficult and frustrating time scrolling in some applications like windows 10 audio device list and Xodo PDF reader. In chrome pages vertical scroll works but horizontal scroll exhibits the same symptoms. I suspect this is happening only when both inputs for horizontal and for vertical scroll are triggered at the same time (or close to each other), since it's impossible to scroll up or down without scrolling slightly sideways on the trackpoint. This is the screencast of what the issue looks like on the chrome page, I'm just holding the trackpoint pointing it to upper left corner so the scroll should always go up and to the left, however you can see that the page jerks slightly to the right from time to time, and there are positive values in the log as well.

https://www.dropbox.com/s/j7yq98w37py0qgp/trackpoint%20issue%203.mkv?dl=0

mondalaci, 2021-10-01:
I'm wondering if this issue is reproducible when vertically scrolling with a wheel of a mouse, and at the same time, limiting UHK scrolling to the horizontal axis and scrolling horizontally with it. Maybe some operating systems or applications handle interspersed vertical/horizontal scrolling strangely.

ewancoder, 2021-10-01:
hm so can I configure it with test firmware @kareltucek ? or do we need to make another one? set module.MODULEID.scrollAxisLock 1 or 0 gives me ERR on the LED and nothing changes

kareltucek, 2021-10-01:
No, I can't reproduce it. 

Also, I am not sure if this is sufficient evidence to blame UHK - scroll mode differs from cursor mode just in putting the value into a different data field - if it was uhk, you should be able to reproduce it with regular mouse movement.

Correct formula is of course:
```
set module.trackpoint.scrollAxisLock 1
```

Also, here is a firmware that is (hopefully - didn't test it) restricted to horizontal scroll only - [uhk-firmware-8.10.10-scroll_issue_v2_x_only.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7265946/uhk-firmware-8.10.10-scroll_issue_v2_x_only.tar.gz)

kareltucek, 2021-10-01:
(Also, there is scroll macro action.)

ewancoder, 2021-10-01:
Sorry yes I've used trackpoint instead of MODULEID of course, it gave me ERR for some reason.

Yes, v2 limits trackpoint to horizontal scroll only successfully. And yes I'm able to reproduce it by using trackpoint to scroll horizontally while using a regular mouse to scroll vertically. Both in Xodo and Windows 10 audio devices. And on that huge web page as well.

kareltucek, 2021-10-01:
> Sorry yes I've used trackpoint instead of MODULEID of course, it gave me ERR for some reason.

You can use `printStatus` command to print out the reason into a text editor.

kareltucek, 2021-10-01:
So you need to move in both directions in order to reproduce the problem, right?

Horizontal scroll alone or vertical scroll alone causes no issues?

ewancoder, 2021-10-01:
> $onInit: 4: 7: parameter not recognized:: scrollAxisLock 1
(this is on v2 firmware though, didn't check with v1 yet)

So I guess this is OS/Software issue then. Since I'm using UHK to scroll only horizontally and a mouse to scroll vertically. The symptoms are the following: the harder I press on the trackpoint (to the side), the harder it is to scroll up or down (in xodo & audio devices list). On the web page when I'm holding trackpoint to scroll to the left and then I scroll the separate mouse up or down, page jumps a bit to the right before going back to the left which is very strange. This is the screencast of all these cases, first slightly touching trackpoint then pressing on it firmer https://www.dropbox.com/s/bniapmi4ckc4r72/trackpoint%20issue%204.mkv?dl=0

Yes. If I disable horizontal scroll everything works flawlessly for me. This is probably what I'm going to do as an end-user if this feature is available later in the firmware (to disable horizontal scroll on the trackpoint).

ewancoder, 2021-10-03:
No mouse jumps with scroll-issue-v1 firmware so far even though I use trackpoint all this time.

kareltucek, 2021-10-03:
Great!

Would you try the tp_hunt_v8 version with `scrollAxisLockEnabled 1`? The thing does not disable x axis - it rather snaps scrolling to one of the grids. I would like to know whether the feature works reliably for you or causes some issues. If things go as planned, it is going to be accepted into stock master soon, so there would not need to be any further modifications...

ewancoder, 2021-10-03:
Yes it works, at least it's much better than without snapping. There's one issue with it: if you scroll up or down slowly and move trackpoint to the left or to the right (I guess to the extend that it's tilted more to the side than vertically) then page in Xodo jumps back. But I don't think this is the issue for me, I can reliably scroll the page up or down without jumps if I just tilt the trackpoint only vertically without side movements (which is rather easy to do).

ewancoder, 2021-10-04:
An offtopic question: do you know why on v8-hunt and v1/2 firmware labels are not working anymore? `$label: some command`. It says "Label not found". Did some format of labels change in the latest build?

kareltucek, 2021-10-04:
It probably does not like the `$` at the beginning. 

Reason behind is that the fork has already moved to using the command action only (https://github.com/UltimateHackingKeyboard/agent/pull/1545), so the `$` notation was dropped alongside with the text action support. These testing firmwares are just "hacked" to parse all text actions as command actions, since there is still no Agent release that supports command action.

kareltucek, 2021-10-04:
> . But I don't think this is the issue for me, I can reliably scroll the page up or down without jumps if I just tilt the trackpoint only vertically without side movements (which is rather easy to do)

Happy to hear that!

So I guess this issue can be closed then. Of course, feel free to follow up...

closed 2021-10-04 by kareltucek

ewancoder, 2021-10-04:
Thanks! Yes we can close it. My macro works perfectly now that I removed $ signs :)