UltimateHackingKeyboard/firmware #410: Trackpoint module is bricked, cannot recover

ebr, 2021-10-05:
I submitted a request via the contact form, but this is weird enough to make an issue about.

After updating to firmware 8.10.10, the trackpoint module appeared bricked. I tried the recovery process, it is not being recognized at all:

```
[UhkHidDevice] Used device: {"vendorId":"0x1D50","productId":"0x6122","path":"/dev/hidraw1","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v1","release":258,"interface":0,"usagePage":65280,"usage":1}
[UhkHidDevice] Device communication closing.
[UhkHidDevice] Device communication closed.
[DeviceOperation] Cannot ping the bootloader. Please remove the "Trackpoint" module, and keep reconnecting it until you see this message.
[UhkOperations] Couldn't connect to the "Trackpoint".
[DeviceService] Module recovery error {"message":"[UhkOperations] Couldn't connect to the \"Trackpoint\".","stack":"Error: [UhkOperations] Couldn't connect to the \"Trackpoint\".\n at t.UhkOperations.<anonymous> (/tmp/.mount_UHK.Ag11f2Bf/resources/app.asar/electron-main.js:2:1251190)\n at Generator.next (<anonymous>)\n at s (/tmp/.mount_UHK.Ag11f2Bf/resources/app.asar/electron-main.js:2:1307260)"}
```

I then updated the keyboard *only* to 8.10.11, which did not help.

The weird part: If I disconnect the left half completely and connect the trackpoint, the Agent recognizes the trackpoint as a "left half". If I then reconnect the left half without disconnecting the trackpoint, the indicator LEDs (the 14-segment display and status icons) light up fully clearly indicating an error. **It seems like** the left half firmware 8.10.10  got flashed to the trackpoint instead of the trackpoint firmware (not sure if that's possible, but seems like it).

ebr, 2021-10-05:
I just fixed it. 

:arrow_right: TL;DR: It IS possible to flash the _left half_ firmware onto the trackpoint module, after which the trackpoint module appears bricked and unrecoverable. *This should not be possible* - the Agent should prevent this. I fixed it by tricking the updater into flashing the trackpoint firmware *without the left half connected*.

### How I fixed it:

Hypothesis: trackpoint got flashed with the left half firmware.

Experiment:

1. Untar the firmware file into a directory
2. `cd modules`
3. Copy both trackpoint files over their corresponding left half files: `cp trackpoint-right.map uhk60-left.map; cp trackpoint-right.bin uhk60-left.bin` 
4. create a new "frankenfirmware.tar.gz` archive
5. **disconnect** the left half fully and connect the trackpoint
6. flash the frankenfirmware. Trackpoint is instantly recognized.
7. Reconnect the left half and delete the frankenfirmware file - I assume it's possible to brick the left half in a similar way.

mondalaci, 2021-10-05:
Thanks for sharing your findings! I can't see how flashing the left half firmware to a module is possible under normal circumstances, and we haven't heard of such an issue yet.

It's useful to know that right-clicking on the "Flash firmware" button and choosing "Unbrick trackpoint module" should unbrick it.

Given the one-off nature of this issue, we won't look into it, but if it reoccurs, we will.

closed 2021-10-05 by mondalaci

ebr, 2021-10-09:
@mondalaci FYI, I strongly suspect this is possible when the left half is disconnected during the firmware update process. I went to update the firmware in the first place because my KB was behaving strangely, but it turned out that one of the RJ11 connectors was flaky.

If the trackpoint module is connected but the left half isn't, the module becomes a target for the left half update. This seems like a problem and I believe the FW update process needs to guard against this, as it could happen due to an unreliable cable connection.

I would test this myself bit I'm worried about bricking the module permanently this time.

mondalaci, 2021-10-09:
@ebr The proposed scenario shouldn't be possible because the bootloaders of the left half, right half, left modules and right modules [all have different I2C addresses](https://github.com/UltimateHackingKeyboard/firmware/blob/e68ead6ba36728011e1eabdc49663336b77a9f06/shared/i2c_addresses.h#L31-L42).