UltimateHackingKeyboard/firmware #416: Enhancement: TrackPoint speed configurable

vorburger, 2021-11-07:
(Caveat: I've got my UHK v2 *yesterday* and am at the very beginning of exploring this **ultimate gadget** - Thank You for having created this!)

Creating an issue to track possible future TrackPoint speed configurability - given that the Agent's Mouse speed setttings don't seem to have anything to do with the TrackPoint (if I understood correctly).

Understood that one can use OS' setting for this, e.g. Linux GNOME Mouse & Touchpad's slowest Mouse Speed.

But UHK default TrackPoint speed appears fairly high, and noticeable higher than e.g. a [ThinkPad USB keyboard](https://www.google.com/search?q=thinkpad+usb+keyboard) connected to a (Linux) desktop; only lowest setting seems about useful (to me, of course YMMV).

vorburger, 2021-11-07:
I peeked for all of about 10s ;-) at https://github.com/UltimateHackingKeyboard/firmware/blob/master/trackpoint/src/module.c to try to understand what determines the speed there... hacking this, on a hackable keyboard, could be fun - but I couldn't (immediately) make sense of how that works, and modifying which line could change the speed for a custom firmware build.

kareltucek, 2021-11-07:
214 / 215

You could also do the same in right half's firmware in mouse_controller.c

Or simply wait for the "smart macro" feature.

mondalaci, 2021-11-07:
@vorburger Thanks for your kind words!

Closing this issue as a duplicate of #351.

closed 2021-11-07 by mondalaci