UltimateHackingKeyboard/firmware #418: Option to automatically change between Mac <-> PC keymap

vvolkgang, 2021-11-12:
Lately I've been switching between Mac and PC frequently, I only use QWERTY so would appreciate if there was an option to detect the OS and automatically switch between Mac / PC versions of the default keymap.

kareltucek, 2021-11-12:
This has been already discussed in https://github.com/UltimateHackingKeyboard/agent/issues/1209 .

Currently you have two options:

- set up a script on your machines as mentioned above.
- set up an "init" keymap which will force you to pick one keymap or the other whenever you reconnect a keyboard. Not automatic, but will save confusion from having one of the keymaps as a default one.

mondalaci, 2021-11-12:
Thanks for providing the correct answer, @kareltucek. Closing this issue, but feel free to follow up, guys.

mondalaci, 2021-11-12:
@vvolkgang Actually, I'm wondering if it'd be useful if Agent implemented the automatic keymap switch feature. For this to happen, you'd have to keep Agent running in the background all the time.

vvolkgang, 2021-11-12:
> @vvolkgang Actually, I'm wondering if it'd be useful if Agent implemented the automatic keymap switch feature. For this to happen, you'd have to keep Agent running in the background all the time.

Was initially going to post this in Agent for that reason, it would be enough for my use case (I only switch between PC/Mac's I own and can install the agent).

kareltucek, 2021-11-12:
>  it would be enough for my use case

But would you fee running the Agent all the time in the background justified for that usecase? (Electron is quite a heavy thing after all.)

> Actually, I'm wondering if it'd be useful if Agent implemented the automatic keymap switch feature.

Are you thinking about saving the configuration agent side or keyboard side? (Both have their pros and cons.)

mondalaci, 2021-11-12:
We'll be introducing features that will depend on Agent due to their implementation details, so running Agent in the background will be unavoidable for such features. Rewriting Agent in another language or platform is not practically feasible. It'd be possible to extract the relevant parts of Agent and create a separate service, but I don't think it's worth the effort unless users start complaining.

This feature wouldn't involve saving the configuration. Agent would send a switch keymap command upon connecting to a UHK.