UltimateHackingKeyboard/firmware #425: Reapply default smart macro values on save configuration

mondalaci, 2021-11-21:
While experimenting with various smart macro values, I often uncomment or delete some lines, try others, then save my configuration. For example, I swap axes, then delete the relevant line, and the axes stay swapped. In this case, I'd prefer the axes not to stay swapped. I can explicitly restore the axes with a command or power-cycle my UHK, but I'd rather not spend time with it.

I think it'd be practical to reapply default smart macro values on saving the configuration (and afterward call `$onInit`, then call `$onKeymapChange ${currentKeymap}` as usual).

kareltucek, 2021-11-22:
Is it really worth the trouble? (Yes, my laziness speaking.)

Also, how to implement it? The best idea I can think of now is to move all configurable variables into its own "configuration manager" module and there to create a "config" struct which would have two instances - one constant for defaults and the other for actually applied values.

mondalaci, 2021-11-22:
I can feel your pain, but I think the suggested behavior is more intuitive than the current one.

The implementation you suggested is the cleanest I can think of.

mondalaci, 2022-09-03:
I'm closing this issue because maybe I'm alone with this request, and I don't want to cause extra work for no good reason.

closed 2022-09-03 by mondalaci