UltimateHackingKeyboard/firmware #426: fade leds after some time of inactivity

jalberto, 2021-11-22:
I would like to see a way to configure a save-energy mode where the leds fade out after some minutes of inactivity.

Drizzt321, 2022-04-29:
Ohh, I'll second this request. When I go to bed, it's nice if the backlights were to go off, eventually, after a short timeout to have less light floating through my apartment.

mhantsch, 2022-05-03:
Fade to complete darkness, or fade to a very low light level?

mondalaci, 2022-05-03:
I think "fade to complete darkness" is what the vast majority wants, so I wouldn't introduce further options now.

kareltucek, 2022-05-03:
So this is about both the backlight and segment display, right?

Drizzt321, 2022-05-03:
> I think "fade to complete darkness" is what the vast majority wants, so I wouldn't introduce further options now.

I'd like fade to complete darkness. If it's just as easy to put in an option for fade to a specified point that'd be neat, but as @mondalaci said, complete darkness is what I'd like/prefer.

mhantsch, 2022-05-03:
> So this is about both the backlight and segment display, right?

Ah yes, so this would work on UHKv1 too, and fade the LED segments (and icon indicators). I like!

mondalaci, 2022-05-04:
Closing this issue as a duplicate of #437.

closed 2022-05-04 by mondalaci