UltimateHackingKeyboard/firmware #429: Limited number of actions in a single macro

bistabil, 2021-11-30:
I don't have an issue report but just some questions regarding the macro engine.

I've created some long macros in past but never before did I need a macro consisting of 300 actions or more.
After a certain number of actions are added, I've noticed Agent becomes less responsive - scrolling, editing, opening actions.

1. Is the beforementioned drop in responsiveness influenced by the keyboard hardware or how powerful the host machine is?
If this is not related to the keyboard then maybe the underlying handling of actions inside of Agent can be optimized because I have i9 9900k and wouldn't expect this huge performance drop. I actually had to continue creating the macro (after 100 or 150 actions) by editing the configuration exported to JSON, then importing it back after modifications were done because Agent was just unresponsive.

Next, I've found out that after a certain point (somewhere between 200 and 300 actions) when trying to run the macro, that "ERR" appears on the display.
I've checked if I missed something in the macro multiple times and I am pretty sure I didn't make a mistake (using Karel's version of firmware)

2. Is there a hardcoded limit of the number of actions that can be registered inside of a single macro?
If there is one, it would be useful to show up the remaining number of actions inside of Agent's interface.
If there is no such limit, and the number of actions is limited by let's say amount of free memory then I could provide the whole macro for inspection.

kareltucek, 2021-11-30:
> I've created some long macros in past but never before did I need a macro consisting of 300 actions or more. 

Current limit is 255. (Original was ~64000, @mondalaci this was merged with fork merge, if you object, It can be reverted, but I am afraid that such revert would cost us 1024 bytes of RAM.)
 
>     1. Is the beforementioned drop in responsiveness influenced by the keyboard hardware or how powerful the host machine is?

Primarily by how efficiently the Agent is written. More powerful host machine will improve matters, but I wouldn't rate it as main cause. I would recommend to open ticket in Agent repository.

UHK's MCU performance has nothing to do with it.

> Next, I've found out that after a certain point (somewhere between 200 and 300 actions) when trying to run the macro, that "ERR" appears on the display. I've checked if I missed something in the macro multiple times and I am pretty sure I didn't make a mistake (using Karel's version of firmware)

You can use `$printStatus` macro to print a log of what went wrong. But in this case, you almost surely violated the 255 action count.

>     2. Is there a hardcoded limit of the number of actions that can be registered inside of a single macro?
>        If there is one, it would be useful to show up the remaining number of actions inside of Agent's interface.

Sounds as a reasonable request to me, but it is up to @mondalaci to decide.

>        If there is no such limit, and the number of actions is limited by let's say amount of free memory then I could provide the whole macro for inspection.

I am very interested in why you need so many actions within single macro anyway. 

If we are talking macro commands, then with next release, it will be possible to enter multiple commands within single action. This can help aleviate the problem, but in case you need so many actions in a single macro, then macro is probably wrong level of abstraction for your problem, and some other solution should be found.

mondalaci, 2021-12-01:
@bistabil I've just created a macro with about 200 macro actions, and Agent was quite fast. Please attach your configuration for us to reproduce the issue.

Why do you need so many macro actions in a macro?

bistabil, 2021-12-01:
I will be away from the keyboard until the end of the week but I will post the configuration asap.

Regarding the need for so many actions, well there is a use case in an MMORPG I am playing but it's of course very specific and I doubt anyone in the right mind ever thought to automate the whole quest involving 5 NPC's from start to finish with the use of keyboard macro.
Macro would be way longer if I didn't use some kind of "programming" using $goTo commands from Karel's fork to reuse some of the repeating fragments.
Can't wait to get feedback like "you were using UHK the wrong way all this time" after you see all that mess in the configuration 😆

bistabil, 2021-12-14:
Weirdly, I don't experience slow edits of long macros anymore.
Maybe I had some bloatware on my machine (I did clean windows installation a few days ago).
I am still attaching my configuration with the infamous 200-actions macro for your amusement but you can close this ticket.
Macro name: _TTSumSingleAccSwitch
[UserConfiguration.zip](https://github.com/UltimateHackingKeyboard/firmware/files/7710529/UserConfiguration.zip)

kareltucek, 2021-12-14:
> I am still attaching my configuration

Nice!

>  but you can close this ticket.

Allright!

closed 2021-12-14 by kareltucek

mondalaci, 2021-12-14:
@bistabil You were using the UHK the wrong way all this time.

Sorry, I couldn't help myself! ;)

Actually, I'm incredibly impressed by your macros. It truly showcases the insane power of Karel's firmware. It'd be an understatement to say that you're an advanced user.

If you ever find the time, please make a video of you using these macros. I don't understand them thoroughly, but it'd be amusing to see them in action.

For the time being, we don't plan to increase the number of macro actions per macro, but we can in the future if encountering some more use cases like yours.

Cheers!