UltimateHackingKeyboard/firmware #431: Cannot build on Windows

benedekkupper, 2021-12-05:
The symptom:
`../../scripts/Makedefs.mk:193: *** multiple target patterns.  Stop.`
The error line is this:
```
# Rebuild all objects when the Makefiles change.
$(OBJS): $(MAKEFILE_LIST)
```
This is how those variables expand on my setup (by inserting `$(info $(OBJS): $(MAKEFILE_LIST))` line above):
`build_make/E:/Workspaces/NXP-MCUXpresso/UHK-firmware/right/src/led_pwm.o  [... all other object files ...] Makefile ../Makefile ../../scripts/Makedefs.mk`

The absolute path on Windows has the drive letter followed by a colon, this colon is falsely interpreted by *make* as another target pattern, causing the error. I haven't found any easy solution to this problem, there is no way to escape the colon without breaking the path, so either *make* will throw this error, or it will give no rule to make target errors as it cannot find the source file on the modified path.

The solution would be to avoid using absolute paths, which is far from straightforward. There are [suggestions out there](https://make.mad-scientist.net/papers/multi-architecture-builds/), but I'm not that well-versed to figure this out on my own. The other solution would be to have a root Makefile, which only works with *make* arguments (e.g. right/left, version), instead of needing to navigate to the correct folder to run *make*.

I have tried to adapt a Makefile of my own which uses `vpath`, and gotten quite far, but couldn't get the linking to work, so I left that approach, and instead copied the right Makefile to the firmware's root directory, and changed to relative paths. It's building all the way now, which is good enough for me. I hope this issue will help others stuck in the same situation to get out of it faster than I did.

mondalaci, 2021-12-06:
@benedekkupper Thanks for sharing your journey and the solutions you found!

@kareltucek I'm not a fan of a root Makefile, and my Make Fu is not strong. How about yours?

kareltucek, 2021-12-08:
Probably not this strong.

Maybe a silly suggestion, but have you tried some build environment which uses unix-like paths? Afaik, things like wsl, mingw, cygwin, etc all use unix-like paths and mount drives like `/mnt/c/program\ files`.

benedekkupper, 2021-12-08:
Running the build inside the MCUXpresso IDE with the ARM Cross GCC toolchain as described in the readme, or running mingw32-make result in the same error.

mondalaci, 2021-12-08:
@Lauszus I dare to summon you as you're the author of the Makefiles. Can you possibly help us?

Lauszus, 2021-12-09:
Try replacing it with:

```make
# Make sure the dependencies are not included in the Makefile list.
CONTAINING = $(foreach v,$2,$(if $(findstring $1,$v),$v))
NOT_CONTAINING = $(foreach v,$2,$(if $(findstring $1,$v),,$v))
MAKEFILE_LIST_FILTERED = $(call CONTAINING,Make,$(MAKEFILE_LIST))

# Rebuild all objects when the Makefiles change.
$(OBJS): $(MAKEFILE_LIST_FILTERED)
```

This is what I use in another project, as there was some issues with `MAKEFILE_LIST`.

benedekkupper, 2021-12-10:
The problem isn't with the MAKEFILE_LIST, it's with the OBJS expanding to absolute paths, with the Windows drive letter's colon triggering the "multiple target patterns" error.

kareltucek, 2021-12-10:
@Lauszus your makefile hoards the builded obj files under `build_make` directory using absolute paths (like `firmware/right/uhk60v1/build_make/<absolute filesystem path>/firmware/right/src/something.o`). The trouble is that with windows-native builds these absolute paths contain semicolons (which make is inherently not compatible with) - e.g., `build_make/E:/firmware/right/src/something.o`.

Lauszus, 2021-12-11:
Try replacing:

```make
C_OBJS := $(addsuffix .o,$(addprefix $(BUILD_DIR)/,$(basename $(filter %.c,$(abspath $(SOURCE))))))
CPP_OBJS := $(addsuffix .o,$(addprefix $(BUILD_DIR)/,$(basename $(filter %.cpp,$(abspath $(SOURCE))))))
S_OBJS := $(addsuffix .o,$(addprefix $(BUILD_DIR)/,$(basename $(filter %.S,$(abspath $(SOURCE))))))
```

with:

```make
C_OBJS := $(addsuffix .o,$(addprefix $(BUILD_DIR)/,$(basename $(filter %.c,$(SOURCE)))))
CPP_OBJS := $(addsuffix .o,$(addprefix $(BUILD_DIR)/,$(basename $(filter %.cpp,$(SOURCE)))))
S_OBJS := $(addsuffix .o,$(addprefix $(BUILD_DIR)/,$(basename $(filter %.S,$(SOURCE)))))
```

However if I remember correctly it caused some issues with this project. That is why I added `abspath`.

benedekkupper, 2021-12-11:
This was one of the first things I tried myself as well, it's one of the solutions that fall into the above mentioned "it will give **no rule to make target** errors as it cannot find the source file on the modified path" category.

mondalaci, 2022-03-19:
I'm not sure if we have the time or expertise to address this issue properly, so I'm closing it. The suggested workaround should be very useful for affected parties, and I'm willing to allocate more resources if we get more requests.

closed 2022-03-19 by mondalaci