UltimateHackingKeyboard/firmware #435: Smart macros 1.0.0

mondalaci, 2021-12-26:
I've merged #421 because it was quite huge as it is. I'm unsure about the current smart macro version. We can call it 0.1.0, but in any case, I don't consider it to be 1.0.0. I propose the following changes to hit smart macros 1.0.0, which I copied from https://github.com/UltimateHackingKeyboard/firmware/pull/421#issuecomment-981174424

---

Instead of divisors, we could use percentage values like `scrollSpeedPercent`. It'd allow for proportional behavior like for other variables.

The mini trackball of the key cluster module emits merely 8 pulses over 360 degrees, so it's very low resolution. Consequently, I don't think we should provide a divisor value for it.

---

It'd probably be a better idea not to expose `caret`, `media`, and `zoom` navigation modes, but only the discreet mode. I'd name it `custom` mode and make `up`, `down`, `left`, and `right` directions configurable as arbitrary key actions.

I do think it has merit to configure the sensitivity of the custom mode on a per-module and per-layer basis. There are very different use cases, such as cycling through workspaces, windows, characters (caret), adjusting volume (media), and a hardcoded solution won't be satisfactory for all these cases.

---

As for the double (PC and Mac) zoom shortcuts, I wasn't aware of their implementation challenges, but I strongly favor them because they just work.

---

`axisLockStrengthFirstTick` is perfectly suited for the mini trackball of the key cluster module. Some users reported accidentally activating the mini trackball, so it'd definitely be useful.

However, I can't see this option being useful for any other module. I think it should be made specific to the mini trackball of the key cluster.

---

We shouldn't expose developer options like `cursorAxisLock` via smart macros along with other options. I really can't think of any practical uses of this option, and we shouldn't burden users with it. If users want to experiment with axis locking, we should enable them via navigation modes for which it's useful, such as `scrollAxisLock`.

---

How about ditching axis lock enabling and only exposing axis lock strength? The way I see it, it'd reduce redundancy and allow for the same configuration possibilities.

kareltucek, 2022-01-26:
> Instead of divisors, we could use percentage values like `scrollSpeedPercent`. It'd allow for proportional behavior like for other variables.

Actually I think you don't mind the behaviour of the values, but rather the "divisor" part of the name. Would it help to simply rename them to `scrollStep`, etc?

> I do think it has merit to configure the sensitivity of the custom mode on a per-module and per-layer basis. There are very different use cases, such as cycling through workspaces, windows, [...]

All those usecases require ability to controll the emmision with granularity up to a single event, which means that they need the same value. Also, having to always change value for all layers and keymaps is going for a really bad user experience. 

Summed up, I think this is a case of KISS and YAGNI - let's go with the minimal version for now and let's see if anyone minds. We can always expand if needed.

(Also, this could be solved with `$onLayerChange` smart macro event, which is in the end much more pragmatic solution in my opinion as it allows altering *any* value on per-layer basis.)

> It'd probably be a better idea not to expose `caret`, `media`, and `zoom` navigation modes, but only the discreet mode. I'd name it `custom` mode and make `up`, `down`, `left`, and `right` directions configurable as arbitrary key actions.
> 
> As for the double (PC and Mac) zoom shortcuts, I wasn't aware of their implementation challenges, but I strongly favor them because they just work.

This is just creating new problems.

As a user, I want to configure caret mode once and then simply use it across all layers and keymaps. I do not want to have to remap something like `<4 directions>*<12 layers>*<2 keymaps>*<3 modules> = 286` actions whenever I decide to slightly alter behaviour my caret modes.

Custom mappings or assignments have the potential to solve the PC and MAC zoom problem without using dirty tricks (which are likely to blow up in specific contexts), but the way you suggest implementing them, it just does not solve the problem.

The way I see it, there are two reasonable ways to go:

- minimalist - current state of affairs with fixed but sufficiently big number of caret mode slots which are smartMacro-remappable and smartMacro-assignable.

- maximalist - let's have a GUI which will allow to fully configure and even add custom caret mode slots. This is somewhat problematic without introducing some sort of dynamic memory allocation though.

As for the zoom:

- It would be nice if we simply could simply define two caret mode slots (`zoomPc` and `zoomMac`) and assign them somehow to the keymaps. (For instance, extending keymap definition by a PC/Mac flag would solve the problem (and enable a bunch of other usecases too).)

- I am thinking about defining a macro which would send the two shortcuts in sequence and then binding it in the zoom mode. There are some technical details which make it troublesome (such as defining the macro, or having to wait until till the macro finishes).

- But yeah, I guess I can hardcode that dirty proposal.

> The mini trackball of the key cluster module emits merely 8 pulses over 360 degrees, so it's very low resolution. Consequently, I don't think we should provide a divisor value for it.
>
> `axisLockStrengthFirstTick` is perfectly suited for the mini trackball of the key cluster module. Some users reported accidentally activating the mini trackball, so it'd definitely be useful.
> 
> However, I can't see this option being useful for any other module. I think it should be made specific to the mini trackball of the key cluster.
> 
> We shouldn't expose developer options like `cursorAxisLock` via smart macros along with other options. I really can't think of any practical uses of this option, and we shouldn't burden users with it. If users want to experiment with axis locking, we should enable them via navigation modes for which it's useful, such as `scrollAxisLock`.
> 
> How about ditching axis lock enabling and only exposing axis lock strength? The way I see it, it'd reduce redundancy and allow for the same configuration possibilities.

I find these invalid. All the values have their purpose.

As for the "let's selectively prohibit some of them in some contexts", it is strongly against the philosophy on which I have built the fork - which is providing small but generic building blocks. In case of the mouse subsystem, I have deliberately designed a system which can cover all those different requirements without having to use any special case branching, and then provided defaults which do cover those requirements. I believe that is the right approach.

If you insist, I am fine with throwing errors with `#ifdef` in those cases which you believe should be selectively prohibited, but I do not plan to remove them from the fork, and would very much prefer not to clutter the documentation and the grammar with tons of special-case comments explaining that "Value A does this and that, except when used with module X where A is selectively ignored and a hard-wired constant C is used instead".

If you prefer selectively throwing errors, what should the errors read?

mondalaci, 2022-01-29:
My main objection regarding divisors is their inversely proportional nature. I think the original names are more intuitive than the suggested step name, so let's leave them unchanged.

As for the per-module and per-layer custom mode sensitivity settings, I find your arguments valid, so let's keep them this way. $onLayerChange will probably be a useful addition, but let's not add it yet.

Your slot-based suggestion is the way to go regarding the custom navigation modes. We should use a fixed number of slots for the sake of simplicity, and 12 should be more than enough. We won't build a GUI in Agent yet, only later when eventually moving tried and true configuration options to the binary configuration. I prefer named slots, which would read intuitively, like `set module.trackball.navigationMode.fn workspaceSwitch`.

I'd like the suggested "dirty" zoom mode to be implemented. I understand it's not nice from an implementation perspective, but if it works as well as expected, it'd be very neat from a user standpoint.

I figured out a way to minimize the chance of users messing up their configurations with nonpractical settings, so there is no need to address this on the firmware level.

mondalaci, 2022-02-19:
I didn't mean slots with hardcoded names but arbitrarily specifiable names. I'm unsure regarding slot identifier naming restrictions, but I assume your fork may provide such conventions.

kareltucek, 2022-02-19:
I don't think that dynamic naming is worth all the trouble at the moment. As long as this feature lives in the macro engine domain, I would actually find it very unintuitive, since being able to write anything and have the engine allocate a slot on its own is not how things are done at the moment.

It starts making sense once/if you wish to make a GUI for it, in which case the names will be specified in the agent and saved in the binary config, and will not have to change at runtime.

mondalaci, 2022-02-19:
Fair enough, numbers will be fine. How about having 8 slots?

kareltucek, 2022-02-21:
Agreed.

mondalaci, 2022-02-21:
Do you think there's anything left? I'm getting lost.

kareltucek, 2022-02-21:
Those mentioned in this thread yes, but I remember you have not carried over all of the relevant points. Since the problematic parts are solved, I will go through the original thread again and write down the rest.

kareltucek, 2022-02-21:
> but caretAxisLock does make sense, so it should be added.

- [ ] make the discreet navigation mode compatible with non-axis-locked operation.

- [ ] figure out how to make distinction between "locking skew" and "secondary axis zeroing" (in docs).

(Trouble with current code is that it does not distinguish the signal-to-discreet-events transformation from axis locking. I.e., that discreet modes are always subject to zeroing the secondary axis. Indeed `caretAxisLock = false` makes sense, and semantically means turning off axis lock skews *and* allowing the secondary axis to accumulate over time. This also explains why disabling axis lock is not equivalent to setting the skews to 1.)

> scrollAxisLock is very useful, so it should be enabled by default.

- [ ] check this.

> Axis locking worked for me very reliably regardless of the value of axisLockStrength. How about you? This variable may not be needed and a reasonable default suffices.

This depends on granularity of the signal (which varies greatly with different modules) and user's ability to drag the cursor in smooth lines (which again varies depending on module and user). I am quite confident that it should stay. Still you are right that it is a bit nitche property and it makes sense not to acknowledge its existence in any simplified user guide.

----------------

Also one more detail which should significantly enhance usability of the locking mode.

- [ ] convert "axisLockStrength" back into "axisLockSkew". Furthermore, zero both axes when first event is produced. 

This will allow configuration with "axisLockFirstTickSkew = 2", to increase sensitivity for first event.

kareltucek, 2022-02-22:
> Do you think there's anything left? I'm getting lost.

(Now, with the above PR, I believe everything of importance has been addressed.)

closed 2022-02-26 by mondalaci