UltimateHackingKeyboard/firmware #437: Feature request: LED brightness control from macro action

R0unin, 2021-12-28:
Not everyone needs the LEDs on all the time, but pulling up 'UHK Agent' each time they do need it is cumbersome.
Would it be possible to get the LED brightness control added as a Macro action?
Ideally at least 3 actions would be nice to have.
1. Toggle LED off/on (off=0 and on=Last brightness)
2. Increase brightness
2. Decrease brightness

Optionally having a toggle in the "LED Brightness" section to turn on LEDs only when Modifier keys (mod, fn, and mouse) are held down would be nice for those that just want to highlight the extra programmed keys.

mondalaci, 2021-12-28:
I think we'll implement `toggleLeds`, `adjustLedBrightness <relativeValue>`, and `setLedBrightness <absoluteValue>` smart macro commands to cover the above use cases.

john-soda, 2022-04-06:
Is there an estimated time plan when this feature will arrive in the firmware?
 I personally miss this feature heavily.

mondalaci, 2022-04-06:
Probably 3-4 months.

kareltucek, 2022-05-03:
So, should this affect both the segment display, and the backlight LEDs?

Also, I am wondering whether we can come up with a syntax which would be more consistent with the rest of the features. For instance, following config variables seem to make more sense (also #426 is thrown in):

```
set leds.enabled <0/1>
set leds.brightness <0-255>
set leds.fadeTimeout <0-??? minutes>
```

Being able to adjust any numeric/boolean value also seems reasonable, but I am unsure of a syntax which would work well and still be comprehensible in docs and the grammar.

I am thinking about something like:

```
toggle leds.enabled
adjust leds.brightness 0-255
```

or 

```
set leds.enabled toggle
set leds.brightness {adjustBy | adjustTimes} 0-255
```

The second one should allow for a bit more readable grammar, but I am not entirelly satisfied with either of them... 

I am not strongly against the original `toggleLeds, adjustLedBrightness <relativeValue>, and setLedBrightness <absoluteValue> ` suggestion, but I see it as a strictly *temporary* solution which is to be removed when deprecated by another syntax.

mondalaci, 2022-05-04:
I love the general nature of the proposed `toggle` and `adjust` commands, so please implement them.

Regarding `adjust`, a negative number as its argument should be valid. I emphasize this because you mentioned the 0-255 interval which doesn't include negative numbers. Ideally, float increments should be supported, too.

`leds.enabled` should affect every LED, but I'm unsure regarding `leds.brightness` as we have 3 different brightness values. Any ideas?

Let's use an `uint16_t` for `leds.fadeTimeout` and `0` should mean disabled timeout. I'm wondering whether we should default to a reasonable timeout.

kareltucek, 2022-05-05:
Well, we can either expose all the three one by one (3x 0-255 value), or expose a new joint control value (0-1 float), which would multiply the other three values (and thus allow easy runtime control).

I slightly favor the latter.

kareltucek, 2022-05-05:
Computational overhead? I don't see any.

mondalaci, 2022-05-05:
When introducing a new joint control value, the LEDs of the three LED classes currently exposed by Agent would have to be multiplied by the global brightness value to have the final brightness values.

kareltucek, 2022-05-05:
Not really. They are already being multiplied by the per-key brightness value. All you need is to multiply this value before you begin updating the led map.

Bringing in a global brightness multiplier would bring in some code clutter, but no computational overhead.

mondalaci, 2022-05-08:
You're right, and I've just merged #472, which is very well done.

Will you implement `toggle` and `adjust` as the next step?

kareltucek, 2022-05-09:
Hopefully and eventually yes. In any case, I consider it a standalone issue.

I have some PoC implemented at https://github.com/kareltucek/firmware/tree/generic_adjust , but am not happy about it.

- I am not entirelly sure about proper form of the commands: 
  - for many values, it would make much more sense to provide multiplicative steps, so I am considering syntax like `adjust <variable name> { *FLOAT | /FLOAT  | NUMBER }`.
  - since we have many variables with 0 as special value, toggle should probably be implemented for non-booleans too. Either as `toggle <variable name>` toggling between 0 and some default value, or as `toggle <variable name> NUMBER NUMBER` toggling between the two (or N) provided values.

- I am very about the resulting grammar. So far, I consider the grammar as the main source of documentation, and therefore am dead set on keeping it brief and human-friendly. But I also want to keep it formally correct so that it can be used for a proper autocomplete in agent in the future. The above basically means that:
  - either the config sections of the grammar will have to be duplicated for each of the commands, leading to a grammar which is easy to read, but hard to navigate and maintain.
  - or the config variables will need to be gathered in their own nonterminal rule classes. This will make the grammar very hard to understand, and will loose us LL-0 parsability.
  - or the toggle / adjust can be easilly expressed by a context-sensitive rule. This is human friendly and keeps the grammar brief, but discards any hope for machine understability.

- Adjust will need to know metadata about every config value (such as valid range and maybe some others). I am wondering whether the configuration manager should be implemented first - maybe things will make more sense then.

- I am also wondering whether we should drop the idea of being able to reload config on demand and reset keyboard state on demand - and simply restart the firmware whenever state reset is required. There are many stateful features in the keyboard and maintaining selective state reset (without strange bugs) is going to be difficult. Simply reloading the firmware would make the firmware much more dependable, as correctness of such step is assured.

Any thoughts anyone?

mondalaci, 2022-05-10:
Please provide practical examples regarding `adjust` multiplication and division.

For easy maintenance, the config variables should be handled as first-class citizens, not bound to `set`. Why would it make the grammar hard to understand and parse?

By "configuration manager", I assume you mean the embedded smart macro documentation that will be rendered in Agent's sidebar. I'm still working on it, and I should be ready with it in a couple of days. It's a good idea to wait for it regarding this issue.

Regarding reloading the config on demand, do you mean resetting variables to their defaults when related smart macro commands are deleted, for example? If so, please explain an alternative way of resetting the affected variables, preferably as easily as possible.

kareltucek, 2022-05-10:
> Please provide practical examples regarding `adjust` multiplication and division.

```
ifShift adjust leds.brightness *1.1
ifNotShift adjust leds.brightness /1.1
```

> to understand

Obviously because abstracting the variables into their own classes means that commands which are now clearly expressed as ready-to-use commands will fall apart into multiple rules with multiple indirections, which any reader of the grammar will have to understand and follow.

> to parse

At the moment, the grammar roughly falls into LL0 class - which means that it can be parsed from left hand side, and moreover that the grammar rule can be always decided without having to use lookaheads when reading its first token. This property makes it very easy to parse (for instance with third party tools).

With the suggested command pattern, argument format depends both on variable type and on the command that is used. This means that without having parsed the variable name, you cannot know the format of the value, and therefore cannot decide which grammar rule to use by just reading the `adjust|toggle|set`.

(E.g., by just reading `set`, you cannot know which of the rule `set VAR_BOOL BOOL` and `set VAR_INTEGER INTEGER` applies.)

This means that either a lookahead is needed, or a workaround which duplicates the rules of the grammar for each command and each config variable.

For the C parser, this is not really an issue - the state can be carried over. The trouble really is that it makes the grammar much harder to read for the user, and it will make things a bit more cumbersome when we want to provide proper autocomplete in agent.

> By "configuration manager", I assume you mean the embedded smart macro documentation that will be rendered in Agent's sidebar. I'm still working on it, and I should be ready with it in a couple of days. It's a good idea to wait for it regarding this issue.

No. I mean #425.

> an alternative way

The proposed alternative is to restart the firmware, so that entire RAM is reinitialized into default state.

mondalaci, 2022-05-10:
As for the `adjust` multiplication/division example, why not increment/decrement the value?

To make the grammar easier to parse, we could make the commands non-overloaded, and use one command per signature, like `setBool` and `setInteger`, but that would be a usability disaster. Another solution could be to change the order of the arguments, like `adjust * 1.1 leds.brightness`. It'd be helpful to list every overloaded command signature to consider further solutions.

I wouldn't restart the firmware to reinitialize the RAM into the default state because it'd also involve reenumerating the UHK via USB. Agent could abstract the enumeration process and make it seem like it's not happening, but in practice, it takes several seconds during which the UHK is unresponsive, and some OS-es may display that a new USB device was connected. I'd much prefer reinitializing the values while the firmware is running. I know it's a pain to implement it, but it'd provide a great user experience.

mondalaci, 2022-05-16:
I'm using the latest master, and the LEDs go dim after about half an hour or an hour of inactivity. Can you reproduce it? If it's intentional, I'd rather leave the original behavior unchanged until the fade timeout setting becomes part of the main configuration, easily reconfigurable via Agent.

kareltucek, 2022-05-16:
Yes, I had set the default to 15 minutes. Will turn it off...

jalberto, 2022-09-22:
will this go into next release?

mondalaci, 2022-09-23:
The latest firmware contains smart macro commands that set absolute LED brightness, but a new Agent has to be released to use them. We plan to release a new Agent in weeks.

kareltucek, 2023-06-21:
Adjust -> https://github.com/UltimateHackingKeyboard/firmware/issues/620

Otherwise, led controls are available...

closed 2023-06-21 by kareltucek