UltimateHackingKeyboard/firmware #438: Semantics of semver spaces - request for comments

kareltucek, 2021-12-28:
Could you please provide info (i.e., ideally fill in comments somewhere in the codebase) about what those various different semver spaces mean, and when they are intended to be increased?

I.e., following package.json values:
```
  "firmwareVersion": "8.10.12",
  "deviceProtocolVersion": "4.7.1",
  "moduleProtocolVersion": "4.1.0",
  "userConfigVersion": "4.2.0",
  "hardwareConfigVersion": "1.0.0",
  "smartMacrosVersion": "0.1.0",
```

and furthermore `DataModelVersion` which is present in config parser, but does not seem to correspond to any of the `package.json` values.

In the (already merged) caret refactor, I have added read of `noFingers` to touchpad module. I have not bumped moduleProtocolVersion, which I believe is correct since touchpad does not have a dedicated UHK firmware.

Furthermore, I am a bit confused about `userConfigVersion` and `DataModelVersion`. I implemented new features (command action and more layers) by branching on `DataModelVersion`, and did not bump any of the values.

mondalaci, 2021-12-29:
See https://github.com/UltimateHackingKeyboard/firmware/blob/master/VERSIONING.md

I wrote this document quite a while ago, and it may be incomplete and inexplicit, so feel free to ask questions or suggest improvements. PRs are also welcome.

kareltucek, 2021-12-29:
Oh, nice. I never noticed it.

Still:

- is "userConfigVersion" same thing as "Data model version"?
- is "moduleProtocolVersion" same thing as "Slave protocol version"?

Assuming the answers are both yes, then how is major data model upgrade intended to proceed from version standpoint? I assume that Agent has to be able to read lower user config versions, and upon flashing new firmware, it is also supposed to automatically flash an upgraded user config.

- even if firmware is backward compatible across major user config version changes, its major version needs to be increased in order to signal compatibility with new format to the Agent.
- it makes sense to bump versions on releases, since even though current firmware is compatible with both 4.x and 5.0 user configs, its version upgrade needs to be coordinated with Agent releases.

kareltucek, 2021-12-29:
Oh, I see - data model version is version of the actually flashed user config, which isn't necessarily the same one as the one in package.json/userConfigVersion (which denotes firmware compatibility).

Still any comments on any misunderstandings in my conclusions?

mondalaci, 2022-01-03:
I've updated VERSIONING.md in the meantime. The answers are both yes, indeed.

Please see the related https://github.com/UltimateHackingKeyboard/agent/issues/562 and https://github.com/UltimateHackingKeyboard/agent/issues/1666 issues.

Any further questions, feel free to ask!

kareltucek, 2022-01-04:
I consider this solved.

closed 2022-01-04 by kareltucek