UltimateHackingKeyboard/firmware #440: MD5 checksum-based firmware updates

mondalaci, 2022-01-01:
For new firmware versions, Agent updates the firmware images of both halves and all modules, even though usually only the firmware image of the right half changes, which is problematic for two reasons:

1. The update process is unnecessarily long, especially due to BusPal, the USB to I2C proxy, which requires re-enumeration over USB.
2. Rarely, some customers brick their module or left keyboard half, which gets returned to us for hardware-flashing. Theoretically, the bootloader should always recover when flashing is incomplete, but practically, for some unknown reason, sometimes it doesn't.

We'll add MD5 checksums to every firmware image, including both halves and all modules. Agent will only flash firmware images whose MD5 checksums differ unless explicitly asked to flash them.

Given that the checksums are contained in the firmware images, they affect the checksums of the firmware images, so `make-release.js` will implement a two-step build process. In the first step, the 128-bit MD5 checksums will contain all zeros. In the second step, the MD5 checksums of firmware images of the first step will be used as the MD5 values contained in the final firmware images to be released.

We'll have to extend the device and module protocols to transfer the checksums and expose them to Agent.

Drizzt321, 2022-04-29:
As an aside, for hardware flashing, is this over serial with the pins exposed on the board if we were to take apart the cases? Probably a number of us have FTDI USB->Serial available and can use them, so maybe posting instructions somewhere would be useful to a (probably very small) sub-set of users who do accidentally end up bricking one or both halves.

mondalaci, 2022-04-30:
@Drizzt321 One needs a compatible SWD adapter, not an FTDI USB adapter. Related instructions are at https://github.com/UltimateHackingKeyboard/bootloader

This issue is not related to hardware flashing.

closed 2023-08-20 by mondalaci

AkechiShiro, 2023-08-20:
Thanks a lot @mondalaci !
Sorry for polluting this issue, but I would like to know, when this will be released in a new release ?

Also, is there any lacquer for the wood palm that is recommended to be very durable, mine is alll gone, now I'm sad, but don't want to go for a cheap spray for cars, lacque, what would be the most durable/best without taking into account the price ?

mondalaci, 2023-08-20:
@AkechiShiro You're welcome! This feature will be included in the next firmware release and will be supported by the next Agent version.

My colleagues may be able to recommend a lacquer type. Please contact them at https://ultimatehackingkeyboard.com/contact-us

PR #666 (2023-08-20): Implement checksum-based builds.
Closes #440