UltimateHackingKeyboard/firmware #442: Can the UHK emit key codes for extended Fn keys on macOS?

matthewsinclairdv, 2022-01-11:
I would like to be able to have the UHK emit key codes for extended Fn keys on macOS. At the moment, it does not appear to be possible to emit F14, F15, etc into macOS. I don't mind if these are on the MOD or FN layer (see attached images). As it stands, F13 works but it is the only key above F12 that does so.

[image]

[image]

All of these keyboards will properly emit extended Fn key codes:
- Apple Extended USB (white, numeric keypad and dedicated function, arrow, del, etc keys)
- Apple Extended Bluetooth (black, numeric keypad and dedicated function, arrow, del, etc keys)
- Apple External USB (just the alpha keys without the number pad)
- Apple External BT (just the alpha keys without the number pad)

I am pretty sure that I have the latest versions of keyboard firmware and desktop client app (see attached screenshots).

[image]

[image]

Extended Fn keys obviously work with Apple 1st-party keyboards, and I can see from products available on the internet that there are also 3rd party keyboards (https://www.imore.com/best-keyboard-mac) that feature extended Fn keys, so I can only assume that they work as well.

It would be great if a future firmware update allowed the UHK to emit key codes for all of the extended Fn keys in the keymap when connected to a Mac.

Thanks,
M@

mondalaci, 2022-01-11:
Please make a screenshot of the firmware page of Agent.

mondalaci, 2022-01-11:
Please try to reproduce the issue with [firmware 8.10.9](https://github.com/UltimateHackingKeyboard/firmware/releases/download/v8.10.9/uhk-firmware-8.10.9.tar.gz) and with [uhk-firmware-8.10.12-usb-fix.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/7725857/uhk-firmware-8.10.12-usb-fix.tar.gz).

matthewsinclairdv, 2022-01-11:
> Reply to: UltimateHackingKeyboard/firmware ***@***.***>
>
> EXTERNAL email from: ***@***.***

Ok, so I tried both of the above with mixed results.

The first one put the keyboard into a weird state where none of the keys worked. Luckily, the mouse still worked ok so I was able to try the second one. There were some errors in the firmware flash log (attached) but the keyboard worked again after the second flash.
Unfortunately, the F14 and F15 keys (or PrtSc, ScrLk, or Pause) did not work in either scenario.

BCG Digital Ventures
This email message may contain confidential and/or privileged information.
If you are not an addressee or otherwise authorized to receive this message,
you should not use, copy, disclose or take any action based on this email or
any information contained in the message. If you have received this material
in error, please advise the sender immediately by reply email and delete this
message.  
We may share your contact details with other BCG entities and our third party service providers. Please see BCG privacy policy https://www.bcg.com/about/privacy-policy.aspx for further information.
Thank you.

matthewsinclairdv, 2022-01-11:
Ok, so I tried both of the above with mixed results.
 
The first one put the keyboard into a weird state where none of the keys worked. Luckily, the mouse still worked ok so I was able to try the second one. There were some errors in the firmware flash log (attached) but the keyboard worked again after the second flash.

Unfortunately, the F14 and F15 keys (or PrtSc, ScrLk, or Pause) did not work in either scenario.

[20220111-2143_UHK_Firmware_Update.log.zip](https://github.com/UltimateHackingKeyboard/firmware/files/7850090/20220111-2143_UHK_Firmware_Update.log.zip)