UltimateHackingKeyboard/firmware #444: Firmware 9.0.0 todo

mondalaci, 2022-01-24:
I'm really looking forward to firmware 9.0.0, which will be a huge release including smart macros and 12 layers, thanks to @kareltucek.

There are two missing features that I'd like to include.

1. Shift layers should be handled by the firmware as Agent will expose them.
2. [Smart macros 1.0.0](https://github.com/UltimateHackingKeyboard/firmware/pull/421#issuecomment-981174424) support - Admittedly, this is less critical, but I'd really try to prevent breaking smart macros later as it can lead to a significant support burden.

kareltucek, 2022-01-26:
What exactly do you mean by `Shift layers should be handled by the firmware as Agent will expose them.`?

mondalaci, 2022-01-26:
Assuming the "1" key is mapped to "A" on the Shift layer, I'd expect Shift+1 to emit "A" which is not what happened when I tested common trunk.

kareltucek, 2022-01-26:
By "A" you mean "a", or "shift+a"?

kareltucek, 2022-01-26:
Also, what exactly happened?

mondalaci, 2022-01-26:
I meant "a". Sorry for being unclear. The firmware didn't seem to use the Shift layer and Shift+1 was emitted regardless of the mapping.

kareltucek, 2022-01-26:
So you are implying that when a modifier layer is enabled, corresponding modifier should be treated as layer switcher rather than a modifier?

I would expect that in order to access shift layer, you would have to map shift to the corresponding layer switcher. Otherwise, the behaviour is ill-defined for non-default layers.

kareltucek, 2022-01-26:
Also such behaviour immediately breaks mouse interaction when external mouse is used.

mondalaci, 2022-01-26:
> I would expect that in order to access shift layer, you would have to map shift to the corresponding layer switcher. Otherwise, the behaviour is ill-defined for non-default layers.

I don't see why.

Layer switcher actions with modifier layers cannot be specified in the pending "add more layers" Agent PR by purpose. Modifier layers shouldn't alter the current behavior when none action is mapped on the corresponding modifier layer. If a modifier layer action other than none action is specified, however, the action should be invoked without any further configuration.

I should also add that when it comes to modifier layers, by far the most usual feature request is to override Shift+number keys with alternative symbols.

> Also such behaviour immediately breaks mouse interaction when external mouse is used.

How's that?

kareltucek, 2022-01-26:
Well, could you please write down the specification of your vision for the modifier layer behaviour somewhere?

mondalaci, 2022-01-26:
I think I just did :)

While holding a modifier key, activate the relevant modifier layer. If additional modifiers are held afterward, they should compose. Then when pressing a key, look it up on the relevant modifier layer; If its action is a none action then fall back to the base layer, otherwise trigger the action.

kareltucek, 2022-01-26:
So should a modifier be added to the USB report when a modifier is pressed, or not?

mondalaci, 2022-01-27:
Except for the modifier key that activated the current modifier layer, modifiers should compose. Let's say Shift is held, and as a result, the Shift layer is activated. At this point, Shift shouldn't compose. But if afterward Alt is held, too, it should compose.

If left Shift is held and activates the Shift layer, then right Shift is held afterward, I think Shift shouldn't compose.

kareltucek, 2022-01-27:
So, if I have a non-UHK mouse in one hand and want to `shift+click`, or even just `shift + mouseover`, the shift just activates shift layer without sending shift to the OS. 

> If left Shift is held and activates the Shift layer, then right Shift is held afterward, I think Shift shouldn't compose.

`LShift + RShift` is often used to toggle system keyboard layout/language.

mondalaci, 2022-01-27:
I'm not proposing to suppress modifiers, and the mentioned use cases should work fine.

What I'm proposing is to emit the relevant modifier-layer action when it's other than none action. Depending on the action, it may be needed to artificially release the already active modifier. For example, to emit "a" instead of "A" on the Shift layer, Shift must be artificially released for such an action.

kareltucek, 2022-01-27:
So, by "Except for the modifier key that activated the current modifier layer, modifiers should compose." you meant "Yes, until a shift-layer-mapped key that does not contain modifier is activated"?

How should other modifier layers be treated? Especially left alt and right alt? (Right alt is usually used for entering special characters, while left alt is used for accessing application menus.)

mondalaci, 2022-01-27:
> So, by "Except for the modifier key that activated the current modifier layer, modifiers should compose." you meant "Yes, until a shift-layer-mapped key that does not contain modifier is activated"?

Correct.

> How should other modifier layers be treated? Especially left alt and right alt? (Right alt is usually used for entering special characters, while left alt is used for accessing application menus.)

Let's release those modifiers for modifier-layer actions that don't have such modifiers enabled, during the activation of the action.

kareltucek, 2022-01-27:
I meant, should alt layer be activated by left alt, or by right alt or by either of them? 

The relevant usecase: with czech layouts, right alt makes keyboard produce different characters, while left alt is used for triggering menu actions. This means, that as a user, I might want to use right-alt modifier layer to remap these special keys (just as some users will want to remap the number row), but in that case, I want only right-alt get affected, while I want left alt to continue acting as a simple composing modifier.

Furthermore, I will want `Right alt + shift + key` compose as `shift + alt-layer action`.

Am I correct in assuming that whenever more than one modifier is pressed at the same time, no modifier layer should be activated?

--------------------

What about non-scancode actions mapped in shift layer (e.g., mouse click)? Should modifier be applied to them?

Jopie01, 2022-01-27:
Does it make sense to release a 'in between' version of the firmware? Just to test the normal operation is not broken. Or are there too much changes which will break the agent and normal operation of the keyboard?

mondalaci, 2022-01-27:
@kareltucek Great point! Even though I use the US layout, we also have AltGr in Hungary.

I propose the introduction of the `modifierLayerTrigger.${layerName} = {both | left | right}` smart macro variable to address AltGr and similar use cases in a general way. We should default to `both` for every modifier layer.

The first modifier pressed could be used as the current modifier layer when multiple modifiers are held.

I think we should probably not compose modifiers with modifier-layer non-scancode actions, but I may not be able to see the relevant use cases, and this seems a quite niche use case which I don't feel strongly about.

@Jopie01 I don't feel comfortable releasing an intermediate version now. We'll take our time, test the new version to death, and release it when the time comes.

kareltucek, 2022-02-04:
I am not very sure how composition rules should work when multiple modifiers are pressed. Unless you object, I will implement the simplest possible version (= modifier layers are activated only if at most one modifier is active at a time).

mondalaci, 2022-02-04:
I'd stick to my suggestion, which would allow composition with modifier layer switching:

> The first modifier pressed could be used as the current modifier layer when multiple modifiers are held.

Then the modifiers after the first modifier should compose.

GrauLab, 2022-02-05:
Hi, if I understand this thread correctly, you are discussing about a feature similar to `QWY.base.LeftAlt := $holdLayer myAltLayer`, and wonder what should happen if the user then presses `QWY.myAltLayer.RightAlt`? If so, why not let us users define via Agent on `myAltLayer`? This could include both cases discussed above and would be more flexible:

1. Just hold a modifier for actions on the target layer:
For example, `QWY.myAltLayer.RightAlt := $holdKey RA-`. (Or `$holdKey LA-`, which I use to map right alt to left alt on my mouse layers for Alt-drag&drop actions.)

2. Hold a modifier but go back to the base layer:
For example, my current custom Alt layer for entering special characters provides a way to hold the original `AltGr` on the base layer by pressing both alt keys. This is implemented via: `QWY.myAltLayer.LeftAlt := {$pressKey RA-; $holdKeymapLayer QWY base; $releaseKey RA-;}`.

I think "undoing" the layer switch if followed by pressing any key that has `QWY.myAltLayer.(#key):=NONE` is a great idea for some contexts, but not for all:
- I use this logic regularly in "type mod contexts", i.e. a letter key acts as trigger for a mod layer here. Briefly, it is implemented as a look-ahead in the trigger macro using `$postponeKeys ifNotKeyDefined %0 unToggleLayer`.
- However, for my Alt layer of special characters above, I actually prefer to map NONE to "no action, but stay on QWY.myAltLayer.*", because I want to be able to correct myself (by pressing the intended/correct key next).

-----

Generally speaking, I would agree with @kareltucek to implement a simple solution, but would leave room for user customizability. 

Another general design thought: If the macro language already supports a particular feature, an alternative to a dedicated firmware level implementation might be to just provide an Agent/GUI level implementation that _writes basic macros for the user_: This could open a door for users who are new to this macro language. It would also help avoid maintaining double structures that essentially realize the same thing.
For single key remapping, you could use `$ifShift`. However, I admit that implementing shift layers via macros might look a bit complicated for macro novices. On the trigger key like LShift you would at least need something like: 
```
  $pressKey LS-
  $postponeKeys ifNotPending 1 goTo @0 //lookahead.
  $postponeKeys ifKeyDefined nameOfUserShiftLayer #key holdLayer nameOfUserShiftLayer
  $delayUntilRelease
  $releaseKey LS-
```
I wonder, what benefit do you see in holding a mod like shift for a whole target layer? Why not just let the user record `LS-` in Agent on those target layer keys where he needs/wants it? 
(I think the latter can be more flexible, solve questions of mod composition, and avoid a `$releaseKey LS-` handling if the user wants to play a non-shift character on his shift layer like `=`.)

----

PS: Integration of macros in the main firmware sounds great; thank you! :) I am really looking forward to compress my many single-line write-`$text` actions into multiline macros. 
[...especially since my 32kB config space is running full *again*, although I have already deleted all my old macros and performed several rounds of notes cleaning. When UHK macros become more and more popular, I guess we will need to start zipping macros or at least their inactive `//comments and notes` parts in the future! ;D ]

mondalaci, 2022-02-05:
@GrauLab I'm not sufficiently familiar with the macro features of Karel's fork, but I want to provide a simple way of exposing this feature via a smart macro variable. I'm not against more advanced macro commands, which are planned to be included in the official firmware later.

Agent will be decoupled from smart macros, so Agent will not generate them. This is by design, so Karel's fork and the official firmware can implement their own set of smart macro features without affecting Agent. We're in the process of implementing a [collapsible sidebar](https://github.com/UltimateHackingKeyboard/agent/issues/1670) in Agent, however, which will allow for fancy GUI-based generator features. The firmware will contain the application logic of the sidebar's content.

Popular features, generally smart macro variables, will be eventually moved to the binary configuration, and we'll add dedicated GUIs for them simultaneously. I don't want users to deal with smart macros unless they want a very advanced configuration.

kareltucek, 2022-02-05:
@GrauLab I don't understand the first part of your comment at all (i.e., until the first horizontal line - what are actual proposals for behaviour changes?, what effects are proposed to be applied under what circumstances?).

As for the second part, corresponding macro would have to be much more complex than the one suggested, if possible at all.

> you are discussing about a feature similar to `QWY.base.LeftAlt := $holdLayer myAltLayer`

Not really. I will try to reformulate.

User point of view:

- User wants to rebind the numeric key row. Specifically in such way that shift and non-shift meanings can be remapped independently.

- In the Agent, this will be done by simply remapping the numeric row of the shift layer.

- Modifier mappings or switcher mappings are not to be touched at all during this process. Shifts will still be mapped to simple scancodes (`LS-`, etc). 

- (Shifts should still work as shifts. Shift + click should still produce shift + click.)

Developer point of view:

- (Currently, when a key is activated, the action in question is first copied into a key action cache and then activated from there. This ensures that whenever a key is activated, it will retain its original meaning for as long as it is held, even if layer or keymap changes in the meanwhile.)

- The way I understand it, the new behaviour should be such that when a key is activated, the cache is populated depending on current state of modifiers. Namely:

  - Assume that base layer is active, definition of a shift layer exists and some shift key (i.e., a key mapped to a simple scancode `LS-`) is (already) active. Furthermore assume that a key A gets pressed, and that A is defined on shift layer. 
    - Then the cache record of key A should be populated from the shift layer, and the shift key's modifier effect should be canceled until either another key is pressed or A is released. 
    - Otherwise, A's cache should be populated from base layer. (No cancelling effects should be applied.)
 
  - When more than one modifier is pressed, only the first one should be treated in the special way as stated above (according to Laszlo's request). All others should be applied and no cancelling effects should apply to them. (This means that a complex and somewhat-unstable state will have to be maintained and updated. Contrary to the simple proposal, which can be resolved in a mostly stateless manner.)

- Sticky modifiers should still apply according to their own logic and should probably not be affected by any canceling effects except their own.

- (Note that behaviour of the shift layers is somewhat similar to sticky modifiers, although not exactly.)

- Regarding generallity, I will see what I can do when I get to implementing it.

GrauLab, 2022-02-06:
Thanks for your replies and the explanations. I understand the design decision @mondalaci/"don't want users to deal with smart macros unless they want a very advanced configuration".
@kareltucek: I have thought again about how to define the intended feature of mod layers in an unambiguous and generic way that is minimally invasive for the base layer. Maybe it helps a little.

----
First, one question on "...and the shift key's modifier effect should be cancelled...":
- If I understand you correctly, the user can and must explicitly record `LS-` in Agent for each key on the shiftLayer where he wants a shift-scancode, and can simply omit `LS-` in Agent to map non-shift scancodes on his shiftLayer. I think this is a good/necessary decision. 
- Why do you not want to apply the same logic to non-scancode actions, too? I.e. let the user also record `LS-click` in Agent if he wants `shift` to "passthrough" to the key defined for clicking. This would also allow non-shift click actions on the shift-layer. More importantly, no lookahead or classification logic for key actions would be needed when adopting this rule.
---
I think the following solution might be minimally invasive to the base layer. Maybe it is the same you already mean with "the cache is populated depending on current state of modifiers". I am not sure, because I think in your macro language from a user's perspective. Assume you had this pseudocode on each potential target key:
```
$ifShift ifFirstModWas shift ifKeyDefined shiftLayer #key  final suppressShift replayThisKeyOn shiftLayer
//same one-liner for $ifAlt, $ifCtrl, $ifGui.
//here starts the baseLayer.#key's default action like holdKey x.
```
- `replayThisKeyOn shiftLayer` means to (a) inject `holdLayer shiftLayer`, and (b) replay the hardware `#key` press there. 
  - I.e. restore/prepend the `#key`'s press event at/to the front of the queue, and let it call the action defined on `shiftLayer.#key` this time, instead of triggering above pseudocode from the base layer again.
- `suppressShift` essentially means sending a `releaseKey LS-` when needed, but I see two special cases:
  - To avoid sending superfluous `rLS-/pLS-` events to the OS (maybe harmless for shift but important for alt because of menu accelerators in Windows), a "passthrough logic" is needed: If `shiftLayer.#key` is assigned an action that has the `LS-` mod defined in Agent, just do nothing, i.e. do not inject `releaseKey LS-` in this case.
  - Importantly, if the key that originally sent `pressKey LS-` is still pressed after the target key has been released, it must be restored if a `releaseKey LS-` was injected (to let the user continue shifting when back on the base layer).
    - In pseudocode and directly after the injected `holdLayer shiftLayer` above, we need a line like `ifKeyActive #shift ifNotMod LS- pressKey LS-`. Here, `#shift` points to the key ID that caused `ifShift` to become true in the first line of the baseLayer.#key macro above. Is this key ID already stored somewhere?
    - Internally, you might require at least two "mod info structs" for this: one from the macro trigger time, and one for "now", i.e. the execution time of the current line. (I.e. I assume `ifShift` means `ifMod LS- atMacroStart`, but then `ifMod LS- now` requires something different, I guess?)
---
I see one remaining definition problem with mod chords, like simultaneously pressing `alt+shift`:
- For such chords, `ifFirstModWas shift` might have an undefined meaning in the user's head (who thinks that he has pressed mod keys simultaneously).
- For an `alt->shift` chord and in case of no defined action for the target key on altLayer, should we still invoke shiftLayer if it contains a definition, even though `alt` arrived a few milliseconds earlier? I think yes. 
- If both shiftLayer and altLayer define an alternate action for the same target key, the mod chord ambiguity is the user's fault, I'd say. (I would just invoke the one that came physically first here, even if only by a few ms. We already have macros to branch a chord of keys to another layer, anyway...)

Have a nice Sunday!

kareltucek, 2022-02-06:
> * If I understand you correctly, the user can and must explicitly record `LS-` in Agent for each key on the shiftLayer where he wants a shift-scancode, and can simply omit `LS-` in Agent to map non-shift scancodes on his shiftLayer. I think this is a good/necessary decision.

Maybe correct, maybe not. 

The user is supposed to remap only those scancodes which should behave "oddly". 

For instance, if the user wishes to remap the number/symbol row, he is supposed to map only those keys on shift layer. All other keys should remain mapped to `none` on shift layer.

> * Why do you not want to apply the same logic to non-scancode actions, too? I.e. let the user also record `LS-click`...

1) Because neither the serialized config format nor the in-memory representation can express `LS-click`. 

2) Because whenever I used the connection `shift + click`, there is 50% chance that I meant click with an external mouse (which UHK knows nothing about).

-------

> ... Maybe it is the same you already mean with "the cache is populated depending on current state of modifiers". ... 
> 
> ```
> $ifShift ifFirstModWas shift ifKeyDefined shiftLayer #key  final suppressShift replayThisKeyOn shiftLayer
> //same one-liner for $ifAlt, $ifCtrl, $ifGui.
> //here starts the baseLayer.#key's default action like holdKey x.
> ```

This seems correct. 

> * `replayThisKeyOn shiftLayer` means to (a) inject `holdLayer shiftLayer`, and (b) replay the hardware `#key` press there.

To be honest, I am not sure whether `(a)` should take place. My attempt at a specification above assumes that it should not.

Result will differ in sticky modifier behaviour and in whether or not `press shift; press mod` will switch into mod layer or not.

>  `#shift` points to the key ID that caused `ifShift` to become true in the first line of the baseLayer.#key macro above. Is this key ID already stored somewhere?

No.

> (I.e. I assume `ifShift` means `ifMod LS- atMacroStart`, but then `ifMod LS- now` requires something different, I guess?)

It means at the time when the condition is first evaluated within that evaluation instance. E.g.:

- `ifShift delayUntil 500; goTo @-1;` it will be evaluated again every time before `delayUntil` (may) starts executing, and may yield different result each time.

- `ifShift ifSecondary holdKey a`, `ifSecondary ifShift holdKey a` and `ifShift ifSecondary ifShift holdKey a` have all different meanings since ifSecondary will insert a delay. 

> * Importantly, if the key that originally sent `pressKey LS-` is still pressed after the target key has been released, it must be restored if a `releaseKey LS-` was injected (to let the user continue shifting when back on the base layer).

Correct. The solution will be more along the lines of having multiple modifier "classes" (differing by whether or not they are considered by ifModifier conditions and whether or not they are affected by modifier suppression masks) and composition rules for them.

-------

> * For such chords, `ifFirstModWas shift` might have an undefined meaning in the user's head (who thinks that he has pressed mod keys simultaneously).

Indeed. 

> * For an `alt->shift` chord and in case of no defined action for the target key on altLayer, should we still invoke shiftLayer if it contains a definition, even though `alt` arrived a few milliseconds earlier? I think yes.

I think no. `alt + shift + baseLayerKey` should be produced.

> * If both shiftLayer and altLayer define an alternate action for the same target key, the mod chord ambiguity is the user's fault, I'd say. (I would just invoke the one that came physically first [...]

Yes.

GrauLab, 2022-02-07:
>`alt + shift + baseLayerKey` should be produced.

Right/I agree.

>Because neither the serialized config format nor the in-memory representation can express LS-click.

- I see, but the price for this default behavior for all mouse operations will be that the user is unable to define their non-shift versions on the shiftLayer, e.g. scrolling without holding LS-. (This is okay with me; I am just observing. We have the mouse layer, and it has dedicated modifier keys like shift/alt/ctrl, anyway.)
- The next related question is how to treat keys assigned to macros on shiftLayer. 
  - If a `releaseKey LS-` in the macro suffices(?) to manually "close" the `pressKey LS-` scope for subsequent macro lines (despite the #shift key is still held! not sure how sticky it is...), then this is uncritical. 
  - But the restore logic should still work after such a user macro was called on shiftLayer. I understand from your reply that `ifShift` already refers to the modifier state "now" (the postponer queue should be empty, I guess). Then, once `holdLayer shiftLayer` returns, the line `ifKeyActive #shift ifNotShift pressKey LS-` should suffice, _whatever_ was played on shiftLayer before.

>Because whenever I used the connection shift + click, there is 50% chance that I meant click with an external mouse 

I don't understand: `UHK.shift + extMouse.*` should work just fine as before, because `holdLayer shiftLayer` would only be executed after receiving a press event from a target key that has `shiftLayer.key~=none`. What sequence do you mean?

>Result will differ in sticky modifier behaviour ...

(Don't know enough about sticky implementation to comment.)

> ...and in whether or not `press shift; press mod` will switch into mod layer or not.

As the `holdLayer` is only injected _after_ a `press(targetKey)`, I guess you mean a sequence like `shift -> targetKey1 -> mod -> targetKey2` (where the target keys are defined on shiftLayer and modLayer, respectively). This would indeed lead to different results when using `holdLayer shiftLayer`, but only if `releaseTime(targetKey1)>pressTime(targetKey2)`. And in this scenario, it might be user-intended branching. (Or is `mod` a target key that is defined on your shiftLayer?)

>To be honest, I am not sure whether (a) [`holdLayer shiftLayer`] should take place. 

- This depends on whether only the narrow specification for 2-key `mod->modLayer.targetKey` "escape actions" is the goal with this feature, or things involving >=3 keys, too. Like `alt -> altLayer.W:=upArrow -> altLayer.A:=leftArrow`, where `W/up` is still pressed when `A/left` follows...
- To enable more general/continuous mod layer behavior like WASD->arrowing or WASD->mousing, modLayer needs to be held as long as _any_ of its defined action keys is pressed (transitive). 

From a specification point of view, I see this just like an usual independent layer with a special trigger logic that 

  1.  avoids a lookahead that would postpone every default `pressKey LS-` on `#shift` (and every `pressKey LA-` on `#alt` )
  2. suppresses (i.e. temporarily releases) the modifier, if so defined by the user on modLayer.targetKey (attention here: `pLA->rLA=program menus` and `pGui->rGui=start menu` in Windows => needs a hack like inserting `tapKey(some never used scancode)` to cancel after the press event has already been sent to the OS!)
  3. after holdLayer returns (held from first to last target key), restore the press(mod) state if the original `#mod` key is still pressed.

But it all comes down to whether these mod layers should just provide a few "escape roles for some target keys on the number row", or should become additional general-purpose layers for the user to freely configure. 

And, of course, how much existing code/logic you can reuse from macro commands to realize them. I admit that I am a bit biased here: As I would find a command like `replayThisKeyOn myVanillaLayer` also very helpful/elegant/mighty on macro level (see our old discussion at https://github.com/kareltucek/firmware/issues/55), I would vote for "realize as a macro first, and then reuse that code". ;)

kareltucek, 2022-02-08:
@GrauLab sorry, but I am lost.

mondalaci, 2022-02-08:
I'm lost beyond measure! :)

I trust the judgment of @kareltucek regarding the implementation of this feature. I propose not to get lost in details now but rather create a working version and iterate afterward if needed.

GrauLab, 2022-02-08:
Sorry; I didn't mean to confuse. Sure, implement as you see fit, and iterate from there if needed! :)

For comparison, and hopefully a bit clearer, here is one possible and hopefully logically complete implementation in commented pseudocode, as I see it for your specification.

1. Check if we should branch to a modLayer:
    <details>
          <summary>Code</summary>
    
    ```
    QWR.baseLayer.fcnExecuteKeyAction := {
      foreach #modScancode in {LS,RS, LA,RA, LC,RC, LG,RG}
        postponeKeys ifModIsPressed #modScancode  ifFirstModWas #modScancode  ifKeyDefined getLayerFor(#modScancode).#key  final replayThisKeyOn getLayerFor(#modScancode)
      exec baseLayer.#key.agentAction //the current firmware just directly executes this line.
    }

    //To avoid triggering mod layers on chording (like simultaneous alt+shift), consider a timing condition in addition to ifFirstModWas #modScancode.

    //replayThisKeyOn is defined as { a) restore the hardware key press that has triggered this macro (prepend it at the front of the queue).  b) final holdLayer getLayerFor(#modScancode) }
    ```
    </details>

2. On entering a mod layer, prepare the restore logic needed below by storing a copy of the current modScancode mask, and of the hardware key IDs that caused it:
    <details>
      <summary>Code</summary>

      ```
    foreach #layer in {shiftLayer, altLayer, ctrlLayer, guiLayer}
        QWR.#layer.onBeforeEnter := {
          foreach #modScancode in {LS,RS, LA,RA, LC,RC, LG,RG} {
            set #layer.initialModMask.#modScancode := (ifModIsPressed #modScancode)
            set #layer.keyIDofOriginFor.#modScancode := the hardware key that caused (ifModIsPressed #modScancode), or -1 if this mod is not pressed.  //the firmware needs to keep track of these key IDs of origin whenever the mod mask is updated.
          }
      }
      ```
    </details>

3. On a modLayer and if required by the user's definition of modLayer.#key.agentAction, just-in-time release #modScancode (whose pressKey/keyDown event has already been sent to the OS!):
    <details>
      <summary>Code</summary>

      ```
      QWR.shiftLayer.fcnExecuteKeyAction := {
        postponeKeys if shiftLayer.initialModMask.LS  ifModIsPressed LS  ifNot shiftLayer.#key.agentAction.definedModMask.LS  releaseKey LS-
        postponeKeys if shiftLayer.initialModMask.RS  ifModIsPressed RS  ifNot shiftLayer.#key.agentAction.definedModMask.RS  releaseKey RS-
        exec shiftLayer.#key.agentAction
      }
      QWR.ctrlLayer.fcnExecuteKeyAction := ... //analogous to shiftLayer.
      QWR.altLayer.fcnExecuteKeyAction := {
        //Problem: tapKey LA- causes opening the menu in Windows. To effectively cancel pressKey LA- in hindsight, this hack inserts a tapKey of a "never" used scancode before releaseKey LA-:
          postponeKeys if altLayer.initialModMask.LA  ifModIsPressed LA  ifNot altLayer.#key.agentAction.definedModMask.LA  tapKey f20  //if possible, replace f20 by a custom new scancode that is really never used anywhere.
        //here, continue analogous to the shiftLayer.
      }
      QWR.guiLayer.fcnExecuteKeyAction := ... //analogous to altLayer, since tapKey LWin causes opening the Windows Start menu, but Win+F20 does nothing.
      ```
    </details>

4. On leaving the mod layer, restore the possibly cancelled/released #modScancode, provided that its originally sending hardware key (like LShift) is still physically pressed: (Necessary in order to continue, for example, normal shifting on the baseLayer.)
    <details>
      <summary>Code</summary>

      ```
      foreach #layer in {shiftLayer, altLayer, ctrlLayer, guiLayer}
        QWR.#layer.onAfterLeave := {
          foreach #modScancode in {LS,RS, LA,RA, LC,RC, LG,RG}
            if #layer.initialModMask.#modScancode  itNotModIsPressed #modScancode  ifKeyActive #layer.keyIDofOriginFor.#modScancode  pressKey #modScancode 
        };
      ```
    </details>

kareltucek, 2022-02-09:
@GrauLab I am still quite unclear on what is the actual purpose of this exercise.

- If the purpose is to discuss what the feature should do (or not do), I would find discussing "user stories" much more useful.

- If the purpose is to discuss or propose actual implementation solution, I would love to see a PR of actual PoC implementation. The above pseudocode is nice, but has little to do with the actual reality of internal workings of UHK firmware.

GrauLab, 2022-02-09:
At first, I (wrongly) thought: "This can already be done by macros, why a double structure?" Then this shifted to: "What is the minimal necessary extension of the macro grammar to realize this feature in a logically complete way?" The purpose/idea is to make your macro language even more general, and mighty enough in the future to realize such (and many additional) advanced features. I don't think that much is missing: Essentially only special macro names like `keymap.layer.onEnter` for event-triggered macros, and `replayThisKeyOn`. Most of the remaining pseudocode above is just syntactical sugar that could be replaced with already existing commands like `ifShift/setReg/ifRegEq...`.
Regarding PRs: I have looked at the firmware code last year, and compiled it with a small change to a macro command. But I last coded in C during my PhD, and to become fluent and make useful PRs on the firmware level, I would need more time than I realistically have in the foreseeable future, unfortunately (even though it would be fun).

kareltucek, 2022-02-10:
I see.

As to the macro support, the way I see it, the *usecase* itself is already supported by `ifShift suppressMods tapKey ...; ifNotShift tapKey ...`. Indeed when thinking about it more, it has some problems, but no one ever complained about those, so I still consider it to be a relevant solution for the problem.

I don't think we need to be able to implement this ticket on macro level, as this feature itself is not about broadening the capabilities of the keyboard. The way I see it, the modifier layers as Laszlo envisions them are supposed to simply provide a simple and user-friendly idiomatic way of remapping the modifier keys. *Without having to touch macros at all.*

As to the macro first approach, it simply is not the reality at the moment. The "native" and "macro" subsystems are two distinct and strongly isolated "ecosystems". 
- The native actions have no state, occupy very little space in the config format, can be executed very cheaply, and any number of them can be active at a time.
- Macro system on the other hand is quite "heavy". Macros are stateful. Each instance occupies a nontrivial large state, allocated to it from a relatively small state pool (16 at the moment). Execution very is inefficient, since the macro language is not compiled in any way at the moment.

As to the possibility of making UHK a macro-first firmware, it is possible, but would require someone to take the firmare and redesign it mostly from scratch. Definitely doable, but requires someone who has time and energy to invest into it. At the moment, I do not intend to become that person.

GrauLab, 2022-02-11:
@kareltucek Thanks for the explanation regarding "native vs macro". I can fully understand your decision. Have a nice weekend.

closed 2022-03-19 by mondalaci