UltimateHackingKeyboard/firmware #445: Problem with Write Text Macro (outputting different chars)

ggbecker, 2022-01-31:
Agent version: 1.5.17
Right keyboard half firmware: 8.10.12
Left keyboard half firmware: 8.10.12
Fedora release 34 (Thirty Four)

Steps to reproduce: 

- Create a new Write Text Macro and type `./build_product`

- Save and load the macro into the keyboard

- Activate the macro. It generates some weird results. Note that it changes the `_` with `-` and change letters to uppercase sometimes.

`./BUild-product./buILd-product./bUIld-product./build_product./build_product./buILd-product./buILd-product./buIld-product./buIld-product./buILd-product./buILd-product./buiLd-product./buiLd-product./buILd-product./buILd-product./buILd-product./bUIld-product./bUIld-product./build_product./bUIld-product./bUIld-product./buILd-product./Build-product./build_product./build_product`

kareltucek, 2022-01-31:
First please try to reproduce the problem with different programs. Chances are that your program just can't keep up with the fast input. (I have this exact problem with my terminal, however, input in browser works just fine.)

Assuming that slowing down the input solves the problem, you can use https://github.com/kareltucek/firmware firmware and use (e.g.) `set keystrokeDelay 10` option, which will make every usb report last at least 10 ms before moving on. (Also, you probably wish to put it into a macro named `$onInit`.)

ggbecker, 2022-01-31:
> First please try to reproduce the problem with different programs. Chances are that your program just can't keep up with the fast input. (I have this exact problem with my terminal, however, input [...]

That's true, activating the macro on a web browser works almost every time, but it's still not perfect. I understand that this is probably a problem with the program that is not able to keep with the fast input. While this is probably solved by your fork, is there any progress on making the `keystrokeDelay` option available in the official firmware?

I would love to use this feature with the terminal, especially any kind of them.

kareltucek, 2022-01-31:
> is there any progress on making the keystrokeDelay option available in the official firmware?

Yes, the code is actually already merged into uhk/master. As for the release, there are some things still to be done (see https://github.com/UltimateHackingKeyboard/firmware/issues/444).

mondalaci, 2022-04-16:
`set keystrokeDelay` is merged into the master and will be contained in the next release, so I'm closing this issue.

closed 2022-04-16 by mondalaci

ggbecker, 2022-11-21:
I can confirm that setting a macro called `$onInit` and defining a command `set keystrokeDelay 10` fixes the issue with the bogus input. As described in the docs: https://github.com/UltimateHackingKeyboard/firmware/blob/0d584b17dc19e50bfd92185a1bf28fa62ba891ba/doc/dist/index.html