UltimateHackingKeyboard/firmware #452: Make Mod+D work like Alt+Tab for UHK trackball

mrpond, 2022-02-23:
1. Install the latest Agent: 
Agent version: 1.5.17
2. Use Agent to update to the latest firmware:
 Right keyboard half firmware: 8.10.12
 Left keyboard half firmware: 8.10.12
 Key cluster firmware: 8.10.12
 Trackball firmware: 8.10.123. 

Current behavior:
After triggering Windows switcher (Mod+D) and keep holding Mod button,
UHK's Trackball not working
 
Alt-Tab and holding Alt with UHK trackball -> cursor move
**Mod-D and holding Mod with UHK trackball -> cursor not move**
Mod-D and holding Mod with non-UHK trackball -> cursor move

Expected behavior:
After triggering Windows switcher (Mod+D) and keep holding Mod button,
UHK's Trackball should working like
triggering Windows switcher (Alt+Tab) and keep holding Alt button.

mondalaci, 2022-02-23:
@mrpond Do you use the scroll feature of your trackball module via the Mod layer?

mrpond, 2022-02-23:
> @mrpond Do you use the scroll feature of your trackball module via the Mod layer?

nope, I use left module for this task.
and sometimes I use Mod + I,J,K,L for cursor while coding.
[image]

mondalaci, 2022-02-23:
Then you will be able to use the `set module.trackball.navigationMode.mod cursor` [smart macro](https://github.com/UltimateHackingKeyboard/firmware/issues/351) in the next Agent release.

In the meantime, [Karel's UHK firmware fork](https://github.com/kareltucek/firmware) probably already implements this feature.

kareltucek, 2022-02-23:
@mrpond Laszlo's is trying to say that the feature which you ran into is called "navigation modes". Navigation modes are assigned to layers, and default navigation mode for mod layer is scroll mode, which means, that your "hold Mod and move using UHK trackball" produces scroll events instead.

Unfortunatelly, there is no official documentation for this feature (@mondalaci which you should fix by mentioning them in tutorials, and by adding a short entry into knowledgebase). You can find some mention in https://github.com/kareltucek/firmware/blob/master/MACRO_DOCS.md, but that documentation already assumes knowledge of basic UHK features since it focuses on documenting the extended macro engine rather than on documenting the UHK as whole.

kareltucek, 2022-02-23:
Taking back, it *is* mentioned in tutorials, although having a separate entry explaining just navigation modes would still be nice to have.

mondalaci, 2022-02-23:
I'll improve the documentation as soon as we release smart macros.

mrpond, 2022-02-23:
Here how I manage to solve it,
correct me if I'm wrong.
1. Download firmware https://github.com/kareltucek/firmware
2. Flash them use official UHK agent
3. Create new Macros on agent
[image]

4. Map it to the Key on your choice ( I use Fn + Q)
[image]

and okay now. Mod+D then hold Mod. UHK's trackball work as I want it to be.

I'll wait for new version of firmware + agent.

mrpond, 2022-02-23:
@mondalaci @kareltucek but I ran into other issue,
with custom firmware and set module.trackball.navigationMode.mod cursor macros in place

While scrolling with left module I can't move UHK's trackball cursor.

kareltucek, 2022-02-23:
You might also wish to rename the macro `$onInit` - it will then be automatically executed on UHK startup.

mrpond, 2022-02-23:
> 

nice trick!