UltimateHackingKeyboard/firmware #453: Shortcut triggering bug

mondalaci, 2022-02-23:
I'm running stock master, and I can only invoke Alt+F4 if I hold Mod first. If I hold Alt first, the shortcut is not invoked.

Attached is my configuration.
[UserConfiguration.json.zip](https://github.com/UltimateHackingKeyboard/firmware/files/8128265/UserConfiguration.json.zip)

kareltucek, 2022-02-24:
I can't import that config, but let me guess - you have Alt layer active?

In that case alt layer (not mod layer) is active, and by definition that is correct. 

Suggested solution: let non-modifier layers override modifier layers.

mondalaci, 2022-02-24:
Yes, your assumption is correct.

Thanks for the fabulously quick fix! Works like a charm.

closed 2022-02-24 by mondalaci