UltimateHackingKeyboard/firmware #456: Snap trackball arrow key emulation to cardinal directions?

DarwinAwardWinner, 2022-03-11:
In theory the Fn + trackball to emulate arrow keys could be a really useful feature, but in practice I find it very hard to use effectively, because I have to scroll the trackball at *exactly* the right angle in order to move left and right across a line without accidentally moving up or down as well.

Basically, I would like to propose that if the direction of trackball scrolling is within a certain angle of horizontal or vertical, it should "snap" to exactly horizontal/vertical. This is analogous to how analog thumbstick inputs are interpreted in video games. In the screenshot below, the idea is that any input within the bright yellow angled regions is snapped to the corresponding horizontal/vertical direction.

[image]

Based on my testing, I'm not sure whether the firmware already implements something like this for the arrow key emulation, but if it does so, the effect is obviously not enough to prevent my inputs from straying away from the cardinal directions, so if this is the case, I'd like to request that the strength/angle of this snapping be made adjustable. (Honestly, I think my ideal angle would be 45 degrees, i.e. always snap to nearest cardinal.)

Obviously, it might also be nice to have this same option for both scrolling and mouse movement as well.

kareltucek, 2022-03-11:
Axis locking is already implemented and will be part of next release. 

You can get a preview of the functionality with https://github.com/kareltucek/firmware . (It uses the 45 degrees angle solution. It is enabled by default for all navigation modes except cursor.)

DarwinAwardWinner, 2022-03-11:
That's good to hear! I'll try it out when I get the chance.

closed 2022-03-11 by kareltucek