UltimateHackingKeyboard/firmware #457: Under Linux, UHKv2 fails to be detected (device descriptor read/64, error -71 or device not responding to setup address))

AkechiShiro, 2022-03-11:
Since I've received my UHKv2 in February, I've been noticing this issue under multiples Linux machines.

Firmware : 8.10.2
Agent version: 1.5.17
Computer : Lenovo Thinkpad T495 has an AMD Ryzen 7 3700U (maybe relevant)
OS: ArchLinux (Linux)
But the issue also happens under an Intel I7 laptop running on ArchLinux with the uhk-agent installed.

I have been affected by the following issue (from my dmesg) :
```
usb 2-2: new full-speed USB device number 25 using xhci_hcd # Plugged USB
usb 2-2: device descriptor read/64, error -71
usb 2-2: device descriptor read/64, error -71                                # Unplug
usb 2-2: new full-speed USB device number 26 using xhci_hcd # Plugged USB
usb 2-2: device descriptor read/64, error -71
usb 2-2: device descriptor read/64, error -71
usb usb2-port2: attempt power cycle                                            # Unplug
usb 2-2: new full-speed USB device number 27 using xhci_hcd # Plugged USB
usb 2-2: Device not responding to setup address.
usb 2-2: Device not responding to setup address.
```

I have also tried, using a USB-C dock/hub (or directly USB-C), had the same errors, it happens whether the UHK agent is running or not, the UHK agent looks like it's trying to query and query again and again with no success.

Sometimes it gets successful when I change the USB port, or just plug in USB-C (fails then plug back in USB 3.0 works).

Looking at the uhk-agent, I've seen this : 
```
23:31:36.773 › [AppEffect][processStartInfo] payload: {
  commandLineArgs: { modules: false },
  platform: 'linux',
  osVersion: '5.16.12-arch1-1',
  udevFileContent: '# Ultimate Hacking Keyboard rules\n' +
    '# These are the udev rules for accessing the USB interfaces of the UHK as non-root users.\n' +
    '# Copy this file to /etc/udev/rules.d and physically reconnect the UHK afterwards.\n' +
    'SUBSYSTEM=="input", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="612[0-7]", GROUP="input", MODE="0660"\n' +
    'SUBSYSTEMS=="usb", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="612[0-7]", TAG+="uaccess"\n' +
    'KERNEL=="hidraw*", ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="612[0-7]", TAG+="uaccess"\n'
}
```

There is one thing that caught my eyes here, the GROUP="input", could it be related to the user groups on Linux ? Do my main user need to be inside the `input` group in order for the first udev rule to work as non-root ?

This error during startup (but probably not related to the issue but to Electron itself I guess): 
```
[254927:0311/233133.759784:ERROR:sandbox_linux.cc(376)] InitializeSandbox() called with multiple threads in process gpu-process.
```

Is anyone else having this issue from time to time ? It doesn't happen consistently, but it happens quite a lot of times, kind of unexpectedly.

mondalaci, 2022-03-12:
Please give a try to this firmware version: https://github.com/kareltucek/firmware/files/8153164/uhk-firmware-9.0.0-df83a18.tar.gz

AkechiShiro, 2022-03-12:
I'm still having the same exact issue, firmware update went alright, I kept the logs just in case, I also have this log of before the firmware update (I tested one USB-C, one USB 3.0 port on the left of my computer) both connections failed.
I then tested one on my USB-C dock/hub, it failed as well.

Finally, I had one last USB 3.0 port that was on the right of the laptop, and that one worked, here are the logs from the UHK agent (mind again that theses logs are under firmware 8.10.12).
```
20:36:54.575 › [UhkHidDevice] Devices before checking permission:
20:36:54.577 › [UhkHidDevice] Available devices unchanged
20:36:54.831 › [UhkHidDevice] Devices before checking permission:
20:36:54.833 › [UhkHidDevice] Available devices unchanged
20:36:55.085 › [UhkHidDevice] Devices before checking permission:
20:36:55.087 › [UhkHidDevice] Available devices unchanged
20:36:55.341 › [UhkHidDevice] Devices before checking permission:
20:36:55.343 › [UhkHidDevice] Available devices unchanged
20:36:55.597 › [UhkHidDevice] Devices before checking permission:
20:36:55.599 › [UhkHidDevice] Available devices unchanged
20:36:55.853 › [UhkHidDevice] Devices before checking permission:
20:36:55.854 › [UhkHidDevice] Available devices unchanged
20:36:56.146 › [UhkHidDevice] Devices before checking permission:
20:36:56.151 › [UhkHidDevice] Available devices changed.
# Plugged on the right USB 3.0 port, UHK keys lights up
20:36:56.151 › [UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"/dev/hidraw2","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":0,"usagePage":65280,"usage":1}
20:36:56.152 › [UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"/dev/hidraw3","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":1,"usagePage":1,"usage":6}
20:36:56.153 › [UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"/dev/hidraw4","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":2,"usagePage":12,"usage":1}
20:36:56.153 › [UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"/dev/hidraw5","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":3,"usagePage":1,"usage":128}
20:36:56.154 › [UhkHidDevice] hasPermission TypeError: cannot open device with path /dev/hidraw2
    at new HID (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/node_modules/node-hid/nodehid.js:49:17)
    at t.UhkHidDevice.hasPermission (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1297595)
    at t.UhkHidDevice.<anonymous> (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1297923)
    at Generator.next (<anonymous>)
    at /tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1364453
    at new Promise (<anonymous>)
    at l (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1364198)
    at t.UhkHidDevice.getDeviceConnectionStateAsync (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1297792)
    at t.DeviceService.<anonymous> (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:605364)
    at Generator.next (<anonymous>)
20:36:56.163 › [DeviceService] Device connection state changed to: {
  bootloaderActive: false,
  zeroInterfaceAvailable: true,
  hasPermission: false,
  halvesInfo: {
    areHalvesMerged: true,
    leftModuleSlot: 0,
    isLeftHalfConnected: true,
    rightModuleSlot: 0
  },
  hardwareModules: { moduleInfos: [], rightModuleInfo: {} },
  multiDevice: false,
  connectedDevice: {
    id: 2,
    name: 'UHK 60 v2',
    vendorId: 7504,
    keyboardPid: 24868,
    bootloaderPid: 24867,
    buspalPid: 24865
  }
}
20:36:56.163 › [DeviceRendererService] dispatch action [device] connection state changed
20:36:56.418 › [UhkHidDevice] Devices before checking permission:
20:36:56.422 › [UhkHidDevice] Available devices unchanged
20:36:56.423 › [UhkHidDevice] hasPermission TypeError: cannot open device with path /dev/hidraw2
    at new HID (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/node_modules/node-hid/nodehid.js:49:17)
    at t.UhkHidDevice.hasPermission (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1297595)
    at t.UhkHidDevice.<anonymous> (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1297923)
    at Generator.next (<anonymous>)
    at /tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1364453
    at new Promise (<anonymous>)
    at l (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1364198)
    at t.UhkHidDevice.getDeviceConnectionStateAsync (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:1297792)
    at t.DeviceService.<anonymous> (/tmp/.mount_UHK.AgxGtKhe/resources/app.asar/electron-main.js:2:605364)
    at Generator.next (<anonymous>)
[... 202 lines cut ...]
20:54:48.341 › [UhkHidDevice] Device communication closed.
20:54:48.341 › [DeviceOperation] User config size: 32704
20:54:48.346 › [UhkHidDevice] Available devices unchanged
20:54:48.346 › [UhkHidDevice] Used device: {"vendorId":"0x1D50","productId":"0x6124","path":"/dev/hidraw2","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":0,"usagePage":65280,"usage":1}
20:54:48.349 › [DeviceOperation] userConfigSize: 3879
20:54:48.474 › [UhkHidDevice] Device communication closing.
[... cut ...]
20:54:48.572 › [UhkHidDevice] Used device: {"vendorId":"0x1D50","productId":"0x6124","path":"/dev/hidraw2","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":0,"usagePage":65280,"usage":1}
20:54:48.594 › [AppEffects] navigate to { commands: [ '/' ] }
```

Something I also noticed, when I plug the UHK on a port that doesn't work the first time, it looks like no matter how much I unplug replug, this doesn't help much also **udev rules are not firing** (according to `udevadm monitor --kernel --udev`) unless the USB port works from the first try.

AkechiShiro, 2022-03-12:
I've also added my user to the `input` group under Linux, but this did not help in any way, must've been totally unrelated.

mondalaci, 2022-03-12:
This looks like a one-off issue. Please send an email to support@ultimatehackingkeyboard.com to discuss it further.

closed 2022-03-12 by mondalaci