UltimateHackingKeyboard/firmware #461: Keyboard backlight stays on after putting computer to sleep

bfpepper, 2022-03-26:
Was told to open a new firmware ticket by László. 

When I press Fn + \ to put the computer to sleep, the keyboard stays lit. 

My config from the agent is attached. I am using Mac OSX 12.2.1 on the M1 chip. I am running firmware 8.10.12 for the UHK and 1.5.17 for the agent. 

[UserConfiguration.json.zip](https://github.com/UltimateHackingKeyboard/firmware/files/8355383/UserConfiguration.json.zip)

Please let me know if you need anything else from me.

mondalaci, 2022-03-27:
Thanks for reporting this issue!

The UHK firmware listens for USB sleep and wake-up events and disables/enables backlighting swiftly accordingly on Linux and Windows.

As for macOS, on my old MacBook, the UHK disables its LEDs 45 seconds after putting the MacBook to sleep, and its LEDs get enabled when waking it up. However, the second time I put my MacBook to sleep or afterward, the UHK doesn't disable its LEDs, so this is likely a macOS-specific firmware bug.

We'll look into the mentioned issue on newer Macs, too.

mondalaci, 2022-12-13:
Everyone affected by this issue, until it gets fixed, please use the LED fade timeout [smart macro](https://ultimatehackingkeyboard.com/blog/2022/11/18/agent-2-next-level-uhk-and-module-configuration).

nader-eloshaiker, 2023-02-12:
Just as an update, running Ventrua 13.1 on MacBook M1 Pro and this is still an issue.

cusxio, 2023-04-21:
Yea, same here. The smart macro option does not seem to fix the issue.

mondalaci, 2023-08-17:
The LED fade timeout option is not supposed to fix the issue directly but indirectly by disabling LED backlighting after a certain time of inactivity. In the meantime, this option has been added to "LED settings" page of Agent, so you can easily set it without using smart macros.

We'll fix the core issue, but I don't have an ETA yet.