UltimateHackingKeyboard/firmware #463: Finalizing smart macros 1.0.0

mondalaci, 2022-04-01:
@kareltucek I'm scrutinizing the smart macros 1.0.0 feature set before releasing a new firmware version to minimize future breakages. I'll keep this as short as possible.

1. On second thought, let's rename `xceleration` to `acceleration` as the former sounds weird. In the documentation, it can be explained that acceleration also affects deceleration.
2. `set module.trackpoint.navigationMode.blah caret` switches the base layer to caret mode instead of getting ignored and throwing an error.
3. `set module.blah.navigationMode.fn2 caret` doesn't throw an error in `$onInit`, and neither does `set modulex.touchpad.navigationMode.base cursor`. As a rule of thumb, every command and argument should be sanitized, and errors should be thrown when justified.
4. When using `set module.trackpoint.navigationMode.shift caret`, Shift is held alongside the caret mode when using it via the Shift layer. I don't think that the modifier of modifier layers should compose this way.
5. What's the practical use of `set keymapAction.*`? Please provide some examples.
6. `set macroEngine.*` is very technical, and it will make the heads of most readers spin. Please provide some practical examples of its use.
7. I'm against introducing `backlight.strategy` and `backlight.constantRgb.rgb`. Please let's rather introduce separate variables for the 8 functional backlighting colors under the `backlight.colors` namespace. I appreciate the convenience of having constant RGB variable, but I'd much rather have Agent expose functional backlighting colors via a dedicated page in the near future via which all of the colors will be adjustable together.
8. Please describe the function of `set emergencyKey KEYID` in simple terms and provide an example of its use including a concrete command.

kareltucek, 2022-04-01:
> 1. On second thought, let's rename `xceleration` to `acceleration` as the former sounds weird. In the documentation, it can be explained that acceleration also affects deceleration.

If I may provide a second opinion on this, it sounds fine and serves the purpose. I don't really see a point in converting it back.

> 2. `set module.trackpoint.navigationMode.blah caret` switches the base layer to caret mode instead of getting ignored and throwing an error.
>
> 3. `set module.blah.navigationMode.fn2 caret` doesn't [...]

- [ ] Todo - sanitize above inputs.

> 4. When using `set module.trackpoint.navigationMode.shift caret`, Shift is held alongside the caret mode when using it via the Shift layer. I don't think that the modifier of modifier layers should compose this way.

This is obscure, exotic and inherently ambiguous usecase. Please do provide a reasonable motivation for even doing such a thing, and then for minding the actual result.

I don't think we should clutter the code by pseudofixes of ambiguous cases like this one.

> 5. What's the practical use of `set keymapAction.*`? Please provide some examples.

Usecase number one:

```
//Bind bound on fn.t
ifGesture 79 exec BindArrows
```

```
//BindArrows
set keymapAction.base.15 BindArrowsLeft
set keymapAction.base.16 BindArrowsDown
set keymapAction.base.17 BindArrowsRight
set keymapAction.base.8 BindArrowsUp
setLedTxt 500 ARR
```

Now, if I tap `fn+t+a`, right jikl cluster of current keymap gets rebound to arrows.  Useful for gaming, since having arrows bound in layer is extremely awkward, just as is having to maintain a separate keymap just in order to have arrows available.

Usecase number two:

```
//$onKeymapChange any
set keymapAction.base.193 BindMiddleClick
```

Since you have rejected the proposal for configurable touchpad actions...

(Admittedly, binding macros only is very clumsy, and I do intend to change the grammar to `macro MACRONAME | scancode SHORTCUT` rather than simply `MACRONAME`, which will require increase of major semver version.)

> 6. `set macroEngine.*` is very technical, and it will make the heads of most readers spin. Please provide some practical examples of its use.

Yes, it is a development option, which should be left out of any beginner-to-intermediate guides. 

Practical examples of its use - let's say that someone finds some inherent issue of the `blocking` execution strategy (which is still somewhat experimental), and needs to disable it. 

Or maybe the new strategy breaks someone's old macro flow. (I know there are people who actually do use the macro engine in so mind-bogglingly complex fasion that it is quite likely  - for instance Michael Grau.)

> 7. I'm against introducing `backlight.strategy` and `backlight.constantRgb.rgb`. Please let's rather introduce separate variables for the 8 functional backlighting colors under the [...]

Then why introduce a complex scheme of functional color variables?

Also, why does it offend you?

I do not intend to remove the feature from the fork unless and until there is a functionally equivalent feature which provides the same level of convenience. As far as I am aware, your backlighting proposal does not provide decent replacement, as it would still require setting all 8 colors manually, which is a cardinal nuisance. 

Assume the convenience of the following macro:

```
ifGesture transitive 75 final set backlight.constantRgb.rgb 255 32 0  //red
ifGesture transitive 84 final set backlight.constantRgb.rgb 192 255 0  /green
ifGesture transitive 91 final set backlight.constantRgb.rgb 128 192 255 //blue
ifGesture transitive 14 final set backlight.constantRgb.rgb 255 192 0 //yellow
ifGesture transitive 90 final set backlight.constantRgb.rgb 192 64 255 //violet
...
```

With this bound on fn's r, `fn+r+r` simply turns the keyboard red, `fn+r+g` green, and so on.

(I do mind functional highlighting because I do need to know meanings of my keys by heart anyways, so it does not bring me any added value, yet it looks extremely random, especially given irregularly spaced macros across my keymaps. I.e., it makes the backlight look just broken.)

> 8. Please describe the function of `set emergencyKey KEYID` in simple terms and provide an example of its use including a concrete command.

Again, a development feature.

Assume `set emergencyKey 79` in `$onInit`. Also assume `diagnose; printStatus` macro bound on key `a` (whose keyid is 79).

Let's assume that I mess up badly some macro which meddles with postponing logic, and it makes the keyboard to lock up time to time. (I.e., it makes it enter endless postponing loop.)  Now even though the keyboard reacts to no keys, I still can tap `a` in order to get execution state of all currently running macros. 

- [ ] Todo: check that my docs of 8 and 5 are intelligible.

----------------

Having said this, do you wish me to remove any of the above features from the stock firmware by means of an `#ifdef`?

Also, please let's keep the feature set minimal, not recklessly adding configuration variables and commands for everything, unless there is either a reasonable motivation for doing so or people actually requesting it...

mondalaci, 2022-04-02:
1 - Ok, I don't insist on changing `xceleration` back to `acceleration`, although I'm wondering whether it'd be possible or even practical to expose separate acceleration and deceleration settings.
4 - Fair enough. This is quite a niche use case, so let's leave it as it is.
5 - Thanks for elaborating! I will include such examples in the documentation because otherwise, the practical use of many commands is not trivial. Do you have an image of the key ids labeled on the UHK? It'll be necessary for decent usability of related commands. For the time being, however, please disable this command in master. I want to include it along with many others but later.
6 - Please also disable this command. I'll think about it more.
7 - I'm concerned about the non-orthogonality of your API. In other words, what you achieve could be achieved with the variables I proposed without introducing an additional backlighting mode. How about simply having a pseudovariable that would set all the 8 functional backlighting colors?
8 - Please disable `set emergencyKey` in master for the time being.

kareltucek, 2022-04-02:
> 1 - Ok, I don't insist on changing `xceleration` back to `acceleration`, although I'm wondering whether it'd be possible or even practical to expose separate acceleration and deceleration settings. 

That's an interesting question. I don't have a simple way, but do have some ideas. 

Question is: Is it worth to rethink the acceleration driver from scratch?

> 5 - Thanks for elaborating! I will include such examples in the documentation because otherwise, the practical use of many commands is not trivial. **Do you have an image of the key ids labeled on [...]

I do not understand that question.

At the moment, you figure out the keyid simply by the `resolveNextKeyId` command. Just map it on `k`, then focus text editor, press `k+a` and voila, you have your 79. (Better name for the command is appreciated.)

Figuring out more intuitive way of referencing the keys is a hard problem with questionable tradeoff. Especially given your intention of introducing non-US keycap sets. I don't see a way to do it in firmware, but I think it could be done quite cheaply in Agent. The only problem is integration. Ideal way would be to provide context-aware helper, but that would require us to tackle the grammar-aware autocompletion. (Which in my opinion is doable and not that expensive.)

> 7 - I'm concerned about the non-orthogonality of your API. In other words, what you achieve could be achieved with the variables I proposed without introducing an additional backlighting mode. How about simply having a pseudovariable that would set all the 8 functional backlighting colors?

You are loosing orthogonality by mimicking monochromatic mode via overwriting the colour classes. In my view on the matter, "functional highlighting", "custom led map" and "constant rgb backlight" are all different backlight drivers/building blocks/usecases/elements of the problem domain, and it is only correct that they do not interfere with each other. Even when their problem domains overlap.

Your way, we loose simple way to turn the functional highight on again. (I see that as a bug-by-design caused by non-orthogonality of such system.)

I consider this a "rule of three" situation, which postulates that we should not try to generalize until we have enough experience to generalize it correctly.

Also, with respect to future evolution of the system, I suspect that we are likely to implement more strategies/drivers in the future rather than reduce their number.

----

- [ ] todo: disable 5, 6, 8

mondalaci, 2022-04-02:
1 - I think it may be worth experimenting with splitting xceleration to acceleration and deceleration, but let's not mess with the driver until smart macros 1.0.0 is released.
5 - I meant to include an image featuring the key ids graphically in the newly created smart macros reference Agent sidebar. I'll do it eventually, as it should be very simple to retrieve the key ids with `resolveNextKeyId`.
7 - Well articulated. I had a specific vision of only having two backlighting modes, but chances are we'll end up with an increasing amount of modes, indeed.

kareltucek, 2022-04-02:
> 5 - I meant to include an image featuring the key ids graphically in the newly created smart macros reference Agent sidebar. I'll do it eventually, as it should be very simple to retrieve the key ids with resolveNextKeyId.

I see, I don't. I think Róbert should be able to render such an image easily - he has to have key ids available in the Agent anyways, since keyId is just the position of the key in the action array. (64-based - i.e., left half starts at 0, right at 64, left module at 128, right at 192)

mondalaci, 2022-04-06:
2. How about using an error global for this purpose and branching accordingly in parser functions?
3. I can't reproduce this either now. Please disregard it.

kareltucek, 2022-04-06:
> 2. How about using an error global for this purpose and branching accordingly in parser functions?

Sure!

mondalaci, 2022-04-13:
@kareltucek Have the above issues been resolved?

kareltucek, 2022-04-13:
I believe so, although I have forgotten to test the result. Will do so later today.

kareltucek, 2022-04-13:
Tested. I believe everything is working fine, so yes.

closed 2022-04-13 by mondalaci