UltimateHackingKeyboard/firmware #464: Feature Request: Add "One Shot Keys" Functionality

nwant, 2022-04-04:
Hello,

Wondering if you all would consider introducing [One Shot Keys](https://github.com/qmk/qmk_firmware/blob/master/docs/one_shot_keys.md) for One-Shot Modifiers and Layers. If so, it would also be great if the one-shot timeout could be configurable.

Thank you for your consideration!

likern, 2023-01-19:
Could someone elaborate how it's different to `ifGesture`? It looks like exactly One-Shot Key feature to me.

Layers are just a bunch of `ifGesture` with all desired key codes and `final tapKey ...`.

At least it's exactly how I just rebind this.
So tapping `space`, then tap `j` produces `:` key.
Space is this "One-Shot Key" activator.

Yes, it's tedious. You need to write a bunch of `ifGesture` for every desired key on another `layer`.

But definitely it's doable. Not sure about "One-Shot Modifiers", I suggest they will work the same way, might be `tapKeySeq S-:` is enough (produces `shifted :`). Well done,  copy paste and you have "One-Shot Modifiers".

kareltucek, 2023-01-19:
`oneShot` is expected to make the "modified" command last throughout the next key press. `ifGesture` is not.

If I understand you correctly, your workaround assumes that your entire layer is duplicitly listed in the macro. `oneShot` on the other hand is supposed to truly compose with next action, not to create separate computational branches for all possible next actions.

likern, 2023-01-20:
Yes, I do this 
```
# Right side top row
ifGesture 14 final holdKey *
ifGesture 7 final holdKey =
ifGesture 8 final holdKey "
ifGesture 9 final holdKey ;
ifGesture 10 final holdKey '
...
```

> If I understand you correctly, your workaround assumes that your entire layer is duplicitly listed in the macro.
Yes.

Yes, in terms of ergonomics it's a nightmare :) But macroses in general do not work very well with UI. Many things I can't make with UI and often have to switch to macro for key. So UI needs substantial refactor and redesign to accomodate macros engine nicely.

> oneShot is expected ...

Please, consider also more general solution to support `twoShot` and might be `threeShot`.
I already have use case for this. 
`Left Space,` then key on right side (`oneShot`) to produce important symbols like `,.{[=(`
`Left Space,` then  `D` key, then key on right side (`twoShot`) to produce less important symbols like `@#$%`.
So I never have to reach to 5th column of keyboard and pressing `Shift`, 

**What UI can't make**

**keyQuote** - reversing symbols on key when pressed with `Shift`
```
ifShift suppressMods tapKey '
ifNotShift tapKey "
```

**toBaseLayer** - setting LED screen when changing layer
```
setLedTxt 0 dev
toggleLayer base
```

**switchLayer** - custom or more advanced layer switching logic
```
ifLayer base exec toSymLayer
ifLayer mouse exec fromMouseToBase
ifLayer fn4 exec toBaseLayer
```
**shiftAndToSym** - primary role (as usual) and custom or more complex secondary role
```
holdKey iLS-
ifNotInterrupted exec switchLayer
```

**escapeAndToMouse**
```
ifShortcut timeoutIn 350 97 exec toMouseLayer
ifShortcut noConsume timeoutIn 350 84 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 82 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 81 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 80 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 79 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 77 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 75 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 74 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 73 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 90 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 89 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 88 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 87 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 91 final holdLayer fn4
ifShortcut noConsume timeoutIn 350 72 final holdLayer fn4
tapKey escape
```

kareltucek, 2023-01-20:
> Please, consider also more general solution to support `twoShot` and might be `threeShot`.

Yep, good point. Will need a refactor, but definitely possible.

robbievogt, 2023-05-31:
FWIW, I recently bound this macro to my `;` key:

```
ifShift final holdKey ;      // act as normal if shift is held

toggleLayer fn2
ifInterupted goTo HELD_LAYER  // additional key hit prior to release

// One-shot mode: Wait for a single follow key.
postponeKeys ifNotPending 1 ifNotPlaytime 2000 goTo @0  // wait up to 2 sec for a single pending key
ifPending 1 goTo @0  // wait for the pending key to be cleared

// Held layer mode: Stay in layer until release.
HELD_LAYER: ifNotReleased goTo @0  // wait for the original key release

delayUntil 10        // wait some additional time. without this we lose modifiers
unToggleLayer        // release the layer
```

It acts as a one-shot switcher to fn2 if tapped, or holds the fn2 if held (I'm using fn2 for symbols, which usually occur individually, but sometimes as bi- or more grams). Also shortcuts to `:` if shift is held.

It's using an active polling loop, so not sure how expensive it is to run, or whether there's a better way. Works really well for me though!

PR #628 (2023-07-09): Implement oneShot command.
Closes #464.
closed 2023-07-09 by mondalaci