UltimateHackingKeyboard/firmware #469: Make double tap speed adjustable

JibstaMan, 2022-04-16:
I'd like to have a setting to change the double tapping speed required to lock any of the layers, but especially the mouse layer. The mouse layer is tapped with the pinky finger and with the current configuration speed, I find it hard to tap it fast enough to have it lock the layer.

I'm not sure whether this is already (partly) implemented in UltimateHackingKeyboard/agent#501, but I can't seem to find the setting in Agent (1.2.0).

kareltucek, 2022-04-18:
I believe this is actually already implemented, as (e.g.) `set doubletapDelay 500`. (Is the word "delay" proper in this context?)

mondalaci, 2022-04-19:
@kareltucek I think "timeout" is a better naming. Would you rename the variable accordingly?

kareltucek, 2022-04-19:
Sure. I will create an alias and mark the old version as deprecated so that backward compatibility is not broken...

kareltucek, 2022-07-04:
Done and waiting for release. (Alternatively, already available at https://github.com/kareltucek/firmware .)

closed 2022-07-04 by kareltucek