UltimateHackingKeyboard/firmware #476: Confusing behavior of keys with secondary role when module is used

Felixoid, 2022-05-11:
1. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
2. Use Agent to update to the latest firmware:
   https://github.com/kareltucek/firmware/releases/tag/v8.10.12.kt.2

I've done everything from the template.

Here's my issue. When I use a key with Esc as primary and MOD as secondary role, modules (trackball, trackpad, and key cluster) behave as I expected in the mod layer, but then after releasing the Esc signal is sent. The problem is that the keyboard behaves halfway, it doesn't switch to a second role completely.

Previously (not sure until when) it didn't work at all, and keys w/ secondary role didn't affect modules' behavior at all. IMHO, it's worse now, because it's inconsistent.

kareltucek, 2022-05-11:
Can this still be the power issue?

Felixoid, 2022-05-11:
The mentioned behavior is reproduced on v1 with stock cables after upgrading to kt.2. MOD layer is not switched unless a key is pressed. For "mouse movements" it's not triggering MOD layer, so after key is released the ESC is sent.

BTW, v1 was with release 8.10.12 firmware, there the behavior as described below

> Previously (not sure until when) it didn't work at all, and keys w/ secondary role didn't affect modules' behavior at all. IMHO, it's worse now, because it's inconsistent.

mondalaci, 2022-05-12:
@Felixoid Can you reproduce the issue without the modules attached?

Felixoid, 2022-05-12:
I feel you don't get an issue 100%. It's about the behavior of the secondary role as layer MOD *and* modules mouse function.

If I press any _key_, it works and did work as expected.

mondalaci, 2022-05-12:
Sorry, but I have a hard time understanding the exact issue. Please describe how to reproduce the issue step by step as simply as possible.

kareltucek, 2022-05-12:
1. press secondary role key.
2. use module to move cursor, but do not tap any key. 
  - expected outcome: it behaves according to the navigation mode given by the secondary role.
  - actual outcome: correct.
3. release secondary role key. 
  - expected outcome: nothing, since the key in 1) was used as secondary role.
  - actual outcome: primary action is issued.

mondalaci, 2022-05-12:
By "actual outcome: correct", I assume you mean: "actual outcome: works as expected".

This is kind of an edge case. Do you think that the primary role should be triggered when a pointer action has been invoked since the activation of a secondary role action? I think it shouldn't be triggered.

Felixoid, 2022-12-07:
Hello, it's solved on the following combination of agent and firmware:

Agent version: 2.0.2  
 Right keyboard half firmware: UltimateHackingKeyboard/firmware:9.1.4  
 Left keyboard half firmware: UltimateHackingKeyboard/firmware:9.1.4  
 Key cluster firmware: UltimateHackingKeyboard/firmware:9.1.4  
 Trackball firmware: UltimateHackingKeyboard/firmware:9.1.4

Felixoid, 2022-12-07:
Ah, no, I am wrong. The primary action is still triggered

PR #639 (2023-07-16): Fix secondary roles for mouse
This also includes #638 which I therefore close. These should be 3 separate PRs, but they share a ton of code... ...so read them commit by commits.

Changes:
- Fix `ifPrimary` / `ifSecondary` which was not functional at all after reconnection onto default secondary role mechanism.
- Any module pointer action now triggers secondary roles.
- Artificial delay added between touchpad presses and releases, fixing doubletap-to-select-word for systems which ignore short (<20ms) mouse button releases (e.g., linux). 

Closes #565, Closes #334, Closes #476
closed 2023-07-16 by mondalaci