UltimateHackingKeyboard/firmware #478: Accelerate and decelerate navigation modes

mondalaci, 2022-05-19:
With configurable acceleration and deceleration values. I'm wondering if these values such be module-specific.

kareltucek, 2022-11-13:
What exactly do you mean? You mean that the mouse keys modifiers should apply with configurable coefficients, you mean a whole new mechanism, or you mean remaking the acceleration driver?

mondalaci, 2022-11-13:
Amusingly, I forgot what I meant to suggest exactly. When I suggested, maybe the acceleration and deceleration modifiers didn't affect module speeds which now they affect.

Per-module acceleration/deceleration coefficients could be useful, but nobody has asked them yet, so I better close this issue for now.

closed 2022-11-13 by mondalaci