UltimateHackingKeyboard/firmware #482: Mouse key mapping stops working, have to re-save layout to get it to work

Drizzt321, 2022-06-02:
I'm running the latest UHK, firmware 8.10.12 on keyboard and modules.

I've been taking my keyboard from home into work on the 1 day a week I work from there, and I don't know if it's disconnecting and moving it, disconnecting the modules and reconnecting them, or what, but some of the layout stops working. On the left key button module I've mapped Mouse Button 4 and Mouse Button 5 to the lower left and right buttons, which are to the OS the mouse back/forward buttons.

I have to go into UHK Agent, make a fake change (make a change on one button, then switch it back to what I need), before I can upload the keymap to the keyboard. This fixes the issue.

I've also, while creating this ticket, unplugged, then plugged the left key module and it still worked OK. It's something weird with moving it between systems. My main desktop is Win 10, my laptop (which I use in the office) is Linux, KDE Neon (based on Ubuntu 20.04).

mondalaci, 2022-06-02:
This issue is probably a duplicate of #361. Any further findings, feel free to follow up.

closed 2022-06-02 by mondalaci

mhantsch, 2022-06-02:
I think you are encountering a known issue. When I connect the keycluster module to the left half of my UHK while the UHK is already powered up and running, the mouse buttons on the module will not work until the keymap is reloaded internally. You can achieve this either by re-saving the keymap in Agent, unplugging and replugging the UHK, or simply switching keymaps back and forth directly on the UHK. I have learned that whenever I plug in the keycluster module, I then type a keymap toggle button twice to ensure the mouse buttons on the keycluster module work correctly.

See #348, #404.