UltimateHackingKeyboard/firmware #483: Received strange `HID Vendor-defined hCollection` driver from Windows Update for UHKv2

surban, 2022-06-02:
Firmware: 8.10.12.

Two days ago Windows Update installed a strange driver on one of the USB Input Devices belonging to UHKv2. The driver is called `OEM HID Vendor-defined hCollection` with INF file `acpibus.inf`. The driver file `acpibus.sys` shows  `Copyright Twinhead. All rights reserved.` in its file properties.

The hardware id on the driver is `HID\VID_1D50&PID_6124&REV_0102&MI_00`.

[image]

[image]

This could be a mistake by a vendor, i.e. pushing a driver on Windows Update that matches wrong hardware ids, or some elaborate malware attack. Has anybody observed something similar on their system?

mondalaci, 2022-06-02:
We have never heard about such an issue.

Which Windows version are you using?

surban, 2022-06-02:
> We have never heard about such an issue.

It only started two days ago.
 
> Which Windows version are you using?

Windows 11 Pro, Build 25126.

surban, 2022-06-02:
I've also opened a feedback item at the Windows Feedback Hub.

https://aka.ms/AAh3wth

surban, 2022-06-02:
These are the files from the offending driver, extracted from the Windows Driver Store.

[acpibus.inf_amd64_eebeb703a8a8ab10.zip](https://github.com/UltimateHackingKeyboard/firmware/files/8821969/acpibus.inf_amd64_eebeb703a8a8ab10.zip)

While the INF file says `Copyright (c) 1990-1999 Microsoft Corporation` in its header, the driver is clearly *not* from Microsoft. The authors just seems to have modified an SDK example. 

The SYS file has copyright `Copyright Twinhead. All right reserved.` in its file properties.

mondalaci, 2022-06-02:
Thanks for the details!

Can you reproduce this issue on another computer?

In any case, please keep us in the loop.

mondalaci, 2022-06-08:
Closing this issue due to the lack of replies, but will reopen if it needed.

closed 2022-06-08 by mondalaci

pcooke9, 2022-06-18:
Yesterday I was offered OEM - HIDClass - 18.16.37.672 as an optional update on Win 10 Pro 21H2 OS build 19044.1766.

While searching for info about the driver, I saw @surban commented about it on Reddit. He mentioned having it install on his UHKv2 (I have a UHKv1). Then I remembered seeing this issue posted a while back, so I figured I'd mention it here in case it's helpful to someone.

I haven't installed it, but as you can see in the screenshot, it says: "If you have a specific problem, one of these drivers might help. Otherwise, automatic updates will keep your drivers up to date." I'm guessing Windows thinks there's a problem with the UHK driver?

[image]

mondalaci, 2022-06-18:
@pcooke9 Thanks for the heads-up!

@benedekkupper Any insights regarding this issue?

surban, 2022-06-18:
@pcooke9 The phrase "If you have a specific problem, one of these drivers might help. Otherwise, automatic updates will keep your drivers up to date." just means that Windows Update has found a more specific driver than the generic HID driver for your keyboard. It does not mean that there is anything wrong with the currently used driver.

pcooke9, 2022-06-18:
> It does not mean that there is anything wrong with the currently used driver.

I was just being a bit sarcastic there. I assume if no one here knows where this driver came from, I'm certainly not gonna be installing it from Windows Update.

benedekkupper, 2022-06-18:
I don't have that much knowledge in reading this driver information, what relevant I found in the `.inf` was only this:
`%customCollection% = customCollection.Inst, HID_DEVICE_UP:FF00_U:0001`
What's more worrisome is that I found no reference to filters for USB VID or PID.

If that's really the case, it's totally nonsense of Microsoft to sign this, because HID usage pages in the range of [0xFF00, 0xFFFF] are vendor defined, which means that each vendor can interpret those pages as they like, and the USB host has to consider at least the VID before trying to match that usage (page) to any known driver.

If there is a VID based filtering that is not visible for me, then Microsoft is not directly to blame. The UHKs are using the OpenMoko VID sublicensing scheme, and so can any other open-source USB project. It is not sensible from the submitters of this driver ("Twinhead") though, as they should anticipate conflicts on this VID.

If there is a VID+PID based filtering that I'm not seeing, then the submitters have clearly targeted the UHK to be hooked to their driver. We would need to address this with Microsoft immediately.

So how do we find out which is the case? We need to test how the affected Windows machines assign this suspect driver on the UHK, if we flash it with a different VID / PID. The agent would need a customization for this as well, so the original firmware can be restored on the devices afterwards.

surban, 2022-06-18:
Note that other devices, such as VR headsets, are [affected as well](https://www.reddit.com/r/WindowsHelp/comments/v2ygu9/comment/icts6rl/?utm_source=share&utm_medium=web2x&context=3).

The [Windows Update catalog page](https://www.catalog.update.microsoft.com/ScopedViewInline.aspx?updateid=0f98c4bd-e6c3-49ea-b733-066f689f75a2#PackageDetails) lists `acpi\qmap0001` as supported hardware id. So maybe this is taken into account as well when choosing whether to install this driver.

My theory is that this is some kind of HID filter driver that is supposed to be installed on specific machines matched via an ACPI device. But someone screwed up and the ACPI device name is quite common; thus the driver gets installed on a wide variety of PCs that have no use to it.

benedekkupper, 2022-06-18:
That seems to confirm that there is no USB VID:PID filtering at all, and every vendor-specific HID device gets caught in it. Meaning that Microsoft messed things up here. Unfortunately I cannot access your Feedback Hub link with my account for God knows what reason.

mondalaci, 2022-06-19:
@benedekkupper Thank you for sharing your insights!

@surban Can you submit a bug report to Microsoft? Feel free to link this issue.

pcooke9, 2022-06-23:
Windows Update stopped offering the driver on my system today.

surban, 2022-06-24:
> @benedekkupper Thank you for sharing your insights!
> 
> @surban Can you submit a bug report to Microsoft? Feel free to link this issue.

I don't have the ability to submit bug reports to Microsoft other than via Feedback Hub, which I already did.

surban, 2022-10-01:
The faulty update seems to have been removed from Windows Update.