UltimateHackingKeyboard/firmware #484: Left keyboard half unreliable when split

alethiophile, 2022-06-06:
Intermittently, I'm getting issues with the behavior of the left keyboard half. Keystrokes from the left half of the keyboard will be randomly dropped or doubled, or a held key flicker off and on. The LED display will go out and remain off until the USB cable is unplugged and reinserted in the computer.

These problems occur only when the keyboard is split and connected using the phone cable, not when it's together. The problem is intermittent and only began recently. I have tried using another phone cable; however, it's near the cutoff for maximum length, and that cable doesn't work at all (keystrokes don't register).

mondalaci, 2022-06-06:
The alternative bridge cable should be the culprit. According to https://ultimatehackingkeyboard.com/modding/can-i-have-a-longer-bridge-cable , please make sure that the cable is as short as possible, and straight cables are preferred over spiral cables.

closed 2022-06-06 by mondalaci

alethiophile, 2022-06-06:
Sorry, I should be clearer.

The problems I'm having are with the standard bridge cable that came out of the box. I suspected a problem with that one, and tried another. The other one is near the maximum length and doesn't work at all.

mondalaci, 2022-06-06:
Thanks for the clarification!

The cause of the issue is probably power-related. Please narrow down the cause based on points 1 to 9 of [this guide](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys), and let me know your findings.

alethiophile, 2022-06-06:
1. When the keyboard halves are merged, there's no problem. This isn't a solution, though, since I need it split to use it comfortably.
2. I am already using the stock bridge cable.
3. I am already using the stock USB cable.
4. The problem remains whether connected directly or via a hub.
5. I haven't tried any other computers. However, this computer has successfully powered the UHK for two years; the problem only started in the past week.
6. I don't have a powered USB hub available. However, this hub has always worked before (it's a USB-C hub that should have ample power available).
7. The problem remains when on a battery-powered laptop.
8. The problem remains no matter the location.
9. The problem remains even with LEDs disabled.

I suspected some physical damage to the cable, since I've been carrying it around in a bag and it might have been banged up. However, the only third-party cable I could get doesn't work at all. I've ordered another official UHK cable, but won't have access to it for a few weeks.

alethiophile, 2022-06-06:
One other note: when I open Agent, the left half seems to be flickering in and out of connectivity (intermittently grays out in the UI). I don't know what this implies.

mondalaci, 2022-06-07:
Please shoot an email to support@ultimatehackingkeyboard.com and let's continue there.