UltimateHackingKeyboard/firmware #489: Feature request: show num lock status using background illumination

surban, 2022-06-13:
For some applications I require the use of the numeric pad keys. I've applied an appropriate mapping to my UHKv2, which includes `Fn+~` being mapped to `Num Lock`.

I would now like the background illumination of the `~` key to show the status of `Num Lock` while `Fn` is being held, for example red for on and blue for off.

Ideally this should also be possible for `Caps Lock` and `Scroll Lock`.

closed 2022-06-13 by mondalaci

surban, 2022-06-13:
As an alternative I would be happy to remap the triangle in the LED display to show Num Lock status, if this is easier to implement.

kareltucek, 2022-06-13:
> @kareltucek Does [your fork](https://github.com/kareltucek/firmware/) support this use case, or do you have a workaround in mind? I'm closing this issue, but we'll follow up.

No and no. (Except for via #322.)

> @kareltucek Any objections?

It sounds far too random. I don't think we should try to support every single feature request just because smart macros make it easier that modifying the native config format.

Why exactly not to support the original suggestion? If the answer is computational overhead, why exactly is that a problem?

Alternatively, it might be possible to introduce building blocks which would allow userspace implementation of such feature. That is, `$onLayerChange` macro event and `ifStateKey { capsLock | numLock | scrollLock }`, `set backlight.key.KEYID.rgb ...` commands.

surban, 2022-10-01:
I would say this is not very niche, given that [Windows allows direct entry of ASCII characters using the numpad](https://support.microsoft.com/en-us/office/insert-ascii-or-unicode-latin-based-symbols-and-characters-d13f58d3-7bcb-44a7-a4d5-972ee12e50e0). This is quite useful when you need symbols such as ° that are not on the keyboard.

mondalaci, 2022-10-01:
Fair enough. I plan to add a dedicated icon for num lock on the display of upcoming UHKs. But I don't plan to repurpose the icons of existing UHKs.