UltimateHackingKeyboard/firmware #490: Mac command/mod key press sequence problem

benjaminwood, 2022-06-17:
I initially submitted this to customer support, but they asked me to open an issue here.

I recently connected my V2 to a Mac and I'm experiencing a problem that I haven't been able to figure out.

On macOS, CMD + arrow left/right goes moves the cursor to the beginning or end of the line.

On my UHK V2, I've mapped arrow keys to MOD + "jkl;" (vim natural). The command key is in the position directly to the left of the MOD. The problem is that the normal behavior of moving the cursor to the beginning/end of the line only works when I press the command key before the MOD key. If I press the MOD key before the command key, the arrow keys move a single character at a time and do not skip to beginning/end of the line.

Any ideas what may cause this or how I can fix it? Thank you!

mondalaci, 2022-06-17:
@kareltucek Any ideas?

kareltucek, 2022-06-17:
Sounds like cmd is not mapped in MOD layer...

benjaminwood, 2022-06-17:
Well this is awkward. 😬 

Thanks @kareltucek - that is exactly what the problem was.

Closing this now as it was user error!

kareltucek, 2022-06-17:
> Well this is awkward. 😬

Not really.

kareltucek, 2022-06-17:
Anyways, glad I could help.

(Yes, sending the unfinished comment and reopenin the ticket was accidental on my part.)

closed 2022-06-17 by kareltucek