UltimateHackingKeyboard/firmware #495: Manually installed LEDs on UHKv1 stopped working with recent firmware update

jceb, 2022-07-01:
Hi, I manually installed LEDs in my UHKv1 switches some years ago and they worked great. I wasn't able to control the brightness so they were on all the time. With the installation of the recent firmwares (sorry, I can't pin-point the exact firmware, currently running v8.10.12) the LEDs were suddenly turned off and I couldn't get them to light up ever since. My impression is that this change of behavior has to do with the introduction of UHKv2 that has built-in back lighting.

mondalaci, 2022-07-01:
Hi there! Please try to pinpoint the exact firmware version that introduced this issue. If you cannot pinpoint it, please let me know why and I'll do my best to help.

jceb, 2022-07-01:
I ran a bunch of tests and realized that this issue is related to @kareltucek 's firmware and not to the standard UHK firmware. I'll close this issue.

closed 2022-07-01 by jceb

kareltucek, 2022-07-01:
Chances are that it affects both. I guess I will have to solder on some leds and bisect the code to figure out what is wrong.

kareltucek, 2022-07-03:
@jceb should be good in https://github.com/kareltucek/firmware/releases/tag/v8.10.12.kt.4

jceb, 2022-07-04:
I'll check it tomorrow and let you know.

jceb, 2022-07-05:
@kareltucek works great thank you very much!