UltimateHackingKeyboard/firmware #504: Scrolling often unrecognised on macOS

danpalmer, 2022-08-14:
I've noticed that scrolling with the trackball module, or the trackball in the key cluster module, is often unrecognised for a period of time, and then starts working again after a second or two, or some other actions.

I've tried to find a pattern, but so far only have vague anecdotes.
- It seems to be unrecognised more frequently on un-focused windows, but not always.
- It seems to happen more commonly after a period of minutes of not using scrolling.
- Clicking often causes scrolling to work again.
- Switching from trackball to key cluster, or vice-versa, seems to cause the first one to work again.

My device configuration is attached in case it's useful.

This happens on macOS 12.x and the 13.x Beta – I can reproduce it on two separate Macs. On those Macs I don't run any software that is designed to interrupt and customise mouse behaviour. Other mice plugged into these computers work fine (one Logitech, one Microsoft). On one computer I plug in directly and on the other I plug in via a USB hub built into my monitor. Between all these differing factors, I suspect this is something to do with the UHK firmware, rather than another aspect of my setup, although if there's anything in the setup I should be testing to see if it makes a difference I'm happy to do so.

If there's any further debugging steps I can take to help figure this out do let me know.

[UserConfiguration.json.zip](https://github.com/UltimateHackingKeyboard/firmware/files/9334009/UserConfiguration.json.zip)

mondalaci, 2022-08-15:
Please flash the following firmware via Agent, and let me know whether it fixes the issue.

[uhk-firmware-9.0.0-beta.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/9337232/uhk-firmware-9.0.0-beta.tar.gz)

danpalmer, 2022-08-16:
Thanks @mondalaci, just flashed it and will see how it goes over the next day or two.

danpalmer, 2022-08-16:
This doesn't appear to have fixed it. The same behaviour is still happening.

Is this due to different ways of implementing scrolling at the USB HID level? Or is it that the UHK is sometimes not recognising that the user is attempting to scroll?

mondalaci, 2022-08-16:
The UHK should always recognize scrolling and send related USB HID events accordingly.

Let's focus on your trackball module first. Does it skip a beat every now and then when moving the pointer with it? Please [clean it](https://ultimatehackingkeyboard.com/manuals/trackball), and let me know if it resolves the issue.

danpalmer, 2022-08-17:
I have never noticed it skipping movement, only scrolling. I've cleaned it once recently and did notice an improvement to the feeling when using it, but didn't notice any difference with tracking of any kind, and it wasn't that dirty so I didn't expect it to improve.

danpalmer, 2022-08-17:
I've got some more confidence now:
- This issue has no effect on the pointing with the trackball
- This issue affects scrolling with the trackball module and key cluster trackball equally and always at the same time – if it works for one it works for the other, if it fails for one it fails for the other.

mondalaci, 2022-08-17:
How often is scrolling unrecognized?

Can you reproduce this issue on non-Mac computers?

If the above two questions don't reveal any new findings, can you follow the "advanced troubleshooting" section of [this guide](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys)?

danpalmer, 2022-08-18:
Under regular use this probably happens every hour or two. It usually resolves within a couple of seconds, but I've had times where it hasn't resolved for >10s and I've reverted to scrolling on my laptop's trackpad.

I haven't reproduced this on a non-Mac, but I'm afraid I don't have access to any non-Macs. The best I could do would be a virtual machine, but I wouldn't trust those as it's all going through layers of USB emulation.

I'll try the advanced troubleshooting as soon as I can. Thanks for your continued help on this!

chuanqisun, 2022-08-30:
I've noticed similar issues in Windows 11. I used a browser + javascript to look window.onscroll events. It turns on at some point, UHK just stopped sending the events. If I keep scrolling, the events will never fire again. I must pause and wait a few seconds and the scrolling will work again.

I'm using Key cluster. 8.10.12 firmware on Right/Left/Key cluster/Trackball

chuanqisun, 2022-08-30:
Does the UHK or Window have any debug utilities that allow me inspect raw events?

mondalaci, 2022-08-30:
@kareltucek Any ideas?

kareltucek, 2022-08-30:
Not much. What about power issues? Does the problem persist if you turn off key backlight? (Generally, https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys might contain some ideas on what to debug.)

Regarding debug features, none that would be end-user targeted. (But in case you would like to dive into C code, there is a mini library which can make the segment display show various statistics, or a printline-like mechanism to gather and print logs... )

Feel fee to give a try to this build:

[uhk-firmware-9.0.0-display_scroll_events.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/9454848/uhk-firmware-9.0.0-display_scroll_events.tar.gz)

- If you tap '`', segment display should show RES and then increment  whenever UHK sends a scroll event. (De-facto exit point.)
- If you tap 1, it should increment on every pointer delta change. (De-facto entry point.)

mondalaci, 2022-09-04:
Closing due to the lack of replies.

closed 2022-09-04 by mondalaci

danpalmer, 2022-09-07:
Sorry about this, I've been travelling and not had access to a computer for the re-compile.

I'll try out the steps in a week or so and re-open as necessary.