UltimateHackingKeyboard/firmware #519: Invalid "Launch application" keycodes in v9.0.0

Agrendalath, 2022-09-09:
Agent version: 1.5.17
Firmware version: 9.0.0
Kernel: 5.19.7-arch1-1 x86_64 GNU/Linux
Display server: Xorg

## Previous behavior - in 8.10.12
### Scancode: Launch Calculator (mapped with Agent)
`showkey` reports 140 ([KEY_CALC](https://github.com/torvalds/linux/blob/470776c6b03491a3e82c644737a6da5466b8b3eb/include/uapi/linux/input-event-codes.h#L218) - `XF86Calculator`).
`xev` detects the key press, but doesn't show anything useful.

### Scancode: Launch Email Client (mapped with Agent)
`showkey` reports 155 ([KEY_MAIL](https://github.com/torvalds/linux/blob/470776c6b03491a3e82c644737a6da5466b8b3eb/include/uapi/linux/input-event-codes.h#L235) - `XF86Mail`).
`xev` shows keycode 163, detects it as `XF86Mail`.

## Current behavior - in 9.0.0
### Scancode: Launch Calculator (mapped with Agent)
`showkey` reports 379 ([KEY_VCR](https://github.com/torvalds/linux/blob/470776c6b03491a3e82c644737a6da5466b8b3eb/include/uapi/linux/input-event-codes.h#L452)?).
`xev` doesn't detect the key press at all.

### Scancode: Launch Email Client (mapped with Agent)
`showkey` reports 150 ([KEY_WWW](https://github.com/torvalds/linux/blob/470776c6b03491a3e82c644737a6da5466b8b3eb/include/uapi/linux/input-event-codes.h#L228) - `XF86WWW`).
`xev` shows keycode 158, detects it as `XF86WWW`.

Notes:
1. I have tested these mappings with both left half of the keyboard and the key cluster module.
1. Remapping keys again to "Launch Calculator" or "Launch Email Client" when running 9.0.0 doesn't change the result.
1. Downgrading the firmware (without any additional changes, like remapping keys) restores the previous behavior.
1. I don't know if it's helpful, but one time the downgrade failed for me on the key cluster and trackpoint. When the keyboard was running 8.10.12, and the modules were running 9.0.0, the behavior was the same as described for 8.10.12.

kareltucek, 2022-09-09:
Yes, I can reproduce it.

kareltucek, 2022-09-09:
Should be fixed in [uhk-firmware-9.0.0-media_scancode_fix.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/9535167/uhk-firmware-9.0.0-media_scancode_fix.tar.gz)

Thanks for report!

closed 2022-09-09 by mondalaci

Agrendalath, 2022-09-09:
@kareltucek, yep, it works now. Thanks for fixing this so quickly!

PR #520 (2022-09-09): Fix 16bit media scancode overflow.
Closes #519.