UltimateHackingKeyboard/firmware #521: Missing values of the extra layers in secondary role enum

ert78gb, 2022-09-10:
There is the UltimateHackingKeyboard/agent#1840 issue about the missing extra layer options in the secondary role dropdown list.
I checked the [secondary role enums](https://github.com/UltimateHackingKeyboard/firmware/blob/53535a906e77fd45d527b8c51773b73cb94a69b5/right/src/secondary_role_driver.h#L33) and it does not contains the new values layer options.

If I see well the `Fn2`, `Fn3`, `Fn4`, `Fn5` values are missing.

mondalaci, 2022-09-10:
@kareltucek Can you implement this easily? If so, I'd like to make it part of the next Agent release.

kareltucek, 2022-09-10:
Oops. Should not be a problem.

mondalaci, 2022-10-21:
Resolved by f72bc9f0ff256bf129a9470a4f4d2ea2e0e90bcc.

closed 2022-10-21 by mondalaci