UltimateHackingKeyboard/firmware #523: Expose long git sha

mondalaci, 2022-09-25:
As it turns out, Agent needs the long SHA to resolve informal documentation resources via GitHub raw URLs, such as https://raw.githubusercontent.com/UltimateHackingKeyboard/firmware/885a92a0d6b730fcb74ac7fd7c256d10ab215750/doc/dist/agent-onInit.png

Please make `tag` only expose the git tag if present, and make the new `sha` field always expose the long SHA. If I'm correct, this change affects the module protocol, the device protocol, and packages.json

This feature is a prerequisite for Agent 2.0.0

kareltucek, 2022-09-25:
Does it? Short version https://raw.githubusercontent.com/UltimateHackingKeyboard/firmware/885a92a0/doc/dist/agent-onInit.png seems to work fine.

Do we actually want it as a new field as opposed to setting the tag to long sha?

ert78gb, 2022-09-26:
We web ui resolve the short sha, but the API does not.

https://api.github.com/repos/UltimateHackingKeyboard/firmware/git/commits/885a92a => does not work
https://api.github.com/repos/UltimateHackingKeyboard/firmware/git/commits/885a92a0d6b730fcb74ac7fd7c256d10ab215750 => works

kareltucek, 2022-09-26:
I see. And it does not universally accept tags either.

So be it new sha field then, indeed.

kareltucek, 2022-09-26:
Actually, @ert78gb, one more try: can you not simply resolve it using https://docs.github.com/en/rest/commits/commits#get-a-commit ?

Seems to work fine, and if you specify media type, it gives sha only. (Docs explicitly mention it probably is the intended usecase.)

```
$ curl -H "Accept: application/vnd.github.sha" https://api.github.com/repos/UltimateHackingKeyboard/firmware/commits/v9.0.0
3a7d4aeeb042aa48ceb1397c3915b084a924915a
$ curl -H "Accept: application/vnd.github.sha" https://api.github.com/repos/UltimateHackingKeyboard/firmware/commits/885a92a0
885a92a0d6b730fcb74ac7fd7c256d10ab215750
```

mondalaci, 2022-09-26:
We've tried it. The API is heavily rate-limited, and only allows 50 requests per hour per IP, so we'd use the raw URLs.

ert78gb, 2022-09-26:
I will check Karel's recommendation on the evening. My URL and his url are same . I use the `git/commits` path his url uses only the `commits`

kareltucek, 2022-09-26:
(I will wait then.)

ert78gb, 2022-09-26:
@kareltucek thanks for the hint. I found the method in the [octokit](https://github.com/octokit/rest.js) lib too.

mondalaci, 2022-09-26:
@kareltucek We still need the new `sha` field because tags may be deleted or changed.

kareltucek, 2022-09-26:
Well, I can build and publish a build and not push the code into github...

Why would we delete or change tags? Does such risk justify the added complexity and consumed right-half memory?

mondalaci, 2022-09-26:
Fair enough. Given your findings, a tag is not needed after all.

Thanks for your help!

closed 2022-09-26 by mondalaci