UltimateHackingKeyboard/firmware #524: How to change cursor speed (baseSpeed, speed, xceleration) of keycluster mini trackball

rftw, 2022-09-26:
Hello, 

In the macros [reference manual](https://github.com/kareltucek/firmware/blob/master/doc-dev/reference-manual.md#tm_11_0:~:text=modifies%20speed%20characteristics%20of%20right%20side%20modules.), it is stated that you can modify the speed characteristics of right side modules using `set module.MODULEID.{baseSpeed|speed|xceleration}` command. Is there any command to modify the speed characteristics of keycluster mini trackball (left side module)? I tested the above commands by replacing the  `MODULEID` with `keycluster` but it has no effect. Any pointers would be appreciated.

Thank you

kareltucek, 2022-09-27:
Which values have you tried? Due to its stepping nature, only whole number multiples make sense.

It should definitely work, although it is not an officially supported usecase.

rftw, 2022-09-27:
ok, they are working.

In the commands page, I saw the below and thought the maximum values they can take is 10. Now I tried with values greater than 10 and got it working as desired.
```
COMMAND = set module.MODULEID.baseSpeed <speed multiplier part that always applies, 0-10.0 (FLOAT)>
COMMAND = set module.MODULEID.speed <speed multiplier part that is affected by xceleration, 0-10.0 (FLOAT)>
```
Thank you

closed 2022-09-27 by rftw

kareltucek, 2022-09-27:
Those are recommended values, and are indeed meant with respect to right side modules. 

Glad it works!