UltimateHackingKeyboard/firmware #526: Make the `invertScrollDirection` invert in both x and y axis?

kareltucek, 2022-09-29:
Transfering from https://github.com/kareltucek/firmware/issues/108 :

> Currently the `module.MODULEID.invertScrollDirection` only inverts the y axis. Inverting the x axis as well makes it feel more like the so called natural scrolling, when there is overflow on both axes.

@mondalaci, what do you think? (I have no idea about natural scrolling conventions.)

mondalaci, 2022-09-30:
I think natural scrolling should invert both axes, but I doubt it's desired by many. I'd rather wait to see its demand and potentially introduce `module.MODULEID.invertScrollDirectionX`.

old9, 2022-10-02:
AFAIK the inversed version is actually the default behavior of the touchpad on most Windows or Mac laptops.

kareltucek, 2023-07-21:
> I think natural scrolling should invert both axes, but I doubt it's desired by many. 

Does this mean that you think there are users who strictly prefer the current state of affairs over the proposed "correct" natural scrolling?

mondalaci, 2023-07-21:
There is no "correct" scrolling. See https://bootcamp.uxdesign.cc/natural-scrolling-mac-vs-reverse-scrolling-windows-e48656275081

Given that we haven't received any other complaints about this, I still suggest introducing `module.MODULEID.invertScrollDirectionX`

kareltucek, 2023-07-21:
What does the link have to do with the horizontal vs vertical inconsistency?

Unless there is a reason to believe that there are users who prefer to have inverted vertical axis but non-inverted horizontal axis, I would vote for simply inverting both axes as requested. (And adding no additional config values.)

If there is such a reason, I think we should probably make the default `module.MODULEID.invertScrollDirection` invert _both_ axes, and introduce `module.MODULEID.invertScrollDirectionX` _and_ `module.MODULEID.invertScrollDirectionY`.

(I have no strong opinion on this, and not much knowledge regarding customs when it comes to horizontal scrolling... it is just that this suggestion seems wrong for reasons of its irregularity.)

mondalaci, 2023-07-22:
Users would have probably already spoken if they preferred inverting both axes. I wouldn't change the current `module.MODULEID.invertScrollDirection` behavior just to reveal those users who are dissatisfied with such a change. I also think that per-axis inversion has merit, and we shouldn't force a policy on users.

kareltucek, 2023-07-22:
So... what should I do about this ticket? Just introduce `module.MODULEID.invertScrollDirectionX`?

mondalaci, 2023-07-22:
I think it makes sense practical sense to introduce `module.MODULEID.invertScrollDirectionX`.

kareltucek, 2023-07-22:
Allright.

PR #645 (2023-07-24): Refactor invertScrollDirection.
Let's do it this way. Leave `invertScrollDirection` as is, but add explicit settings for both axes.

Closes #526.
closed 2023-07-24 by mondalaci