UltimateHackingKeyboard/firmware #527: Key cluster module connection state oscillation

mocknen, 2022-10-01:
Possibly related to #364, #518.

When the key cluster module is connected, the left half of the keyboard becomes unstable.
[image]

This is not only a problem in UHK Agent, but the actual keys become intermittently unresponsive.
The image below shows a scene where the 'u' key on the left side and the 'h' key on the right side are pressed alternately.
Besides, the odd '12345' also appears from time to time, making its use as a keyboard impractical.
[image]

[This](https://github.com/UltimateHackingKeyboard/firmware/files/9689416/uhk.log) is the log of the UHK Agent.
As you can see, `leftModuleSlot` is repeatedly changing between 0 and 2, and `isLeftHalfConnected` is sometimes false.

|    |  Version  |
| ---- | ---- |
| Firmware | 9.0.1 |
| UHK Agent | 1.5.17 |

mondalaci, 2022-10-01:
This is not a firmware issue.

Please [fasten your key cluster's steel guides](https://ultimatehackingkeyboard.com/repair/fastening-module-steel-guide-screws).

Any further issues, feel free to follow up.

closed 2022-10-01 by mondalaci

mocknen, 2022-10-01:
@mondalaci Thanks for your help.

After trying out the UHK v1 and v2 configurations for a while, I realized that the root of the problem lies in the modular cables connecting the left and right sides.
I have been using an unofficial cable for wider left/right spacing, which worked perfectly without the module, but seems to be unstable with the module.
Perhaps it is related to the power supply to the module.
It certainly works correctly with the official cable, but I will now try to find a longer cable that will work without problems.

Anyway, your point about it not being a firmware-related issue was completely correct, and I apologize for bothering you.
If you could add a note to the [knowledgebase](https://ultimatehackingkeyboard.com/modding/can-i-have-a-longer-bridge-cable) on this issue, perhaps it would be helpful to others.

mondalaci, 2022-10-01:
Thanks for your follow-up and suggestion!

[This article](https://ultimatehackingkeyboard.com/modding/can-i-have-a-longer-bridge-cable) already mentions the cable length limitation, and relevant knowledgebase articles link it, so I don't add more articles for the time being.