UltimateHackingKeyboard/firmware #531: Macro command execution via USB

mondalaci, 2022-10-07:
Allowing the execution of macro commands via the USB protocol would enable many advanced use cases.

The only implementation question is whether 63-character-long commands are sufficient practically. If not, a buffer has to be used, and multiple USB transfers are needed.

@kareltucek Thoughts?

kareltucek, 2022-10-08:
> The only implementation question is whether 63-character-long commands are sufficient practically

Yes, they are sufficient practically.

(The extended command set also allows starting another macro, so the 63-character limitation can be easily circumvented.)

> Thoughts?

Implementation wise it would be simpler and safer to expose a USB command to execute an existing macro. But indeed too impractical for short one-liners.

Regarding implementation, I guess the one sane way to do it is to dedicate one macro slot to this and hardcode it to contain a macro command action pointing to a _copy_ of the USB buffer. (And add some sync mechanisms, so it cannot be swapped when another instance of the macro is running.)

mondalaci, 2022-10-08:
Sounds good to me. 63 characters should be sufficient, then.

PR #578 (2022-12-20): Allow executing macro commands from usb.
Closes #531 .
closed 2022-12-20 by mondalaci