UltimateHackingKeyboard/firmware #536: Firmware 9.0.0 breaks KVM's hotkey sequence.

brianhsu, 2022-10-25:
OK, I encountered a really wired issue after upgraded to 9.0.0 firmware series.

I have an UGreen 2 port HDMI / USB keyboard mouse switch device, which has a feature that when input certain hotkey sequence, it switch to other port.

For example, when the device is connected like the following:

```
                  +----> Computer 1
  UHK <---- KVM --+
                  +----> Computer 2
```

When UHK is running at firmware < 9.0.0, I could switch to computer 1 or computer 2 with the following key sequence:

```
1. Press Left Ctrl
2. Press Left Ctrl Again
3. Press 1 or 2
```

But as soon as I updated my UHK to firmware 9.0.0, these hot key sequence just not working anymore. (Yet the computer still receive each key stroke correctly, showed by `xev` program under Linux.)

And if I downgrade my UHK firmware to version 8.10.12, the hotkey sequence works again.

Not sure why the firmware update will make the hotkey not working. Not really a big issue, I still could switch between computers using the physical button on the KVM device. But I guess I will just stay at 8.10.12 since it's still more convenient to use hotkey to switch between them.

mondalaci, 2022-10-25:
Firmware 9.0.0 contains some USB-related fixes that slightly change USB behavior. As most KVM switches, yours must be quite limited when parsing USB HID data, and it probably doesn't parse the updated USB data correctly.

Can you `git bisect` the firmware code between 8.10.12 and 9.0.0, test the code at every stage, and find the offending commit?

brianhsu, 2022-10-26:
Thanks for quick reply, I also intend to do that. Will update if I find the commit that cause this problem.

brianhsu, 2022-10-26:
After some testing, it seems the commit `88fe704eb57b0ee32303b279f71e60ecf796bb9a` cause this problem.

The firmware built from commit `55364` works fine (also need the one line change from `80e7` otherwise it bricks my UHKv2), but the firmware built from `88fe704eb57b` (also with the change from `80e7`) breaks the hotkey sequence.

Also, if I do the following steps, I could build a 9.1.1 based firmware that makes the hotkey sequence working again.

```
$ git checkout v9.1.1
$ git revert 88fe704eb57b0ee32303b279f71e60ecf796bb9a
// fix the conflict by keeps the code from the parent of 88fe704eb57b0
// build the firmware
```  

Unfortunately, I've no knowledge about USB HID / keyboard protocol or N-key rollover, so I couldn't pin point the problem. 

But I'm already pretty happy that I figured out a solution that works for me. Now I could enjoy new feature introduced in 9.0.0 series firmware while keep my hotkey sequence working. 😃

For your information, the following is the log of each commit.

```
commit 80e7996d80887aefaf1edd609f4f77c2bcd077d0 (HEAD -> fail)
Author: Benedek Kupper <kupper.benedek@gmail.com>
Date:   Thu Feb 17 15:32:18 2022 +0100

    allow USB controller to read from embedded flash

commit 88fe704eb57b0ee32303b279f71e60ecf796bb9a
Author: M. Hawn <matthewh@steamraven.com>
Date:   Fri Mar 12 10:42:49 2021 -0700

    nkey rollover using a bitfield
    
    Author: M. Hawn <matthewh@steamraven.com>
    
    Nkey-rollover - conflict fixes and refactors.
    
    Author: Karel Tuček <kareltucek@centrum.cz>
    
    * Take the basic concept from @steamraven, use the refactor from @kareltucek.
    * Clean up the report descriptor, drop the reserved byte and extend size
    to add all scancodes up to the modifier codes.
    * Always send the correct size report.
    * Drop the report that was assembled while the protocol changed.
    
    Signed-off-by: Benedek Kupper <kupper.benedek@gmail.com>

commit 553640d09bc5e314411f03b8aebbb1e93ea3ab93
Author: M. Hawn <matthewh@steamraven.com>
Date:   Sat Mar 13 10:47:06 2021 -0700

    Convert systemkeyboard keys to bitfield
    
    Author: M. Hawn <matthewh@steamraven.com>
    
    Rebase and add before the basic keyboard NKRO changes.
    Honestly I'm puzzled that the report descriptor already described bitfield
    usage, and the interface was operating as scancode bytes.
    
    Signed-off-by: Benedek Kupper <kupper.benedek@gmail.com>

```

mondalaci, 2022-10-26:
Excellent job figuring out the offending commit!

(I've slightly edited your comment; it's worth writing git SHAs of at least 7 characters in length and without backticks because that way GitHub markdown links them.)

@benedekkupper Should we revert the offending commit? According to the above, it should result in better KVM compatibility.

benedekkupper, 2022-10-26:
That commit is what implements NKRO, so if you revert it you also need to remove NKRO advertising from the product. We could add a compile time switch that can disable NKRO, or a specail button combination at boot. Or we can investigate why KVM is acting strange.

mondalaci, 2022-10-27:
@benedekkupper How hard would it be to use 6KRO vs NKRO based on the configuration of the keyboard?

benedekkupper, 2022-10-28:
Not that hard, this is already in place for the boot protocol, the only extra step is to maintain and provide two versions of the report descriptor, depending on the configuration.

mondalaci, 2022-10-28:
@benedekkupper Please expose an API for us to be able to set 6KRO vs NKRO. I assume such an API should be called before enumerating the UHK via `InitUsb()`. Something like `InitUsb(is6KRO) would also work.

hxv, 2022-10-29:
I have kind of similar problem - my laptop has encrypted disk, and I have to enter the password before an OS boots. On firmware 9.1.1 some keys do not work (password is not getting longer after pressing them - that is really all I can see), and some keys generate a beep. I also checked in BIOS - keyboard does not seem to work at all (but I can only check few keys there).

After reverting the commit 88fe704eb57b0ee32303b279f71e60ecf796bb9a everything works fine (thanks @brianhsu), so I'm guessing that solution for this issue will also work for me.
I'm leaving the comment for everybody else with similar problem, and I'll definitely test a fix as soon as it appears.

benedekkupper, 2022-11-04:
> @benedekkupper Please expose an API for us to be able to set 6KRO vs NKRO. I assume such an API should be called before enumerating the UHK via `InitUsb()`. Something like `InitUsb(is6KRO) would also work.

I will get to that soon. It might be possible to soft-disconnect the UHK after the setting changes so it's applied immediately. Another idea that came to my mind is that the UHK could test whether the host supports NKRO mode, by simulating Caps Lock button press, and seeing if the LED is turned on by the host or not.

mondalaci, 2022-11-07:
Let's use whatever approach works best. I love your NKRO to 6KRO graceful degradation suggestion, but we have yet to see if it works well in practice.

elmelik, 2023-02-08:
Is there any updates here?, I also have same issue with this topic. 😢

mondalaci, 2023-02-08:
@elmelik Not yet. Only firmware versions before 9.0.0 implement 6KRO and work as expected with affected KVM switches.

sunghospark, 2023-11-21:
For me the KVM switch (Tesmart didn't work at all until I flashed to PR branch from this issue https://github.com/UltimateHackingKeyboard/firmware/issues/287

It's working flawlessly after installing the firmware from this branch https://github.com/steamraven/firmware/tree/287-reorder-interfaces. Getting recognized by KVM fine and hotkey also works.
For anyone else to try, you will need a [matching agent](https://github.com/steamraven/agent/tree/interface-reorder) to work with this firmware.

mondalaci, 2024-02-26:
This should be fixed by https://github.com/UltimateHackingKeyboard/firmware/commit/f794a563b2053144c2305c84fa5d3bd70c026f69 and it will be included in the next firmware release.

closed 2024-02-26 by mondalaci

mondalaci, 2024-02-29:
Yes, firmware 11.0.0 includes the fix. Specifically, it includes USB interface reordering.

However, it does not include the possibility of switching from NKRO mode to 6KRO mode, which is what fixes the issue, according to your earlier comment. I wasn't sure that 6KRO was actually needed, and I was hoping that USB interface reordering solved the issue for everyone, but that doesn't seem to be the case.

I've just reopened this issue, and we'll implement the 6KRO option, too.

brianhsu, 2024-02-29:
Got it, thanks.

brianhsu, 2024-02-29:
Since agent 4.0 and firmware 11.0 have some pretty interesting new features. I built a custom 11.0 firmware version that disable NKRO, if anyone need it, could download it from https://github.com/brianhsu/firmware/releases/tag/kvm-11.0.0

It works for my KVM without any problem.

kennhardy, 2024-03-01:
Thanks @brianhsu. Worked wonders. I'm able to play on PS5, use KVM and still use the latest agent <3

riccardo-forina, 2024-12-03:
> Since agent 4.0 and firmware 11.0 have some pretty interesting new features. I built a custom 11.0 firmware version that disable NKRO, if anyone need it, could download it from https://github.com/brianhsu/firmware/releases/tag/kvm-11.0.0
> 
> It works for my KVM without any problem.

Hey, thanks for this! I was wondering why my keyboard wasn’t working with the ps5 and this did the trick 👍