UltimateHackingKeyboard/firmware #537: Hold mouse button and Release mouse macros do not have correct behavior

da11an, 2022-10-28:
`Hold mouse button: Left` (see image) and `Release mouse button: Left` macros should allow you to drag and drop by first pressing (and not holding) the hold key then later releasing by pressing the release key. The actual behavior seems to just be a full down/up left click on the hold, making it useless to support drag-and-drop. The desired behavior can be seen when using "mouse keys" in Windows.

[image]

Agent version: 2.0.0
Firmware: 9.1.1

mondalaci, 2022-10-29:
I can't reproduce the issue. The following macro drag & drops for me as expected:

[image]

Please paste your macro here for me to reproduce the issue.

da11an, 2022-10-29:
Thank you for looking at it László. I looked at the documentation last night and found that all keys are released at the end of a macro, so the behavior I'm observing is according to design. I was hoping to duplicate the behavior of windows mouse keys `0` and `.` that allow you to sustain a click (`0`) without holding a key down then later release with another key press (`.`). I will close this issue since it was a misunderstanding on my part, but I still hope to figure out the Windows mouse-keys-like desired behavior within a single macro.

[image]

closed 2022-10-29 by da11an

kareltucek, 2022-10-31:
Actually, mouse states can be controlled using `startMouse` and `stopMouse` and those *are* persistent.

I am wondering whether we should add a global macro state which would allow persistent key locking... Any thoughts @mondalaci?

da11an, 2022-11-04:
> Actually, mouse states can be controlled using `startMouse` and `stopMouse` and those _are_ persistent.
> 
> I am wondering whether we should add a global macro state which would allow persistent key locking... Any thoughts @mondalaci?

@kareltucek, I'm not seeing how to use `startMouse` as you're suggesting. I tried the following.
[image]
Also tried with just `startMouse mouseBtnLeft`. Both just set the LCD status to `ERR` without actually holding the left button. Am I doing something obviously wrong?

kareltucek, 2022-11-04:
Oops. Indeed buttons are not part of the syntax at the moment. Will fix that eventually.