UltimateHackingKeyboard/firmware #541: Feature Request: Add Auto Shift

Warrost, 2022-11-03:
Is it planned to implement something like [Auto Shift in QMK](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_auto_shift.md) ?

Can this work with numbers and special characters (short press '1' to type '1', longer press for '!') aswell?

Thanks in advance!

mondalaci, 2022-11-03:
@kareltucek Do extended smart macro commands support this use case? If not, is it planned?

Warrost, 2022-11-04:
Thanks for your response,

Wish I could help implementing it, however I'm just a lousy sysadmin.

kareltucek, 2023-07-18:
@Warrost what is the actual requested functionality? 

I understand that the auto shift is supposed to apply only to some scancode groups. Which are these? Non-white alphanumeric scancodes? What about arrows, del, backspace? What about macros? (In case of macros, we cannot simply look whether they correspond to a scancode keypress action.)

Warrost, 2023-07-19:
@kareltucek 
Hi there, I'll try to clear things up:

> I understand that the auto shift is supposed to apply only to some scancode groups. Which are these?

As you wrote, at least Non-white alphanumeric characters, but now that I think of it, is there an inherent limitation to those?

> What about arrows, del, backspace?

Exactly, those too.

Just an example: 

If I short press a key ('A' for example) it'll just work as intended, it'll write a lower case 'a'
How about a double/long press on the same key, but now it'll automatically send an upper case 'A'?
Same could work with the number/special character keys.

Afaik a similar feature (tap dancing if I remember correctly) is already implemented in QMK (you can find it [here](https://github.com/samhocevar-forks/qmk-firmware/blob/master/docs/feature_tap_dance.md) )

kareltucek, 2023-07-21:
> As you wrote, at least Non-white alphanumeric characters, but now that I think of it, is there an inherent limitation to those?

Sorry, I don't understand the question?

> Exactly, those too.

Can you be a bit more specific? 

If you look into https://github.com/UltimateHackingKeyboard/firmware/blob/master/right/src/lufa/HIDClassCommon.h , which scancodes should be affected by autoshift?

Or the other way around, which should be excluded?

kareltucek, 2023-07-25:
@Warrost bump?

PR #660 (2023-08-02): Autoshift
Closes #541