UltimateHackingKeyboard/firmware #542: delayUntil does not appear to work

Metaxal, 2022-11-05:
Hi, 

First of all, thanks for the amazing piece of software and hardware! The UHK is beautiful.

I tried the following macro:
```
set backlight.strategy constantRgb
set backlight.constantRgb.rgb 0 0 255
delayUntil 100
set backlight.constantRgb.rgb 0 255 0
delayUntil 100
set backlight.strategy functional
```
and mapped it to a Fn-A, but when calling it it jumps directly to the last line, as if the delays didn't work. The macro is executed since if the backlight is initially in constantRgb mode, it does switch to functional.

Am I doing something wrong?

UHK v2 60 ISO
Firmware: 9.1.2
OS: Linux

kareltucek, 2022-11-05:
Hi! 

Sorry about it, you are completely right. The bug was fixed a few days ago but has not been released yet.

Feel free to use this build in the meanwhile: 
[uhk-firmware-9.1.2-macro_state_fix.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/9944735/uhk-firmware-9.1.2-macro_state_fix.tar.gz)

Metaxal, 2022-11-05:
Awesome, thanks!

mondalaci, 2022-11-05:
@kareltucek Is the fix already in stock master? If so, which commit?

Metaxal, 2022-11-05:
If I understand correctly it's this one: https://github.com/UltimateHackingKeyboard/firmware/commit/cc4344752972128057f1c6b1cae3b674693a1c1d

mondalaci, 2022-11-05:
@Metaxal Thanks, that was my guess!

@kareltucek Which are the affected macro commands? I need them for a solid changelog.

kareltucek, 2022-11-05:
Yes, 55f4531.

Possibly every command which uses the action scope data storage. Especially conditions and modifiers.

mondalaci, 2022-11-06:
@kareltucek Can you phrase "every command which uses the action scope data storage" in a more digestible manner that users of macro commands can clearly understand?

kareltucek, 2022-11-06:
I would probably phrase it as "Fix memory corruption in macro engine core which caused undefined behaviour from time to time.".

If you insist on a comprehensive list, then the following ones are suspects. I don't know which ones are actually affected.

```
COMMAND = [CONDITION|MODIFIER]* COMMAND
COMMAND = delayUntilRelease
COMMAND = delayUntil <timeout (NUMBER)>
COMMAND = delayUntilReleaseMax <timeout (NUMBER)>
COMMAND = holdLayer LAYERID
COMMAND = holdLayerMax LAYERID <time in ms (NUMBER)>
COMMAND = holdKeymapLayer KEYMAPID LAYERID
COMMAND = holdKeymapLayerMax KEYMAPID LAYERID <time in ms (NUMBER)>
COMMAND = resolveSecondary <time in ms (NUMBER)> [<time in ms (NUMBER)>] <primary action macro action index (ADDRESS)> <secondary action macro action index (ADDRESS)>
COMMAND = write <custom text>
COMMAND = writeExpr NUMBER
COMMAND = goTo <index (ADDRESS)>
COMMAND = repeatFor <register index (NUMBER)> <action adr (ADDRESS)>
COMMAND = {pressKey|holdKey|tapKey|releaseKey} SHORTCUT
COMMAND = tapKeySeq [SHORTCUT]+
CONDITION = {ifShortcut | ifNotShortcut} [IFSHORTCUTFLAGS]* [KEYID]+
CONDITION = {ifGesture | ifNotGesture} [IFSHORTCUTFLAGS]* [KEYID]+
CONDITION = {ifPrimary | ifSecondary}
CONDITION = {ifDoubletap | ifNotDoubletap}
CONDITION = {ifInterrupted | ifNotInterrupted}
CONDITION = {ifReleased | ifNotReleased}
CONDITION = {ifKeyActive | ifNotKeyActive} KEYID
CONDITION = {ifKeyDefined | ifNotKeyDefined} KEYID
CONDITION = {ifKeyPendingAt | ifNotKeyPendingAt} <idx in buffer (NUMBER)> KEYID
CONDITION = {ifPending | ifNotPending} <n (NUMBER)>
CONDITION = {ifPendingKeyReleased | ifNotPendingKeyReleased} <queue idx (NUMBER)>
CONDITION = {ifPlaytime | ifNotPlaytime} <timeout in ms (NUMBER)>
CONDITION = {ifShift | ifAlt | ifCtrl | ifGui | ifAnyMod | ifNotShift | ifNotAlt | ifNotCtrl | ifNotGui | ifNotAnyMod}
CONDITION = {ifRegEq | ifNotRegEq | ifRegGt | ifRegLt} <register index (NUMBER)> <value (NUMBER)>
CONDITION = {ifKeymap | ifNotKeymap} KEYMAPID
CONDITION = {ifLayer | ifNotLayer} LAYERID
CONDITION = {ifRecording | ifNotRecording}
CONDITION = {ifRecordingId | ifNotRecordingId} MACROID
MODIFIER = suppressMods
MODIFIER = postponeKeys
MODIFIER = final
MODIFIER = autoRepeat
```

Metaxal, 2022-11-06:
> Feel free to use this build in the meanwhile: [uhk-firmware-9.1.2-macro_state_fix.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/9944735/uhk-firmware-9.1.2-macro_state_fix.tar.gz)

For what it's worth, this new firmware didn't fix the issue, despite a "Firmware update succeeded."
I don't observe any change from the previous firmware.

kareltucek, 2022-11-06:
That's weird. Just tested your macro with 500 intervals and it seems to work fine.  (Changes to blue, after half a second to green and after another half a second to functional.)

What happens if you extend the delays to 1000 (ms)?

Does decreasing backlight brightness help? Weird things happen when power supply is insufficient...

kareltucek, 2022-11-06:
Do you have the extended commands enabled?

 `set macroEngine.extendedCommands 1`

Metaxal, 2022-11-06:
Aha, I needed the extended commands indeed. It works now, thanks!

closed 2022-11-06 by Metaxal

Metaxal, 2022-11-06:
To finish this thread in colors, the fix enables a simple show-off macro (is there a wiki or some other place to post UHK macros?):

```
set macroEngine.extendedCommands 1
delayUntilRelease
set backlight.strategy constantRgb

set backlight.constantRgb.rgb 0 0 0
delayUntil 100

set backlight.constantRgb.rgb 0 0 64
delayUntil 100

set backlight.constantRgb.rgb 0 0 127
delayUntil 100

set backlight.constantRgb.rgb 0 0 191
delayUntil 100

set backlight.constantRgb.rgb 0 0 255
delayUntil 100

set backlight.constantRgb.rgb 0 64 0
delayUntil 100

set backlight.constantRgb.rgb 0 64 64
delayUntil 100

set backlight.constantRgb.rgb 0 64 127
delayUntil 100

set backlight.constantRgb.rgb 0 64 191
delayUntil 100

set backlight.constantRgb.rgb 0 64 255
delayUntil 100

set backlight.constantRgb.rgb 0 127 0
delayUntil 100

set backlight.constantRgb.rgb 0 127 64
delayUntil 100

set backlight.constantRgb.rgb 0 127 127
delayUntil 100

set backlight.constantRgb.rgb 0 127 191
delayUntil 100

set backlight.constantRgb.rgb 0 127 255
delayUntil 100

set backlight.constantRgb.rgb 0 191 0
delayUntil 100

set backlight.constantRgb.rgb 0 191 64
delayUntil 100

set backlight.constantRgb.rgb 0 191 127
delayUntil 100

set backlight.constantRgb.rgb 0 191 191
delayUntil 100

set backlight.constantRgb.rgb 0 191 255
delayUntil 100

set backlight.constantRgb.rgb 0 255 0
delayUntil 100

[... 291 lines cut ...]
set backlight.constantRgb.rgb 255 191 191
delayUntil 100

set backlight.constantRgb.rgb 255 191 255
delayUntil 100

set backlight.constantRgb.rgb 255 255 0
delayUntil 100

set backlight.constantRgb.rgb 255 255 64
delayUntil 100

set backlight.constantRgb.rgb 255 255 127
delayUntil 100

set backlight.constantRgb.rgb 255 255 191
delayUntil 100

set backlight.constantRgb.rgb 255 255 255
delayUntil 100

set backlight.strategy functional
```

I'm sure we can use registers and loops, which would save space on the UHK memory, but I'll learn about these later :)

kareltucek, 2022-11-06:
Nice! Thanks for sharing!

I use following as a color picker. I have it mapped on Fn+r, so fn+r+r yields red, fn+r+v yields violet and so on...

```
ifGesture transitive 0 final set backlight.constantRgb.rgb 255 0 0
ifGesture transitive 1 final set backlight.constantRgb.rgb 0 255 0
ifGesture transitive 2 final set backlight.constantRgb.rgb 0 0 255
ifGesture transitive 75 final set backlight.constantRgb.rgb 255 32 0  //red
ifGesture transitive 84 final set backlight.constantRgb.rgb 192 255 0  /green
ifGesture transitive 91 final set backlight.constantRgb.rgb 128 192 255 //blue
ifGesture transitive 14 final set backlight.constantRgb.rgb 255 192 0 //yellow
ifGesture transitive 90 final set backlight.constantRgb.rgb 192 64 255 //violet
ifGesture transitive 9 final set backlight.constantRgb.rgb 255 128 0 //orange
ifGesture transitive 73 final set backlight.constantRgb.rgb 192 32 0 //wine
ifGesture transitive 21 final set backlight.constantRgb.rgb 128 48 0 //brown
ifGesture transitive 22 final set backlight.constantRgb.rgb 255 192 32 //warm white
ifGesture transitive 65 final set backlight.constantRgb.rgb 0 0 0
ifGesture transitive 66 final set backlight.constantRgb.rgb 32 32 32
ifGesture transitive 67 final set backlight.constantRgb.rgb 64 64 64
ifGesture transitive 68 final set backlight.constantRgb.rgb 128 128 128
ifGesture transitive 69 final set backlight.constantRgb.rgb 192 192 192
ifGesture transitive 70 final set backlight.constantRgb.rgb 255 255 255
ifGesture transitive 89 final set backlight.strategy constantRgb
ifGesture transitive 82 final set backlight.strategy functional
ifGesture transitive 72 final set leds.enabled 0
ifGesture transitive 10 final set leds.enabled 1
```

> (is there a wiki or some other place to post UHK macros?):

The best I can think of is a community on Reddit and Discord...

> I'm sure we can use registers and loops, which would save space on the UHK memory, but I'll learn about these later :)

Maybe better don't. The language is very crude in this respect. It was never designed to be a scripting language.

mondalaci, 2022-11-06:
I plan to set up a Discourse, but I want to wait until January, at which point we should deliver all pending orders, and hence, the forum wouldn't be full of "when will I get my order" threads.