UltimateHackingKeyboard/firmware #546: Gamepad mode introduces weird behaviour in some games

valentinpalkovic, 2022-11-11:
In Call of Duty Warzone and also in some other games, it is not possible to choose your game controller input. I assume the first controller, which is found, is used to control the game. In the past, everything went fine with my UHK and a dedicated PS5 controller. I was able to control the game with the PS5 controller. I recently updated the firmware to 9.1.2, and things were going crazy. In some games, like Cyberpunk 2077, I could still control the game with my PS5 controller (USB). In Call of Duty Warzone through inputs from my PS5 controller were not recognized anymore. Instead, the UHK was behaving as the primary controller input :/

Workaround: I have to disconnect the UHK every time I want to play COD Warzone because otherwise, my PS5 controller is not used to control the game.

Question: Could we create a shortcut to disable the game controller mode? Or even introduce a setting in the UHK Agent? Does a possibility exist in Windows to configure the order of controllers (I didn't find any)?

Related: https://github.com/UltimateHackingKeyboard/firmware/pull/535

mondalaci, 2022-11-11:
@valentinpalkovic Thanks for reporting this issue! We'll make the gamepad USB interface toggleable from Agent, as suggested.

@benedekkupper The best solution I can think of is adding `force6kro` and `disableGamepad` booleans to `HardwareConfig`, and we'd initialize USB from `hardwareConfigurationReadFinished() according to the values read`. Would you send us such a PR?

mondalaci, 2022-11-13:
We've just disabled the USB gamepad interface in 45fd12a188548d0a55cfe5cd4ade59261f1f4dc0, and I'll release a new firmware version soon. We'll re-add the interface later, make its presence configurable, and plan to make it less disturbing by default.

closed 2022-11-13 by mondalaci